notro/fbtft #311: ili9488

kiwifruktish, 2015-08-17:
Im not sure if this is the right place, but is it possible to load ili9488 with flexb from fallowing information:

void Lcd_Init(void)   //ili9488
{
Lcd_Write_Com(0xE0);
Lcd_Write_Data(0x00);
Lcd_Write_Data(0x07);
Lcd_Write_Data(0x10);
Lcd_Write_Data(0x09);
Lcd_Write_Data(0x17);
Lcd_Write_Data(0x0B);
Lcd_Write_Data(0x40);
Lcd_Write_Data(0x8A);
Lcd_Write_Data(0x4B);
Lcd_Write_Data(0x0A);
Lcd_Write_Data(0x0D);
Lcd_Write_Data(0x0F);
Lcd_Write_Data(0x15);
Lcd_Write_Data(0x16);
Lcd_Write_Data(0x0F);
Lcd_Write_Com(0xE1);
Lcd_Write_Data(0x00);
Lcd_Write_Data(0x1A);
Lcd_Write_Data(0x1B);
Lcd_Write_Data(0x02);
Lcd_Write_Data(0x0D);
Lcd_Write_Data(0x05);
Lcd_Write_Data(0x30);
Lcd_Write_Data(0x35);
Lcd_Write_Data(0x43);
Lcd_Write_Data(0x02);
Lcd_Write_Data(0x0A);
Lcd_Write_Data(0x09);
Lcd_Write_Data(0x32);
Lcd_Write_Data(0x36);
Lcd_Write_Data(0x0F);
Lcd_Write_Com(0xB1);
Lcd_Write_Data(0xA0);
Lcd_Write_Com(0xB4);
Lcd_Write_Data(0x02);
Lcd_Write_Com(0xC0);
Lcd_Write_Data(0x17);
Lcd_Write_Data(0x15);
Lcd_Write_Com(0xC1);
Lcd_Write_Data(0x41);
Lcd_Write_Com(0xC5);
Lcd_Write_Data(0x00);
Lcd_Write_Data(0x0A);
Lcd_Write_Data(0x80);
Lcd_Write_Com(0xB6);
Lcd_Write_Data(0x02);
Lcd_Write_Com(0x36);
Lcd_Write_Data(0x48);  
Lcd_Write_Com(0x3a);   
Lcd_Write_Data(0x55);
Lcd_Write_Com(0xE9);
Lcd_Write_Data(0x00); 
Lcd_Write_Com(0XF7);
Lcd_Write_Data(0xA9);
Lcd_Write_Data(0x51);
Lcd_Write_Data(0x2C);
Lcd_Write_Data(0x82);
Lcd_Write_Com(0x11);
delay(120);
Lcd_Write_Com(0x29);
}
 
void Address_set(unsigned int x1,unsigned int y1,unsigned int x2,unsigned int y2)
{
  Lcd_Write_Com(0x2a); // Set_column_address 4 parameters
  Lcd_Write_Data(x1>>8);
  Lcd_Write_Data(x1);
  Lcd_Write_Data(x2>>8);
  Lcd_Write_Data(x2);
  Lcd_Write_Com(0x2b); // Set_page_address 4 parameters
  Lcd_Write_Data(y1>>8);
  Lcd_Write_Data(y1);
  Lcd_Write_Data(y2>>8);
  Lcd_Write_Data(y2);
  Lcd_Write_Com(0x2c); // Write_memory_start        
}

Sadly i am lookin for driver since i cant get my screen running.

kiwifruktish, 2015-08-21:
I did get better color now when making changes to https://gist.github.com/ont/52c91e4ff090ee805e27 with:

    import time
    import RPi.GPIO as g

    g.setmode(g.BOARD)

    pins = [
        7,    # cs
        11,   # wr - clock (rising edge)
        12,   # rs - data/command
        15,   # d1
        16,   # d2
        18,   # d3
        19,   # d4 (???)
        21,   # d6 (???)
        22,   # d5 (???)
        23,   # d7
        24,   # d8
        26    # rst
    ]

    CSX = 7
    DCX = 12
    WRX = 11
    RES = 26
    D = [
        15,   # d1
        16,   # d2
        18,   # d3
        19,   # d4 (???)
        22,   # d5 (???)
        21,   # d6 (???)
        23,   # d7
        24,   # d8
    ]

    def init():
        # prepare raspberry pins for output
        for x in pins:
            print x
            g.setup(x, g.OUT)
            g.output(x, 1)

    # hard reset sequence
    g.output(RES, 1)
    time.sleep(0.005)  ## 5 ms
    g.output(RES, 0)
    time.sleep(0.015)  ## 15 ms
    g.output(RES, 1)
    time.sleep(0.015)  ## 15 ms

    g.output(CSX, 1)
    g.output(WRX, 1)
    g.output(CSX, 0)

    def send_data(cs, byte):
    g.output(DCX, cs)
    g.output(WRX, 0)
    for i in xrange(8):
    g.output( D[i], (byte >> i) & 1 )
    g.output(WRX, 1)

    def cmd(byte):
        send_data(0, byte)

    def data(byte):
        send_data(1, byte)

    init()
    #cmd( 0xE9 )
    #data( 0x20 )
    cmd( 0x11 ) #Exit Sleep
    time.sleep( 0.100 )

    cmd( 0xD1 )     # VCOM Control
    data( 0x00 )    #    SEL/VCM
    data( 0x0C )    #    VCM
    data( 0x0F )    #    VDV

    cmd( 0xD0 )     # Power_Setting
    data( 0x07 )    #    VC
    data( 0x04 )    #    BT
    data( 0x00 )    #    VRH

    cmd( 0x36 )     # Set_address_mode
    data( 0x48 )
    # 0x48 = 0b01001000
    # 0 - top to bottom
    # 1 - right to left
    # 0 - normal mode of columns order
    # 0 - refresh top to bottom
    # 1 - RGB or BGR  (here BGR order)
    # 0 - <skip>
    # 0 - no hflip
    # 0 - no flip

    cmd(0xE0)
    data(0x00)
    data(0x04)
    data(0x0E)
    data(0x08)
    data(0x17)
    data(0x0A)
    data(0x40)
    data(0x79)
    data(0x4D)
    data(0x07)
    data(0x0E)
    data(0x0A)
    data(0x1A)
    data(0x1D)
    data(0x0F)

    cmd(0xE1)
    data(0x00)
    data(0x1B)
    data(0x1F)
    data(0x02)
    data(0x10)
    data(0x05)
    data(0x32)
    data(0x34)
    data(0x43)
    data(0x02)
    data(0x0A)
    data(0x09)
    data(0x33)
    data(0x37)
    data(0x0F)

    cmd(0xB1)
    data(0xB0)

    cmd(0xB4)
    data(0x02)

    cmd(0xF7)
    data(0xA9)
    data(0x51)
    data(0x2C)
    data(0x82)

    cmd(0XB0)
    data(0x00)

    cmd( 0x3A )     # Set pixel format
    data( 0x55 )
    cmd( 0xC1 )     # Display timing setting for normal/partial mode
    data( 0x41 )
    cmd( 0xC0 )     # Set Default Gamma
    data( 0x18 )
    data( 0x16 )
    cmd( 0xC5 )     # Set frame rate
    data( 0x00 )
    data( 0x1E )
    data( 0x80 )
    cmd( 0xD2 )     # Power setting
    data( 0x01 )
    data( 0x44 )
    cmd( 0xC8 )     # Set Gamma
    data( 0x04 )
    data( 0x67 )
    data( 0x35 )
    data( 0x04 )
    data( 0x08 )
    data( 0x06 )
    data( 0x24 )
    data( 0x01 )
    data( 0x37 )
    data( 0x40 )
    data( 0x03 )
    data( 0x10 )
    data( 0x08 )
    data( 0x80 )
    data( 0x00 )
    cmd( 0x2A )
    data( 0x00 )
    data( 0x00 )
    data( 0x01 )
    data( 0x3F )
    cmd( 0x2B )
    data( 0x00 )
    data( 0x00 )
    data( 0x01 )
    data( 0xDF )
    cmd( 0x29 ) #display on
    cmd( 0x2C ) #display on

    x1 = 100
    y1 = 100
    x2 = 200
    y2 = 200
    cmd(0x2a) # Set_column_address
    data(x1>>8)
    data(x1)
    data(x2>>8)
    data(x2)
    cmd(0x2b) # Set_page_address
    data(y1>>8)
    data(y1)
    data(y2>>8)
    data(y2)
    cmd(0x2c) # Write_memory_start

    # write pixel by pixel
    for x in xrange(100):
        for y in xrange(100):
            data(0xee)  ## low part of word
            data(0)     ## hight part of word

    g.output(CSX, 1)    ## end command / data session

[image]

This is the screen:
[image]
[image]

This is what happens when you run win32_screem.rar that was for the 3.6 screen:
[image]

It seams it shouldn't be that much different to get it running, sure its a cheap screen from aliexpress ;)

toxaq, 2015-08-21:
I have the exact same screen (even though I ordered a different one), so very keen on your progress!

kiwifruktish, 2015-08-21:
ok, now i have stuff on the screen :P but it is mirrored :( hahah

sudo modprobe fbtft_device name=flexpfb rotate=180 gpios=dc:18,reset:7,wr:17,cs:4,db00:22,db01:23,db02:24,db03:10,db04:25,db05:9,db06:11,db07:8

sudo modprobe flexfb width=320 height=480 buswidth=8 init=-1,0xb0,0x0,-1,0x11,-2,120,-1,0x3A,0x55,-1,0xC2,0x33,-1,0xC5,0x00,0x1E,0x80,-1,0xE0,0x00,0x04,0x0E,0x08,0x17,0x0A,0x40,0x79,0x4D,0x07,0x0E,0x0A,0x1A,0x1D,0x0F,-1,0xE1,0x00,0x1B,0x1F,0x02,0x10,0x05,0x32,0x34,0x43,0x02,0x0A,0x09,0x33,0x37,0x0F,-1,0x11,-1,0x29,-3

con2fbmap 1 1

Now mirroring is need to be fixed and rotate need to be fixed some how ;)

[image]

notro, 2015-08-21:
Register 0x36 is usually responsible for rotation/flip/mirror/bgr

kiwifruktish, 2015-08-21:
Thanks,worked.

Some stuff left to Workout.

kiwifruktish, 2015-08-21:
Everything's working now but the screen looks like it is "lagging" every lines updates way to slow, any use'?

notro, 2015-08-21:
Yes it's slow to drive a parallel bus with gpiolib.
If you use a Pi and can build your own kernel, you can add the optimization that was left out when fbtft went into staging: https://github.com/notro/fbtft/blob/master/fbtft-io.c#L133
See here for preformance numbers with this optimization: https://github.com/notro/fbtft/wiki/Performance#itdb28

mwilliams03, 2015-08-22:
If you want a visual comparison, I created a short video a month ago of two TFTs running next to each other, one with optimization and the other without the optimization.
https://www.youtube.com/watch?v=TyqHsEw4ids

notro, 2015-08-22:
That's a different optimization: DMA. The display we are talking about here is driven with a 8-bit parallel bus directly, no SPI. Upstream FBTFT uses gpiolib and sets one bit at a time (unless it's the same value as the previous one), whereas the old optimized FBTFT wrote directly to the hw register making it possible to set 8 bit in one go.

kiwifruktish, 2015-08-22:
For any one that whats:

pi@raspberrypi ~ $ cat /etc/modules
snd-bcm2835
fbtft_device name=flexpfb rotate=180 fps=60 gpios=dc:18,reset:7,wr:17,cs:4,db00:22,db01:23,db02:24,db03:10,db04:25,db05:9,db06:11,db07:8
#flexfb width=320 height=480 buswidth=8 init=-1,0xb0,0x0,-1,0x11,-2,120,-1,0x3A,0x55,-1,0xC2,0x33,-1,0xC5,0x00,0x1E,0x80,-1,0x36,0x48,-1,0xB1,0xB0,-1,0xE0,0x00,0x04,0x0E,0x08,0x17,0x0A,0x40,0x79,0x4D,0x07,0x0E,0x0A,0x1A,0x1D,0x0F,-1,0xE1,0x00,0x1B,0x1F,0x02,0x10,0x05,0x32,0x34,0x43,0x02,0x0A,0x09,0x33,0x37,0x0F,-1,0x11,-1,0x29,-3
#
# landscape mode
#
flexfb width=480 height=320 buswidth=8 init=-1,0xb0,0x0,-1,0x11,-2,120,-1,0x3A,0x55,-1,0xC2,0x33,-1,0xC5,0x00,0x1E,0x80,-1,0x36,0x28,-1,0xB1,0xB0,-1,0xE0,0x00,0x04,0x0E,0x08,0x17,0x0A,0x40,0x79,0x4D,0x07,0x0E,0x0A,0x1A,0x1D,0x0F,-1,0xE1,0x00,0x1B,0x1F,0x02,0x10,0x05,0x32,0x34,0x43,0x02,0x0A,0x09,0x33,0x37,0x0F,-1,0x11,-1,0x29,-3

kiwifruktish, 2015-08-22:
notro, stupid question, isent it just possible to change like fb_ili9486.c and make it as a new driver?

toxaq, 2015-08-23:
I asked my supplier for the driver and this is what I got back

```

/*************************************************/
void initi(void)
{
   res=1;
   delay(1);
   res=0;
   delay(10);
   res=1;
   delay(120);
/***************************************/ 
write_command(0xE0); 
write_data(0x00); 
write_data(0x04); 
write_data(0x0E); 
write_data(0x08); 
write_data(0x17); 
write_data(0x0A); 
write_data(0x40); 
write_data(0x79); 
write_data(0x4D); 
write_data(0x07); 
write_data(0x0E); 
write_data(0x0A); 
write_data(0x1A); 
write_data(0x1D); 
write_data(0x0F);  

write_command(0xE1); 
write_data(0x00); 
write_data(0x1B); 
write_data(0x1F); 
write_data(0x02); 
write_data(0x10); 
write_data(0x05); 
write_data(0x32); 
write_data(0x34); 
write_data(0x43); 
write_data(0x02); 
write_data(0x0A); 
write_data(0x09); 
write_data(0x33); 
write_data(0x37); 
write_data(0x0F); 

write_command(0xC0); 
write_data(0x18); 
write_data(0x16); 

write_command(0xC1); 
write_data(0x41); 

write_command(0xC5); 
write_data(0x00); 
write_data(0x1E); //VCOM
write_data(0x80); 

write_command(0x36); 
write_data(0x48); 

write_command(0x3A); //Interface Mode Control
write_data(0x55); //16BIT 

write_command(0xB1);   //Frame rate 70HZ  
write_data(0xB0); 

write_command(0xB4); 
write_data(0x02);   

write_command(0xE9); 
write_data(0x00);
 
write_command(0XF7);    
write_data(0xA9); 
write_data(0x51); 
write_data(0x2C); 
write_data(0x82);

//**********set rgb interface mode******************
write_command(0XB0);  //Interface Mode Control  
write_data(0x00);  //set DE,HS,VS,PCLK polarity

write_command(0xB6); 
write_data(0x30); //30 set rgb
write_data(0x02); //GS,SS 02£¨42
write_data(0x3B); 
 /**************************************************/ 
write_command(0x2A); //Frame rate control
write_data(0x00);
write_data(0x00);
write_data(0x01);
write_data(0x3F);

write_command(0x2B); //Display function control
write_data(0x00);
write_data(0x00);
write_data(0x01);
write_data(0xDF);

write_command(0x21);

write_command(0x11);
delay(120);
write_command(0x29); //display on
write_command(0x2c); 
	   
}
//******************************************
void LCD_Enter_Standby(void)
{
   write_command(0x28);//display off
   delay(10);
  write_command(0x10);
   delay(120);
   RGB_OFF();
   }
void LCD_Exit_Standby (void)
{
   RGB_ON();
   delay(10);
   write_command(0x11);// Exit Sleep/ Standby mode
   delay(120);
   write_command(0x29);
  }
```
Not sure if that's of any use to anyone.

@kiwifruktish How do I use that example you've got above?

notro, 2015-08-23:
> isent it just possible to change like fb_ili9486.c and make it as a new driver?

Sure it's possible.
I'm working on some patches and one of the big changes is moving away from having controller drivers to display drivers instead. The main reason for this is that the Device Tree guys won't allow register initialisation sequences in DT.
https://github.com/notro/linux-staging/commits/next

kiwifruktish, 2015-08-23:
@toxaq i used same example from the supplier, 

"sudo nano /etc/modules" and add to enable screen:
fbtft_device name=flexpfb rotate=180 fps=60 gpios=dc:18,reset:7,wr:17,cs:4,db00:22,db01:23,db02:24,db03:10,db04:25,db05:9,db06:11,db07:8
#flexfb width=320 height=480 buswidth=8 init=-1,0xb0,0x0,-1,0x11,-2,120,-1,0x3A,0x55,-1,0xC2,0x33,-1,0xC5,0x00,0x1E,0x80,-1,0x36,0x48,-1,0xB1,0xB0,-1,0xE0,0x00,0x04,0x0E,0x08,0x17,0x0A,0x40,0x79,0x4D,0x07,0x0E,0x0A,0x1A,0x1D,0x0F,-1,0xE1,0x00,0x1B,0x1F,0x02,0x10,0x05,0x32,0x34,0x43,0x02,0x0A,0x09,0x33,0x37,0x0F,-1,0x11,-1,0x29,-3
#
 # landscape mode
#
flexfb width=480 height=320 buswidth=8 init=-1,0xb0,0x0,-1,0x11,-2,120,-1,0x3A,0x55,-1,0xC2,0x33,-1,0xC5,0x00,0x1E,0x80,-1,0x36,0x28,-1,0xB1,0xB0,-1,0xE0,0x00,0x04,0x0E,0x08,0x17,0x0A,0x40,0x79,0x4D,0x07,0x0E,0x0A,0x1A,0x1D,0x0F,-1,0xE1,0x00,0x1B,0x1F,0x02,0x10,0x05,0x32,0x34,0x43,0x02,0x0A,0x09,0x33,0x37,0x0F,-1,0x11,-1,0x29,-3

"sudo nano /boot/cmdline.txt" Enable the Boot Splash Screen by adding:
fbcon=map:10 fbcon=font:VGA8x8 fbcon=rotate:3

kiwifruktish, 2015-08-25:
@notro 

I made changes to fb_ili9486.c
By changing "static int default_init_sequence[] = {" to:
/* this init sequence matches PiScreen */
static int default_init_sequence[] = {
	/* Interface Mode Control */
	-1, 0xb0, 0x0,
	/* Sleep OUT */
	-1, 0x11,
	-2, 120,
	/* Interface Pixel Format */
	-1, 0x3A, 0x55,
	/* Power Control 3 */
	-1, 0xC2, 0x33,
	/* VCOM Control 1 */
	-1, 0xC5, 0x00, 0x1E, 0x80,
	/* PGAMCTRL(Positive Gamma Control) */
	-1, 0xE0, 0x00, 0x04, 0x0E, 0x08, 0x17, 0x0A, 0x40, 0x79,
	          0x4D, 0x07, 0x0E, 0x0A, 0x1A, 0x1D, 0x0F,
	/* NGAMCTRL(Negative Gamma Control) */
	-1, 0xE1, 0x00, 0x1B, 0x1F, 0x02, 0x10, 0x05, 0x32, 0x34,
	          0x43, 0x02, 0x0A, 0x09, 0x33, 0x37, 0x0F,
	/* Sleep OUT */
	-1, 0x11,
	/* Display ON */
	-1, 0x29,
	/* end marker */
	-3
};

And started "fbtft_device custom name=fb_ili9486 rotate=90 gpios=dc:18,reset:7,wr:17,cs:4,db00:22,db01:23,db02:24,db03:10,db04:25,db05:9,db06:11,db07:8"

It worked after build, but still "laggy" because of gpiolib i guess.

But maybe you want to add a fb_ili9488 to next release ;)

notro, 2015-08-25:
I won't be making a fb_ili9488 because many of these drivers are mipi compliant so I will add a mipi module instead that will cover many of the fbtft supported controllers: https://github.com/notro/linux-staging/blob/next/drivers/staging/fbtft/lcdctrl/mipi-dbi.c

This also solves the problem of the hardcoded register 0x36 values for rotation. Not all displays are equal in this respect, so currently they have to use flexfb.

Then there will be display drivers that use this module: https://github.com/notro/linux-staging/blob/next/drivers/staging/fbtft/ada-mipifb.c

So I'm moving away from controller drivers to display drivers mainly for the DT reason I mentioned earlier.

If I still find fbtft fun to work on when all this is in place, I will have a look at the SMI driver and see if I can add support for that in fbtft: https://github.com/raspberrypi/linux/pull/1105
But anyway that will be months away. I would need to make a generic parallel bus subsystem first.

bigpaulie, 2015-09-10:
Guy's I don't know how you did this but after I added : 
fbtft_device name=flexpfb rotate=180 fps=60 gpios=dc:18,reset:7,wr:17,cs:4,db00:22,db01:23,db02:24,db03:10,db04:25,db05:9,db06:11,db07:8
flexfb width=480 height=320 buswidth=8 init=-1,0xb0,0x0,-1,0x11,-2,120,-1,0x3A,0x55,-1,0xC2,0x33,-1,0xC5,0x00,0x1E,0x80,-1,0x36,0x28,-1,0xB1,0xB0,-1,0xE0,0x00,0x04,0x0E,0x08,0x17,0x0A,0x40,0x79,0x4D,0x07,0x0E,0x0A,0x1A,0x1D,0x0F,-1,0xE1,0x00,0x1B,0x1F,0x02,0x10,0x05,0x32,0x34,0x43,0x02,0x0A,0x09,0x33,0x37,0x0F,-1,0x11,-1,0x29,-3

to my /etc/modules , all I can see it's just a black screen ...

kiwifruktish, 2015-09-11:
did you add any thing to cmdline.txt ? if not you have to add:

fbcon=map:10 fbcon=font:VGA8x8 fbcon=rotate:3

before rootwait i think. and reboot or just type "con2fbmap 1 1" when module is loaded if you dont want to restart for the moment.

bigpaulie, 2015-09-12:
@kiwifruktish , Thank you. 
You were right, also I had to use this FRAMEBUFFER=/dev/fb1 startx to start the x .
It is very very laggy .. :)

gibo65, 2015-10-08:
@kiwifruktish 
Hi, I have obtained the same yellow square with you, but how to di to redirect console on TFT screen?
Sorry but I'm dummy user ;-)
Thanks

callievi, 2015-10-09:
I found a kernel for the ili9488 lcd and a raspbian image with kernel for this display.
I attached the link of the page.

http://www.ittgroup.ee/et/ekraanid-ja-naidikud/626-lcd-tft-ekraan-395-480x320-raspberry-pi-le.html

In the bottom of the page there are two links,Kernel and
Raspberry Pi image, click one of them and enjoy!

Bye.

gibo65, 2015-10-10:
@callievi 
Hi, thanks to info but yunpan360.cn website release always an unknow error to download win32_3.95screem.zip 952.6Mb file so, I ask to show your cloaud public link to download correclly resource.
Thanks ;-)

PS https://www.raspberrypi.org/forums/viewtopic.php?t=122736&p=826897

gibo65, 2015-10-19:
[image]
Why with this distribution when I connect my wireless device IRQ32 disable and ext keyb don't work anymore? ;-/

callievi, 2015-10-19:
I do not know, I used raspbian modify kernel.
You have download the kernel file o the win32 file?

gibo65, 2015-10-22:
...maybe error config into etc/ files! Now I havn't.
into games-python foldere only 640x480 games trailer? ;-)
Thanks Bye

JunkmailVictim, 2015-11-02:
I've been trying to follow along on this thread, but am unfortunately not experienced enough in this environment to get my display as far along as everyone here has.  Mine is still that ugly faded image that doesn't fill up the screen.

Is there any possibility anyone would be willing to create and share an image that has this display working properly?

Thanks.

gibo65, 2015-11-02:
FYI
[image]

JunkmailVictim, 2015-11-03:
Thanks.  That's very thoughtful.  I had done that already, but my display is the same as that faded image that doesn't fill the screen in the earlier picture.  I'm hoping someone sorts out the display, and then makes a new image.  The photo of your screen looks great.

by2coffee, 2015-11-16:
Strangely, the display is working on a Raspberry Pi *B+* with _flexpfb_ using the mentioned steps, but not on *B rev 2*. Although, I am using the same Raspbian Jessie image on both. 

Is this some other random problem, or am I missing something?

1ucian, 2015-12-20:
I have a PiZero with this screen and I have attempted using the kernel. The image shows up on the screen but I'm unable to use any of my USB devices. Can you guys help me figure this out?
Thanks

gibo65, 2015-12-30:
Hi 1ucian,
try to see this https://github.com/notro/fbtft/issues/254

gibo65, 2015-12-30:
...too late! ;-)

1ucian, 2015-12-30:
Haha thanks, I was able to get it running. The only thing I'm trying to figure out now is finding out how to add the optimization to my kernel to make the screen respond a bit faster.

gibo65, 2016-01-01:
...here is the kernel for the mcufriend 3.95 tft screen .,
Should work for the raspberry pi b and raspberry pi B+
Also enable spi in the raspi-config

Rename your kernel.img and add the kernel.img from the zip file,if or when you update your pi, you should put back the orignal kernel file and after updating put back the 3.95 tft kernel file.
Let me know if you discover any info.Thanks

susanaemp, 2016-01-15:
Hola chicoos, tengo un problema... y es que no se como hacer funcionar la misma pantalla de la que hablais (ili9488) en la Raspberry Pi2...
Osea me descargo el win32_3.95screem.zip de la pagina ( http://www.ittgroup.ee/et/ekraanid-ja-naidikud/626-lcd-tft-ekraan-395-480x320-raspberry-pi-le.html ) y escribo la imagen el una tarjeta micro SD virgen (sin nada escrito) y luego la meto en la raspberryPi2 y conecto la pantalla por los GPIO, despues conecto todo a la fuente de alimentacion y la pantalla se queda todo el rato en blanco...
cuando pasa eso intento conectar la raspberry a la pantalla de un ordenador mediante el cable HDMI pero en la pantalla del ordenador no aparece nada.....

ayudaa que puedo hacer?? o como lo hago??
he mirado en esta pagina y no entiendo cômo lo hace o que hace... ( http://appdictive.dk/blog/projects/2015/10/30/cheap_tft_display_on_raspberry_pi/ )
no entiendo los procedimientos o si tiene que tener antes la imagen de raspbian instalada..
alguien me podria dar unas instrucciones que seguir paso a paso???
o un video de como se hace???
un saluditoo

gibo65, 2016-01-15:
Hola chica,
the image is writing for only PI "tarjeta" ;-) please try to test on this card and AFTER move at PI2 
I admit GPIO pinout compatibility for first 26 pins but you can writed an error on image SDcard!
PLEASE community admit UK "idioma" so please write so!
;-)
Asta luego

susanaemp, 2016-01-15:
Jajajaja excuse me i was thinking in my lenguaje...
Now im going to write in UK lenguaje :P

susanaemp, 2016-01-15:
Helloooo,? I have a problem ... and i dont konw who i can running the same screen from which you speak (ili9488) in the Pi2 Raspberry ...
I downloaded the win32_3.95screem.zip  of the page (http://www.ittgroup.ee/et/ekraanid-ja-naidikud/626-lcd-tft-ekraan-395-480x320-raspberry-pi-le.html ) and write the image virgin micro SD card (nothing written) and then get into the raspberryPi2 and connect the screen by GPIO, then connect everything to the power supply and all the while the screen goes white. ..
when that happens I try to connect the raspberry to a computer screen using the HDMI cable but the computer screen does not show anything .....

Help what can i do ?? 
I've looked in this page and i dont understand how it does  ... (http://appdictive.dk/blog/projects/2015/10/30/cheap_tft_display_on_raspberry_pi/)
I dont understand the procedures or if the SD card have to be installed before the image of raspbian ..
someone could give me some instructions to follow step by step ???
or a video of how to do ???

iyayan, 2016-01-18:
I have same case, until now I have Blank white Screen with follow instruction from: http://www.ittgroup.ee/et/ekraanid-ja-naidikud/626-lcd-tft-ekraan-395-480x320-raspberry-pi-le.html

gicuuu3, 2016-04-20:
Hello. I'm hoping someone has found a solution for this. I've compiled the kernel with the fbtft from here, the one that was supposed to help with the lag, but it still didn't solve the issue. I can still see scanlines every time an image loads onto the screen or they change. Is there anyway to solve that problem?

thelinkin3000, 2016-05-02:
@gicuuu3
With the kernel you built, is there any noticeable difference between that and the one without the optimizations? Even if it doesn't render at usable speeds, do you notice an increase in the speed of the display?
I'm looking forward to do this as it'll make my display usable, but in order to compile a Kernel with the module from github I have to read a lot, as I never cross compiled a kernel.

gicuuu3, 2016-05-03:
@thelinkin3000 
It wasn't a major optimization. If you were to read the comment referring to the optimization(the one in the code itself, referred by the link mentioned above), it says that it's gonna give a 40 to 50% boost in speed. Can't say it was noticeable. 
I used this reference to cross compile it(in case you might find it useful): https://www.raspberrypi.org/documentation/linux/kernel/building.md

PaddleStroke, 2016-06-17:
Hello everyone,
I am trying to use this screen and I'm having some troubles.
Using raspberry pi zero. I am willing to use it for gameboy-like device with retropie.

I tried different things : 
- retropie rpi zero latest image v3.8.1 + changing the kernel, screen stays white. The pi LED blinks only once then stop.
- win32_3.95screen image : the screen stays white and the pi LED does not turn on at all.

Anyone made it work well on rpi zero could explain me how to do it ? :)

Sorry I don't know much about LCD drivers/linux kernel etc. @gibo65 could you explain me how to use what you posted for @1ucian ?

PaddleStroke, 2016-06-17:
It would be great if someone can give me the big picture on how this is working : linux-kernel kernel-drivers/rpi gpio/screen spec. 
And how to modify.
It seems that I lack some basic knowledge to understand the different solutions discussed here. Is there somewhere to start? :)

xxxSon1cxxx, 2016-09-13:
Has anyone tried to compile the kernel with this optimization https://github.com/notro/fbtft/blob/master/fbtft-io.c#L133 on Raspberry Pi 2? I used this guide https://github.com/notro/fbtft/wiki/Build-kernel-and-fbtft-drivers but no luck, still laggy((

cu6apum, 2017-04-11:
Hi there.
Thanks for your efforts!

I have a 3.5" ILI9488 module from buydisplays and need to get it working via native SPI. As of now I tried ili9481 and 9486 modules by notro (thank you Sir) and custom init strings from this topic. All I achieved for the moment is the screen (9481 driver) or half the screen (9486 driver) changing from white to gray on modprobe. Nothing more happens whatever I do.

Is there a reason to try new DBI branch? If so, is there anything "for dummies" to start from? I'm new to low-level hardware under linux.

Thank you!!

notro, 2017-04-12:
If wiring and the init string is corrrect it should turn black. Many drivers set register 0x36 in the set_var() function and can cause trouble if it doesn't match your panel. This register controls flip/mirror.
The flexfb driver doesn't set this register, so you can use it with the default setaddrwin=0 setting.
https://github.com/notro/fbtft/wiki/flexfb

cu6apum, 2017-04-12:
Notro, thanks a lot. It does not turn black in any case. This occurred once, when I occassionally told buswidth=8 instead of 9 used, and I could not repeat this. Even when it has, the module did not react to filling it from /dev/urandom or fbi viewer, so it's not initted yet. I keep trying.
Can the init string be different for parallel and SPI modes?!

notro, 2017-04-12:
> Can the init string be different for parallel and SPI modes?!

Not that I know of. The interface type is usally set with the IM[0:3] pins.

cu6apum, 2017-04-12:
I tried both 3-wire/9bit and 4-wire/8bit. The fbtft_device requires dc gpio in both modes. Gray screen and no response in any mode. Bum.
Will try with arduino first, there's too much unknown variables as of now. The init string is the first, as avr/stm forums say very different things from this thread.
Am I right that '-1' stands for command, and the rest is data? Sorry for being that dumb.

notro, 2017-04-12:
> Am I right that '-1' stands for command, and the rest is data?

Correct.

I you use the verbose=3 argument, you'll see the registers getting written to in the kernel log.

cu6apum, 2017-04-12:
I use debug=7 and dmesg.
Unfortunately, init from avr forums is leaving the display white. Heck, how do i debug this!..

cu6apum, 2017-04-12:
ok, let me know if you want to get rid of my posts, SORRY.

Is this the right seq?
`[  387.842762] flexfb spi0.0: fbtft_request_gpios: 'dc' = GPIO1
[  387.842776] flexfb spi0.0: flexfb_verify_gpios_dc()
[  387.842791] flexfb spi0.0: fbtft_init_display()
[  387.842808] flexfb spi0.0: init: write(0xB0) 0x00 
[  387.842829] flexfb spi0.0: fbtft_write_reg8_bus8: b0 00
[  387.842847] flexfb spi0.0: fbtft_write_spi(len=1): b0
[  387.843023] flexfb spi0.0: fbtft_write_spi(len=1): 00
[  387.843092] flexfb spi0.0: init: write(0x11) 
[  387.843106] flexfb spi0.0: fbtft_write_reg8_bus8: 11
[  387.843116] flexfb spi0.0: fbtft_write_spi(len=1): 11
[  387.843221] flexfb spi0.0: init: mdelay(120)
[  387.963260] flexfb spi0.0: init: write(0x3A) 0x55 
[  387.963282] flexfb spi0.0: fbtft_write_reg8_bus8: 3a 55
[  387.963294] flexfb spi0.0: fbtft_write_spi(len=1): 3a
[  387.963379] flexfb spi0.0: fbtft_write_spi(len=1): 55
[  387.963441] flexfb spi0.0: init: write(0xC2) 0x33 
[  387.963451] flexfb spi0.0: fbtft_write_reg8_bus8: c2 33
[  387.963461] flexfb spi0.0: fbtft_write_spi(len=1): c2
[  387.963521] flexfb spi0.0: fbtft_write_spi(len=1): 33
[  387.963582] flexfb spi0.0: init: write(0xC5) 0x00 0x1E 0x80 
[  387.963593] flexfb spi0.0: fbtft_write_reg8_bus8: c5 00 1e 80
[  387.963603] flexfb spi0.0: fbtft_write_spi(len=1): c5
[  387.963655] flexfb spi0.0: fbtft_write_spi(len=3): 00 1e 80
[  387.963708] flexfb spi0.0: init: write(0x36) 0x28 
[  387.963717] flexfb spi0.0: fbtft_write_reg8_bus8: 36 28
[  387.963727] flexfb spi0.0: fbtft_write_spi(len=1): 36
[  387.963779] flexfb spi0.0: fbtft_write_spi(len=1): 28
[  387.963830] flexfb spi0.0: init: write(0xB1) 0xB0 
[  387.963840] flexfb spi0.0: fbtft_write_reg8_bus8: b1 b0
[  387.963850] flexfb spi0.0: fbtft_write_spi(len=1): b1
[  387.963901] flexfb spi0.0: fbtft_write_spi(len=1): b0
[  387.963964] flexfb spi0.0: init: write(0xE0) 0x00 0x04 0x0E 0x08 0x17 0x0A 0x40 0x79 0x4D 0x07 0x0E 0x0A 0x1A 0x1D 0x0F 
[  387.963979] flexfb spi0.0: fbtft_write_reg8_bus8: e0 00 04 0e 08 17 0a 40 79 4d 07 0e 0a 1a 1d 0f
[  387.963991] flexfb spi0.0: fbtft_write_spi(len=1): e0
[  387.964043] flexfb spi0.0: fbtft_write_spi(len=15): 00 04 0e 08 17 0a 40 79 4d 07 0e 0a 1a 1d 0f
[  387.964115] flexfb spi0.0: init: write(0xE1) 0x00 0x1B 0x1F 0x02 0x10 0x05 0x32 0x34 0x43 0x02 0x0A 0x09 0x33 0x37 0x0F 
[  387.964129] flexfb spi0.0: fbtft_write_reg8_bus8: e1 00 1b 1f 02 10 05 32 34 43 02 0a 09 33 37 0f
[  387.964141] flexfb spi0.0: fbtft_write_spi(len=1): e1
[  387.964194] flexfb spi0.0: fbtft_write_spi(len=15): 00 1b 1f 02 10 05 32 34 43 02 0a 09 33 37 0f
[  387.964254] flexfb spi0.0: init: write(0x11) 
[  387.964264] flexfb spi0.0: fbtft_write_reg8_bus8: 11
[  387.964273] flexfb spi0.0: fbtft_write_spi(len=1): 11
[  387.964323] flexfb spi0.0: init: write(0x29) 
[  387.964333] flexfb spi0.0: fbtft_write_reg8_bus8: 29
[  387.964343] flexfb spi0.0: fbtft_write_spi(len=1): 29
[  387.964397] flexfb spi0.0: fbtft_update_display(start_line=0, end_line=319)
[  387.964410] flexfb spi0.0: fbtft_set_addr_win(xs=0, ys=0, xe=479, ye=319)
[  387.964420] flexfb spi0.0: fbtft_write_reg8_bus8: 2a 00 00 01 df
[  387.964431] flexfb spi0.0: fbtft_write_spi(len=1): 2a
[  387.964483] flexfb spi0.0: fbtft_write_spi(len=4): 00 00 01 df
[  387.964536] flexfb spi0.0: fbtft_write_reg8_bus8: 2b 00 00 01 3f
[  387.964547] flexfb spi0.0: fbtft_write_spi(len=1): 2b
[  387.964598] flexfb spi0.0: fbtft_write_spi(len=4): 00 00 01 3f
[  387.964650] flexfb spi0.0: fbtft_write_reg8_bus8: 2c
[  387.964660] flexfb spi0.0: fbtft_write_spi(len=1): 2c
[  387.964712] flexfb spi0.0: fbtft_write_vmem16_bus8(offset=0, len=307200)
[  387.964741] flexfb spi0.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
`

cu6apum, 2017-04-12:
YESSS.
It's been wrong init my friends.
The proper for ILI9488 is:
-1,0xE0,0x00,0x03,0x09,0x08,0x16,0x0A,0x3F,0x78,0x4C,0x09,0x0A,0x08,0x16,0x1A,0x0F,-1,0xE1,0x00,0x16,0x19,0x03,0x0F,0x05,0x32,0x45,0x46,0x04,0x0E,0x0D,0x35,0x37,0x0F,-1,0xC0,0x17,0x15,-1,0xC1,0x41,-1,0xC5,0x00,0x12,0x80,-1,0x36,0x48,-1,0x3A,0x66,-1,0xB0,0x80,-1,0xB1,0xA0,-1,0xB4,0x02,-1,0xE9,0x00,-1,0xF7,0xA9,0x51,0x2C,0x82,-1,0x11,-2,150,-1,0x29,-3

Thank you for patience.

elageneg, 2017-04-26:
Hi Cu6apum, 
I recently bought the ER-TFT035-6 from buy display and I am trying to make it work with SPI interface in the raspberry pi 3. Looks like you solved the problem, however I tried your init sequence and still having trouble. Can you please specify if you finally used 3-wire or 4-wire SPI interface. Details would be really appreciated.
Thanks a lot!

elageneg, 2017-05-08:
Hi Cu6apum and Notro, any comment about my last question? I would really appreciate any help...
All the best

cu6apum, 2017-05-08:
Gentlemen, sorry, was AFK for a week.
I've connected it in 8 bit mode with separate D/C gpio. Init cited above. It's not final, the display is using wrong color scheme - the bitmap rendered by fsb is wrong, I achieved only console prompt for now, in portrait orientation. My goal was to make it work via SPI, and it does. Details later, as I'm busy with PCB design now.
NanoPi Neo (alwinner H3) here, dietpi linux OOB.

rogueturnip, 2017-11-02:
I was wondering if anyone has had success with the ILI9488 SPI (4wire)?  I've got a driver working but the color is all messed up.  My suspicion is it's due to the 18bit mode for 4wire spi.  When I set 18bpp in the driver it doesn't display anything.  The only thing I can think of trying is putting the buswidth to 9bits (although the spec sheet says 4 wire is 8).

Anyone have any ideas?  I've considered the need to pad the 16bits before it's sent to the ILI9488 but I'm not sure where I'd do this.  I started the driver off the ILI9486 one.

sergey-suloev, 2018-03-10:
@elageneg any success with that display yet ?

sergey-suloev, 2018-03-11:
@elageneg I was able to make it work with mainline kernel

[5 more replies]

closed 2021-08-24 by github-actions[bot]