notro/fbtft #313: Driver support for Adafruit 1.5" SPI OLED?

Rasmushauschild, 2015-08-23:
Hi, I'm working on a Raspberry Pi project featuring this small OLED ( https://www.adafruit.com/product/1431 ).
I can't seem to find any driver for this display for the Raspberry pi, but when I asked on the Raspberry Pi forum I was told, that your driver would have support for my screen. Is this correct? If so then what device name should I use in the setup of the driver? Otherwise could you add support for this OLED? Or could I add it myself?

Thanks in advance, Rasmus

notro, 2015-08-23:
That display is not supported directly.

Add to /boot/config.txt:
```
dtparam=spi=on
```
Try this:
```
sudo modprobe fbtft_device name=pioled
```
See here for the default values (gpio): http://lxr.free-electrons.com/source/drivers/staging/fbtft/fbtft_device.c#L764

https://github.com/notro/fbtft/wiki/fbtft_device#usage

Rasmushauschild, 2015-08-27:
Hi Notro, thank you so much for your quick reply! I'm not exactly sure about how to connect the Oled to the PI's GPIO.
Is there a specific way the display should be connected to the GPIO of the Pi? If so, could you please tell me how?
Here's the connections available on the oled:
CD
MOSI
MISO
SDCS
OLEDCS
RESET
DC
SCK
I hope you can give me some advice, since i'm completely new to this whole screen driver thing :)
Thanks in advance, Rasmus

Rasmushauschild, 2015-08-30:
Hi Notro, when I do as you say, and the oled is connected i get this error message:
FATAL: module pioled not found.
Do you know what I should do?
Best regards Rasmus

notro, 2015-09-01:
pioled is not the module, it's the value of the name parameter

Rasmushauschild, 2015-09-02:
What then is the module called?
Also I haven't heard about parameters, could you please explain it for me?
Thank you so much for your help!

notro, 2015-09-02:
As I wrote previously:
```
modprobe fbtft_device name=pioled
```
fbtft_device is the module, name is the module parameter.

Rasmushauschild, 2015-09-02:
I follow circuit basics tutorial ( http://www.circuitbasics.com/setup-lcd-touchscreen-raspberry-pi/#comment-706 ) and in step 4 it tells you to copy and paste a block of text from the tutorial itself into /etc/modules. In that block of text it says waveshare32b multiple times.The tutorial tells you to replace that with the name of your own screen. What name should I put there? Or shouldn't I follow that tutorial?
Thanks again

notro, 2015-09-02:
The following steps are the only ones necessary to test the display on a plain Raspian installation:

Add this to /boot/config.txt to enable the SPI controller driver:
```
dtparam=spi=on
```
Reboot

Load the pioled fbtft device (Linux will autoload the necessary driver):
```
$ sudo modprobe fbtft_device name=pioled

# You will get a 'graphics fb1: ...' message in the kernel log
$ dmesg | tail

# The framebuffer is present
$ ls -l /dev/fb1

# Show modules
$ lsmod
fb_ssd1351
fbtft
fbtft_device
...
```

Redirect the console from hdmi to this new framebuffer:
```
$ con2fbmap 1
console 1 is mapped to framebuffer 0
$ con2fbmap 1 1
$ con2fbmap 1
console 1 is mapped to framebuffer 1
```
If you don't get anything on the display, either wiring or the init sequence is wrong.

Rasmushauschild, 2015-09-02:
If I would display on an instalation on retropie is the process the same then?

notro, 2015-09-03:
> If I would do it on an instalation on retropie is the process the same then?

Yes.

```
MOSI -> MOSI
MISO - not needed
SCK -> SCLK
OLECS -> CE0
DC -> GPIO25
RESET -> GPIO24
```

Rasmushauschild, 2015-09-03:
Thank you so much, I'll try that tomorrow :)

Rasmushauschild, 2015-09-04:
It didn't work.
If I have a completely fresh install of raspbian do I then just Follow your steps above, or are there more? Could you leave me with just a link, or a short guide?
Thank you in advance

notro, 2015-09-04:
I guess Adafruit has an Arduino library for the display. Check the init sequence in that library against http://lxr.free-electrons.com/source/drivers/staging/fbtft/fb_ssd1351.c#L26

Similar display https://www.raspberrypi.org/forums/viewtopic.php?f=41&t=78636

You could also port the arduino library to the pi and see if you can get it working from userspace. It did that once with a display it couldn't get working with fbtft. Or see if someone already has done that.

Rasmushauschild, 2015-09-04:
The similar oled you linked to is almost identical to mine! :)
If I knew how to, I could probably get it working, but again I don't know any guide or tutorial. :(
Do you know any?
Thanks in advance

notro, 2015-09-04:
I don't know what kind of guide you're looking for, but I would have run these 3 commands and see if I got a console login on the display:
```
$ sudo modprobe fbtft_device custom name=fb_ssd1351 buswidth=8 gpios=reset:24,dc:25 speed=10000000 width=128 height=96 bgr=1 init=-1,0xFD,0x12,-1,0xFD,0xB1,-1,0xAE,-1,0xB3,0xF1,-1,0xCA,127,-1,0xA0,0x74,-1,0x15,0x00,0x7F,-1,0x75,0x00,0x7F,-1,0xA1,96,-1,0xA2,0x0,-1,0xB5,0x00,-1,0xAB,0x01,-1,0xB1,0x32,-1,0xBE,0x05,-1,0xA6,-1,0xC1,0xC8,0x80,0xC8,-1,0xC7,0x0F,-1,0xB4,0xA0,0xB5,0x55,-1,0xB6,0x01,-1,0xAF,-3

# verify that the driver is loaded
$ dmesg | grep graphics
... graphics fb1: fb_ssd1351 ...

$ con2fbmap 1 1
```

Rasmushauschild, 2015-09-04:
When I do so nothing happens on the oled, but in the terminal/cmd prompt it does give some feedback, as shown in the attached image:
[image]

notro, 2015-09-04:
I have run out of things to suggest.

Rasmushauschild, 2015-09-04:
unless I am mistaken I have to do more than just typing that one line, right?

Rasmushauschild, 2015-09-04:
By the way, thank you so much for your enormous help! I really appreciate it! :)

Rasmushauschild, 2015-09-04:
Now it's functioning perfectly in the console! :)
However when I type startx nothing happens on the oled.
Also every time I reboot I have to type all three commands, that you posted above, to make the Pi display to the oled.
How can I make the Pi display to the oled automatically, every time it boots up?
Again, thank you so much for all your help! :)

notro, 2015-09-04:
Remove ```sudo modprobe``` from the fbtft_device line and put the rest in /etc/modules. Now it will load on boot.
X windows: Change to /dev/fb1 in /usr/share/X11/xorg.conf.d/99-fbturbo.conf

Rasmushauschild, 2015-09-04:
When I put the fbtft_device line in /etc/modules and then reboot it doesn't display on the oled automatically. First when i type con2fbmap 1 1.
Can I fix this?
The other part with the X windows works fine! :)

notro, 2015-09-04:
https://github.com/notro/fbtft/wiki/Boot-console#enable-console

Rasmushauschild, 2015-09-05:
I'm not exactly sure where or what I should type, to make the oled load automatically.
Sorry for being such a newbie. :)

notro, 2015-09-05:
Compare the init sequence from the library with the one you use. I see that one setting depends on height.
https://github.com/adafruit/Adafruit-SSD1351-library/blob/master/Adafruit_SSD1351.cpp#L376

Rasmushauschild, 2015-09-05:
Everything works now!
Only thing is that the screen is flickering, and making a weak buzzing sound when it displays the rasbian desktop.
Do you know why?

notro, 2015-09-05:
I remember someone having a buzzing sound on another oled and he adjusted some register setting.
Have you verified that you are using the same init as the library?

Rasmushauschild, 2015-09-05:
I haven't verified it, but it does display the console properly.

Rasmushauschild, 2015-09-05:
Also if I try to insert the fbtft_device line on a retropie image it says: ERROR: could not insert 'fbtft_device': Invalid argument
Should I do anything different when I'm working with retropie?

notro, 2015-09-05:
I have never used retropie, so I don't know which kernel they are using.
What does ```cat /proc/version``` show?

Rasmushauschild, 2015-09-05:
It shows:
Linux version 3.18.11+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #781 PREEMPT Tue Apr 21 18:02:18 BST 2015

notro, 2015-09-05:
That's a standard Pi kernel and it should have the fbtft drivers.
Try ```modinfo fbtft_device```
If that fails, do you have the folder ```/lib/modules/3.18.11+/kernel/drivers/staging/fbtft```?

Rasmushauschild, 2015-09-05:
When I type modinfo fbtft_device it says:
filename:       /lib/modules/3.18.11+/kernel/drivers/staging/fbtft/fbtft_device.ko
license:        GPL
author:         Noralf Tronnes
description:    Add a FBTFT device.
srcversion:     42360D1CD0E72DEE0838538
depends:        fbtft
staging:        Y
intree:         Y
vermagic:       3.18.11+ preempt mod_unload modversions ARMv6 
parm:           name:Devicename (required). name=list => list all supported devices. (charp)
parm:           rotate:Angle to rotate display counter clockwise: 0, 90, 180, 270 (uint)
parm:           busnum:SPI bus number (default=0) (uint)
parm:           cs:SPI chip select (default=0) (uint)
parm:           speed:SPI speed (override device default) (uint)
parm:           mode:SPI mode (override device default) (int)
parm:           gpios:List of gpios. Comma separated with the form: reset:23,dc:24 (when overriding the default, all gpios must be specified) (charp)
parm:           fps:Frames per second (override driver default) (uint)
parm:           gamma:String representation of Gamma Curve(s). Driver specific. (charp)
parm:           txbuflen:txbuflen (override driver default) (int)
parm:           bgr:BGR bit (supported by some drivers). (int)
parm:           startbyte:Sets the Start byte used by some SPI displays. (uint)
parm:           custom:Add a custom display device. Use speed= argument to make it a SPI device, else platform_device (bool)
parm:           width:Display width, used with the custom argument (uint)
parm:           height:Display height, used with the custom argument (uint)
parm:           buswidth:Display bus width, used with the custom argument (uint)
parm:           init:Init sequence, used with the custom argument (array of int)
parm:           debug:level: 0-7 (the remaining 29 bits is for advanced usage) (ulong)
parm:           verbose:0 silent, >0 show gpios, >1 show devices, >2 show devices before (default=3) (uint)

Yes, I do have that folder.
Inside are following files:
fb_agm1264k-fl.ko  fb_ili9320.ko  fb_ili9486.ko  fb_ssd1289.ko  fbtft_device.ko  fb_upd161704.ko
fb_bd663474.ko     fb_ili9325.ko  fb_pcd8544.ko  fb_ssd1306.ko  fbtft.ko         fb_watterott.ko
fb_hx8340bn.ko     fb_ili9340.ko  fb_ra8875.ko   fb_ssd1331.ko  fb_tinylcd.ko    flexfb.ko
fb_hx8347d.ko      fb_ili9341.ko  fb_s6d02a1.ko  fb_ssd1351.ko  fb_tls8204.ko
fb_hx8353d.ko      fb_ili9481.ko  fb_s6d1121.ko  fb_st7735r.ko  fb_uc1701.ko

notro, 2015-09-05:
Ok, try loading fbtft_device manually: sudo modprobe fbtft_device ...
Maybe the kernel log has anything more about the error: dmesg

Rasmushauschild, 2015-09-05:
When I try to load it manually it still says:
ERROR: could not insert 'fbtft_device': Invalid argument
When I then type dmesg it says a lot, and down in the bottom it does say failed to register SPI device:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.11+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #781 PREEMPT Tue Apr 21 18:02:18 BST 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model B Plus
[    0.000000] cma: Reserved 8 MiB at 0x0f800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat c083d364, node_mem_map cf5f4000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65024
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708.boardrev=0x10 bcm2708.serial=0xfd4e46dc smsc95xx.macaddr=B8:27:EB:4E:46:DC bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 242404K/262144K available (5881K kernel code, 348K rwdata, 1868K rodata, 336K init, 733K bss, 19740K reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xd0800000 - 0xff000000   ( 744 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc07997a0   (7750 kB)
[    0.000000]       .init : 0xc079a000 - 0xc07ee000   ( 336 kB)
[    0.000000]       .data : 0xc07ee000 - 0xc084513c   ( 349 kB)
[    0.000000]        .bss : 0xc084513c - 0xc08fc8a8   ( 734 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:522
[    0.000025] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483648000ns
[    0.000074] Switching to timer-based delay loop, resolution 1000ns
[    0.000358] Console: colour dummy device 80x30
[    0.001449] console [tty1] enabled
[    0.001495] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.001570] pid_max: default: 32768 minimum: 301
[    0.001942] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002009] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.003014] Initializing cgroup subsys memory
[    0.003106] Initializing cgroup subsys devices
[    0.003165] Initializing cgroup subsys freezer
[    0.003219] Initializing cgroup subsys net_cls
[    0.003270] Initializing cgroup subsys blkio
[    0.003388] CPU: Testing write buffer coherency: ok
[    0.003501] ftrace: allocating 19232 entries in 57 pages
[    0.108314] Setting up static identity map for 0x553398 - 0x5533d0
[    0.111174] devtmpfs: initialized
[    0.128710] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.131752] pinctrl core: initialized pinctrl subsystem
[    0.134444] NET: Registered protocol family 16
[    0.139984] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.141462] bcm2708.uart_clock = 3000000
[    0.144403] No ATAGs?
[    0.144467] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.144529] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.144594] mailbox: Broadcom VideoCore Mailbox driver
[    0.144756] bcm2708_vcio: mailbox at f200b880
[    0.145215] bcm_power: Broadcom power driver
[    0.145270] bcm_power_open() -> 0
[    0.145301] bcm_power_request(0, 8)
[    0.646018] bcm_mailbox_read -> 00000080, 0
[    0.646069] bcm_power_request -> 0
[... 144 lines cut ...]
[   13.382503] fbtft_device:  'fb' Platform devices registered:
[   13.395353] fbtft_device:      bcm2708_fb id=-1 pdata? no
[   13.407726] fbtft_device:  spi_busnum_to_master(0) returned NULL
[   13.421496] fbtft_device: failed to register SPI device
[   22.922418] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   24.511008] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   32.810090] Adding 102396k swap on /var/swap.  Priority:-1 extents:1 across:102396k SSFS
[   33.512731] Bluetooth: Core ver 2.19
[   33.515460] NET: Registered protocol family 31
[... cut ...]
[   33.618487] Bluetooth: RFCOMM socket layer initialized
[   33.618547] Bluetooth: RFCOMM ver 1.11
[   62.268922] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[   62.279945] fbtft_device:  SPI devices registered:
[   62.280019] fbtft_device:  'fb' Platform devices registered:
[   62.280229] fbtft_device:      bcm2708_fb id=-1 pdata? no
[   62.280368] fbtft_device:  spi_busnum_to_master(0) returned NULL
[   62.280388] fbtft_device: failed to register SPI device

notro, 2015-09-05:
Oh, you haven't enabled spi: dtparam=spi=on

Rasmushauschild, 2015-09-05:
Oh, sorry for being so helpless :)
Now it does display the console, but it wont display the GUI.
I have corrected /dev/fb0 to /dev/fb1, so I don't know what it could be, do you?

notro, 2015-09-05:
It depends on what you mean with gui.
You have to use fbcp to sync/copy fb0 to fb1 for most emulators: https://www.raspberrypi.org/forums/viewtopic.php?p=622842#p622842

Rasmushauschild, 2015-09-05:
If GUI I means EmulationStation and emulators?

notro, 2015-09-05:
If they use opengl, you have to use fbcp.
https://github.com/notro/fbtft/wiki/Framebuffer-use#framebuffer-mirroring

Rasmushauschild, 2015-09-05:
Ok, do you know if the instalation process is identical or similar to fbtft?
Also will I be able to use the same fbtft_device line?

notro, 2015-09-05:
You use fbcp together with fbtft.
fbtft are kernel drivers that, on the Pi, creates a framebuffer /dev/fb1.
bcm2708_fb (hdmi) creates /dev/fb0, and has hw accel.

fbcp is an userspace application that takes snapshots of /dev/fb0 every 25ms and copies it to /dev/fb1.
So fb1 displays the same as fb0.

I haven't used fbcp myself, apart from testing it.

Rasmushauschild, 2015-09-05:
Ok, thank you so much for all the help, I really appreciate it!
Can I still get help from you with fbcp, or should I ask somewhere else?
Best regards Rasmus

notro, 2015-09-05:
It would be better to ask about that in the pi forums. Most likely someone there has already setup fbcp with retropie.

Rasmushauschild, 2015-09-05:
That's fine, thanks again :)

closed 2015-11-23 by notro

cosmikwolf, 2015-12-17:
do you guys have any more info about buzzing with the SSD1351? I am experiencing that with an Arduino and I was hoping you guys had some more breadcrumbs I could follow!