notro/fbtft #316: HY28b non standard pendown GPIO

cburger, 2015-09-03:
I am trying to change the hy28b-overlay.dts so I can use GPIO23 instead of GPIO17 for the pen down.
I have modified the source file as follows.

```
brcm,pins = <17 25 18>; --> brcm,pins = <23 25 18>;
interrupts = <17 2>; --> interrupts = <23 2>;
pendown-gpio = <&gpio 17 0>; --> pendown-gpio = <&gpio 23 0>;
```

And then update.

```
sudo dtc -@ -I dts -O dtb -o /boot/overlays/hy28b-overlay.dts hy28b-overlay_modified.dts
```
then restart, but I can still not get the touch screen to work. I have tested my GPIO on Python and it works fine there when touching the screen, so its not a hardware issue.

What am I doing wrong?

notro, 2015-09-04:
It looks correct to me. And it's working with the default gpio17?
What does ```cat /proc/interrupts``` show?

cburger, 2015-09-05:
I have this.

```
sudo cat /proc/interrupts
           CPU0       
  3:      36862   ARMCTRL   3  BCM2708 Timer Tick
 16:          0   ARMCTRL  16  bcm2708_fb dma
 24:        117   ARMCTRL  24  DMA IRQ
 25:       1500   ARMCTRL  25  DMA IRQ
 32:     411131   ARMCTRL  32  dwc_otg, dwc_otg_pcd, dwc_otg_hcd:usb1
 49:          0   ARMCTRL  49  20200000.gpio:bank0
 50:          0   ARMCTRL  50  20200000.gpio:bank1
 65:         10   ARMCTRL  65  2000b880.mailbox
 66:          2   ARMCTRL  66  VCHIQ doorbell
 75:          1   ARMCTRL  75
 77:      17346   ARMCTRL  77  20204000.spi
 79:        767   ARMCTRL  79  20804000.i2c
 80:      37428   ARMCTRL  80  20204000.spi
 84:       6577   ARMCTRL  84  mmc0
411:          0  pinctrl-bcm2835  17  ads7846
FIQ:              usb_fiq
Err:          0
```

notro, 2015-09-05:
The touch driver is using gpio17 for interrupt:
```
411:          0  pinctrl-bcm2835  17  ads7846
```
So your changes haven't taken effect.

cburger, 2015-09-05:
So for some reason my GPIO pin change does not have an effect. Where else can this be configured from?

closed 2015-11-23 by notro