notro/fbtft #317: Adafruit 2.4" tft on Beaglebone Black white flickering screen

FelixNemis, 2015-09-08:
I'm trying to get an adafruit tft (this one https://www.adafruit.com/products/2478) to work on my beagle bone black.

I've followed this guide mostly: https://github.com/notro/fbtft/wiki/BeagleBone-Black

I'm using the ili9341 driver and the output from dmesg looks good but the screen stays white and starts to flicker periodically, I can change the backlight brightness. And then after awhile the backlight turns off (it will wake up if I plug in a keyboard)

Do I need to change something in the initialization sequence or something?

Here's the relevant dmesg output:
```
[  357.658954] bone-capemgr bone_capemgr.9: part_number 'ADAFRUIT-ILI9341', version 'N/A'
[  357.659086] bone-capemgr bone_capemgr.9: slot #8: generic override
[  357.659104] bone-capemgr bone_capemgr.9: bone: Using override eeprom data at slot 8
[  357.659120] bone-capemgr bone_capemgr.9: slot #8: 'Override Board Name,00A0,Override Manuf,ADAFRUIT-ILI9341'
[  357.659211] bone-capemgr bone_capemgr.9: slot #8: Requesting part number/version based 'ADAFRUIT-ILI9341-00A0.dtbo
[  357.659227] bone-capemgr bone_capemgr.9: slot #8: Requesting firmware 'ADAFRUIT-ILI9341-00A0.dtbo' for board-name 'Override Board Name', version '00A0'
[  357.665121] bone-capemgr bone_capemgr.9: slot #8: dtbo 'ADAFRUIT-ILI9341-00A0.dtbo' loaded; converting to live tree
[  357.665571] bone-capemgr bone_capemgr.9: slot #8: #5 overlays
[  357.713569] edma-dma-engine edma-dma-engine.0: allocated channel for 0:43
[  357.713628] edma-dma-engine edma-dma-engine.0: allocated channel for 0:42
[  357.713840] fbtft_of_value: regwidth = 8
[  357.713851] fbtft_of_value: buswidth = 8
[  357.713861] fbtft_of_value: debug = 3
[  357.713869] fbtft_of_value: rotate = 270
[  357.713877] fbtft_of_value: fps = 50
[  357.714076] fb_ili9341 spi1.0: fbtft_gamma_parse_str() str=
[  357.714088] fb_ili9341 spi1.0: 1F 1A 18 0A 0F 06 45 87 32 0A 07 02 07 05 00
[  357.714088] 00 25 27 05 10 09 3A 78 4D 05 18 0D 38 3A 1F
[  357.714141] of_get_named_gpio_flags exited with status 48
[  357.714179] fb_ili9341 spi1.0: fbtft_request_one_gpio: 'reset-gpios' = GPIO48
[  357.714192] of_get_named_gpio_flags exited with status 60
[  357.714207] fb_ili9341 spi1.0: fbtft_request_one_gpio: 'dc-gpios' = GPIO60
[  357.714240] fb_ili9341 spi1.0: fbtft_verify_gpios()
[  357.714250] fb_ili9341 spi1.0: init_display()
[  357.714259] fb_ili9341 spi1.0: fbtft_reset()
[  357.973294] fb_ili9341 spi1.0: set_var()
[  358.055067] fb_ili9341 spi1.0: Display update: 1836 kB/s (81.651 ms), fps=0 (0.000 ms)
[  358.055085] fb_ili9341 spi1.0: set_gamma()
[  358.060621] Console: switching to colour frame buffer device 80x40
[  358.063076] graphics fb0: fb_ili9341 frame buffer, 320x240, 150 KiB video memory, 16 KiB DMA buffer memory, fps=51, spi1.0 at 48 MHz
[  358.063745] fb_ili9341 0.lcd@0: fbtft_probe_common()
... (stuff about failing to add platform device. see guide I used) ...
[  358.088417] bone-capemgr bone_capemgr.9: slot #8: Applied #5 overlays.
```

notro, 2015-09-09:
The init sequence from the driver and the Arduino library that Adafruit provides differs:
http://lxr.free-electrons.com/source/drivers/staging/fbtft/fb_ili9341.c
https://github.com/adafruit/Adafruit_ILI9341/blob/master/Adafruit_ILI9341.cpp#L236

To get you started on the conversion:
```
  writecommand(0xEF);
  writedata(0x03);
  writedata(0x80);
  writedata(0x02);

  writecommand(0xCF);  
  writedata(0x00); 
  writedata(0XC1); 
  writedata(0X30); 
```
translates to:
```
init = <0x10000EF 0x03 0x80 0x02
        0x10000CF 0x00 0xC1 0x30
        ...
        >;
```
OR with 0x2000000 to make a delay.

Example: https://github.com/raspberrypi/linux/blob/rpi-4.1.y/arch/arm/boot/dts/overlays/mz61581-overlay.dts#L67

FelixNemis, 2015-09-09:
Thanks, I converted the sequence from the arduino code and put that into my .dts file
It looks a bit different during startup, but the display still doesn't go black

Here's my dts file:
```
/dts-v1/;
/plugin/;

/ {
	compatible = "ti,beaglebone", "ti,beaglebone-black";

	/* identification */
	part-number = "ADAFRUIT-ILI9341";
	version = "00A0";

	/* state the resources this cape uses */
	exclusive-use =
		/* the pin header uses */
		"P9.31",	/* spi1_sclk */
		"P9.29",	/* spi1_d0 MISO */
		"P9.30",	/* spi1_d1 MOSI */
		"P9.28",	/* spi1_cs0 */
		"P9.42",	/* spi1_cs1 */
		"P9.15",	/* gpio1_16 - reset */
		"P9.12",	/* gpio1_28 - dc */
		"P9.14",	/* backlight */
		"P9.25",	/* touch irq */
		/* the hardware ip uses */
		"gpio1_16",
		"gpio1_28",
		"gpio3_21",
		"ehrpwm1A",
		"spi1";

	fragment@0 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			bb_spi1_pins: pinmux_bb_spi1_pins {
				pinctrl-single,pins = <
					0x190 0x33	/* mcasp0_aclkx.spi1_sclk, INPUT_PULLUP | MODE3 */
					0x194 0x33	/* mcasp0_fsx.spi1_d0, INPUT_PULLUP | MODE3 */
					0x198 0x13	/* mcasp0_axr0.spi1_d1, OUTPUT_PULLUP | MODE3 */
					0x19c 0x13	/* mcasp0_ahclkr.spi1_cs0, OUTPUT_PULLUP | MODE3 */
					0x164 0x12	/* eCAP0_in_PWM0_out.spi1_cs1 OUTPUT_PULLUP | MODE2 */
				>;
			};

			lcd_ctrl_pinmux: lcd_ctrl_pins {
				pinctrl-single,pins = <
					0x040 0x17	/* P9.15 gpio1_16  OUTPUT_PULLUP | MODE7 */
					0x078 0x17	/* P9.12 gpio1_28  OUTPUT_PULLUP | MODE7 */
					0x1ac 0x37	/* P9.25 gpio3_21  INPUT_PULLUP  | MODE7 */
				>;
			};

			pwm_bl_pins: pinmux_pwm_bl_pins {
				pinctrl-single,pins = <
					0x48 0x06       /* ehrpwm1a, MODE6 */
				>;
			};
		};
	};

	fragment@1 {
		target = <&spi1>;
		__overlay__ {
			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&bb_spi1_pins>;

			#address-cells = <1>;
			#size-cells = <0>;

			lcd@0{
				compatible = "ilitek,ili9341";
				reg = <0>;

				pinctrl-names = "default";
				pinctrl-0 = <&lcd_ctrl_pinmux>;

				spi-max-frequency = <48000000>;
				rotate = <270>;
				bgr;
				fps = <50>;
				buswidth = <8>;
				regwidth = <8>;
				dc-gpios = <&gpio2 28 0>;
				reset-gpios = <&gpio2 16 0>;

				init = <0x10000EF 0x03 0x80 0x02
					0x20000CF 0x00 0xC1 0x30
					0x20000ED 0x64 0x03 0x12 0x81
					0x20000E8 0x85 0x00 0x78
					0x20000CB 0x39 0x2C 0x00 0x34 0x02
					0x20000F7 0x20
					0x20000EA 0x00 0x00
					0x20000C0 0x23
					0x20000C1 0x10
					0x20000C5 0x3E 0x28
					0x20000C7 0x86
					0x2000036 0x48
					0x200003A 0x55
					0x20000B1 0x00 0x18
					0x20000B6 0x08 0x82 0x27
					0x20000F2 0x00
					0x2000026 0x01
					0x20000E0 0x0F 0x31 0x2B 0x0C 0x0E 0x08 0x4E 0xF1 0x37 0x07 0x10 0x03 0x0E 0x09 0x00
					0x20000E1 0x00 0x0E 0x14 0x03 0x11 0x07 0x31 0xC1 0x48 0x08 0x0F 0x0C 0x31 0x36 0x0F
					0x2000011
					0x2000029>;

				debug = <3>;
			};
		};
	};

	fragment@2 {
		target = <&epwmss1>;
		__overlay__ {
			status = "okay";
		};
	};

	fragment@3 {
		target = <&ehrpwm1>;
		__overlay__ {
			status = "okay";
		};
	};

	fragment@4 {
		target = <&ocp>;
		__overlay__ {
			backlight {
				compatible = "pwm-backlight";
				pinctrl-names = "default";
				pinctrl-0 = <&pwm_bl_pins>;
				status = "okay";

				pwms = <&ehrpwm1 0 500000 0>;
				pwm-names = "backlight";
				brightness-levels = <0 4 8 16 32 64 128 255>;
				default-brightness-level = <7>; /* index to the array above */
			};
		};
	};
};
```

notro, 2015-09-10:
You are using 0x20000nn when you should be using 0x10000nn.
0x20000nn is used when specifying a delay.

FelixNemis, 2015-09-10:
Yeah I had them as 0x10000nn (except the last command to turn on the display) with the same result.
I changed them to that thinking it might be sending commands too quickly.

notro, 2015-09-10:
Ok, 0x20000nn isn't a way to slow down the command, it's a delay in it self. Often used between sleepout and displayon like in the Adafruit library:
```
  writecommand(ILI9341_SLPOUT);    //Exit Sleep 
  if (hwSPI) spi_end();
  delay(120); 		
  if (hwSPI) spi_begin();
  writecommand(ILI9341_DISPON);    //Display on 
  if (hwSPI) spi_end();
```
This translates to:
```
                    0x1000011
                    0x2000078 // 120 ms delay
                    0x1000029>;
```
Maybe the dispon command is never executed because it's sent to early.

FelixNemis, 2015-09-10:
Oh, so it goes on the command _before_ the delay, thanks I'll try this when I get home.

FelixNemis, 2015-09-10:
Alright, so it's still not working, this is my init sequence:
```init = <0x10000EF 0x03 0x80 0x02
        0x10000CF 0x00 0xC1 0x30
        0x10000ED 0x64 0x03 0x12 0x81   
        0x10000E8 0x85 0x00 0x78
        0x10000CB 0x39 0x2C 0x00 0x34 0x02
        0x10000F7 0x20
        0x10000EA 0x00 0x00
        0x10000C0 0x23
        0x10000C1 0x10
        0x10000C5 0x3E 0x28
        0x10000C7 0x86
        0x1000036 0x48
        0x100003A 0x55
        0x10000B1 0x00 0x18
        0x10000B6 0x08 0x82 0x27
        0x10000F2 0x00
        0x1000026 0x01
        0x10000E0 0x0F 0x31 0x2B 0x0C 0x0E 0x08 0x4E 0xF1 0x37 0x07 0x10 0x03 0x0E 0x09 0x00
        0x10000E1 0x00 0x0E 0x14 0x03 0x11 0x07 0x31 0xC1 0x48 0x08 0x0F 0x0C 0x31 0x36 0x0F
        0x1000011
        0x2000078
        0x1000029>;```
Is the 78 on the delay command important?
Thanks for all your help so far btw.

notro, 2015-09-11:
Yes, 0x78 is 120 which is the time most datasheets says to wait in milliseconds after sleepout.
In the beginning of fbtft there was a Adafruit display that I couldn't get to work with fbtft. So I hacked the Arduino library to work on the Raspbery Pi. I don't remember the details, but it helped me get it working with fbtft.

A search shows that Adafruit has a userspace version for this display: https://learn.adafruit.com/user-space-spi-tft-python-library-ili9341-2-8/overview
See if you can get that working.

closed 2015-12-04 by notro