notro/fbtft #318: cm-fx6 board - ili9341 issue with framebuffer

nancynguyen, 2015-09-11:
Hi notro,

I had encountered a problem of using fbtft device. I was able to enable spi driver for using fb_ili9341 in imx6 board. However when I try to display a rectangle in LCD using the buffer that fbtft_device registered for this LCD, it has "segmentation fault" error. 
I pushed output of framebuffer information and I found that there is no timing information of this fb. 
  activate:      0
  height:        0 (0x00000000)
  width:         0 (0x00000000)
  pixclock:      0
  left_margin:   0
  right_margin:  0
  upper_margin:  0
  lower_margin:  0
  hsync_len:     0
  vsync_len:     0
  sync:          0x00000000
  vmode:         0x00000000
  rotate:        270
Here are the output of dmesg:
root@devel:/home/ngocntt# dmesg | grep fb
fbtft_of_value: buswidth = 8
fbtft_of_value: debug = 0
fbtft_of_value: rotate = 270
fbtft_of_value: fps = 30
graphics fb6: fb_ili9341 frame buffer, 320x240, 150 KiB video memory, 16 KiB DMA buffer memory, fps=33, spi32765.3 at 32 MHz
fbtft_device:  SPI devices registered:
fbtft_device:      m25px16 spi32766.0 20000kHz 8 bits mode=0x00
fbtft_device:      tsc2046 spi32766.1 1500kHz 8 bits mode=0x00
fbtft_device:      ili9341 spi32765.3 32000kHz 8 bits mode=0x00

The fb_ili9341 is using /dev/fb6 as I know. However there is no timing infor for /dev/fb6
root@devel:/home/ngocntt# fbset -fb /dev/fb6 -i

mode "320x240"
    geometry 320 240 320 240 16
    timings 0 0 0 0 0 0 0
    nonstd 1
    rgba 5/11,6/5,5/0,0/0
endmode

Frame buffer device information:
    Name        : fb_ili9341
    Address     : 0
    Size        : 153600
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 0
    YPanStep    : 0
    YWrapStep   : 0
    LineLength  : 640
    Accelerator : No

I tried to set timing for /dev/fb6 manually by "fbset" without success. 
How could you solve this problem? Could you give any hint for that? 

Thanks & best regards,

notro, 2015-09-12:
There's no timing info because the display controller has onboard ram that is scanned out to the display.
FBTFT just syncs the Linux framebuffer with the display framebuffer (and initialises the controller).
When the driver is loaded, you should get a black (zeroed) display with backlight on.

Does these commands give you any output?
```
sudo cp /dev/urandom /dev/fb6
con2fbmap 1 6
```

nancynguyen, 2015-09-15:
Hi Notro,

Thank for your guide, however, I could not write anything to /dev/fb6.
As they said that "NO SPACE LEFT ON DEVICE".
And the result is not displaying anything on LCD. 

Thanks for any helps!!!

notro, 2015-09-15:
You will get debug output to the kernel log with: modprobe fbtft_device ... debug=7

closed 2015-12-04 by notro