notro/fbtft #319: fbtft_device loading by systemd fails whereas successfull with modeprobe

computer007, 2015-09-15:
Hello,
Now that I migrated to systemd I encounter an issue with autoloading my fbtft module at startup.
I followed the explanations given there: https://github.com/notro/fbtft/wiki#make-it-permanent-arch-linux
The only difference is the module name (= ssd1322).

Here is my configuration:
```
In /boot/config.txt:
# Other lines are all commented
dtparam=spi=on
```

In /etc/modules-load.d/modules.conf
```
snd-bcm2835
spi_bcm2708  <= just added it for test but think it is not necessary
fbtft_device
```

In /etc/modprobe.d/fbtft_options.conf
```
options fbtft_device name=ssd1322 debug=3
```

I get the following in dmesg after a reboot:
```
....
[    4.019997] systemd[1]: Started File System Check on Root Device.
[    4.057710] systemd[1]: Started udev Coldplug all Devices.
[    4.111212] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    4.131504] fbtft_device:  SPI devices registered:
[    4.138421] fbtft_device:  'fb' Platform devices registered:
[    4.146073] fbtft_device:      soc:fb id=-1 pdata? no
[    4.153870] fbtft_device:  spi_busnum_to_master(0) returned NULL
[    4.161895] fbtft_device: failed to register SPI device
[    4.206452] systemd[1]: Starting Create Static Device Nodes in /dev...
[    4.221440] systemd[1]: Starting Syslog Socket.
[    4.230910] systemd[1]: Listening on Syslog Socket.
[    4.238039] systemd[1]: Starting Journal Service...
[    4.253337] systemd[1]: Started Journal Service.
[    4.485636] systemd-udevd[110]: starting version 215
[    5.087098] spi spi0.0: setting up native-CS0 as GPIO 8
[    5.098428] spi spi0.1: setting up native-CS1 as GPIO 7
...
```

Now, if after reboot I long on with ssh and run ```sudo modprobe fbtft_device name=ssd1322 debug=3``` it works perfectly as indicated by dmesg:

```
[  614.226255] fbtft_device:  SPI devices registered:
[  614.226284] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[  614.226297] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[  614.226304] fbtft_device:  'fb' Platform devices registered:
[  614.226327] fbtft_device:      soc:fb id=-1 pdata? no
[  614.226372] fbtft_device: Deleting spi0.0
[  614.226723] spi spi0.0: setting up native-CS0 as GPIO 8
[  614.226848] fbtft_device:  GPIOS used by 'ssd1322':
[  614.226858] fbtft_device:    'reset' = GPIO24
[  614.226866] fbtft_device:    'dc' = GPIO25
[  614.226872] fbtft_device:  SPI devices registered:
[  614.226883] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[  614.226893] fbtft_device:      fb_ssd1322 spi0.0 5000kHz 8 bits mode=0x00
[  614.240326] fb_ssd1322 spi0.0: fbtft_gamma_parse_str() str=
[  614.240353] fb_ssd1322 spi0.0: 7 1 1 1 1 2 2 3 3 4 4 5 5 6 6
[  614.240416] fb_ssd1322 spi0.0: fbtft_request_gpios: 'reset' = GPIO24
[  614.240437] fb_ssd1322 spi0.0: fbtft_request_gpios: 'dc' = GPIO25
[  614.240449] fb_ssd1322 spi0.0: fbtft_verify_gpios()
[  614.240462] fb_ssd1322 spi0.0: fbtft_init_display()
[  614.240473] fb_ssd1322 spi0.0: fbtft_reset()
[  614.363792] fb_ssd1322 spi0.0: init: write(0xFD) 0x12
[  614.363872] fb_ssd1322 spi0.0: init: write(0xAE)
[  614.363899] fb_ssd1322 spi0.0: init: write(0xB3) 0xF3
[  614.363934] fb_ssd1322 spi0.0: init: write(0xCA) 0x3F
[  614.363967] fb_ssd1322 spi0.0: init: write(0xA2) 0x00
[  614.364000] fb_ssd1322 spi0.0: init: write(0xA1) 0x00
[  614.364035] fb_ssd1322 spi0.0: init: write(0xA0) 0x14 0x11
[  614.364070] fb_ssd1322 spi0.0: init: write(0xAB) 0x01
[  614.364104] fb_ssd1322 spi0.0: init: write(0xB4) 0xA0 0xFD
[  614.364139] fb_ssd1322 spi0.0: init: write(0xC1) 0x1F
[  614.364172] fb_ssd1322 spi0.0: init: write(0xC7) 0x7F
[  614.364205] fb_ssd1322 spi0.0: init: write(0xB1) 0xF0
[  614.364243] fb_ssd1322 spi0.0: init: write(0xD1) 0x82 0x1F 0xBB 0x0D
[  614.364285] fb_ssd1322 spi0.0: init: write(0xBE) 0x00
[  614.364317] fb_ssd1322 spi0.0: init: write(0xA6)
[  614.364340] fb_ssd1322 spi0.0: init: write(0xAF)
[  614.377952] fb_ssd1322 spi0.0: Display update: 2357 kB/s (13.569 ms), fps=0 (0.000 ms)
[  614.377972] fb_ssd1322 spi0.0: set_gamma()
[  614.378411] graphics fb1: fb_ssd1322 frame buffer, 256x64, 32 KiB video memory, 8 KiB DMA buffer memory, fps=20, spi0.0 at 5 MHz
```
I really wonder what matters.
If you need any additional information, let me know ;)

Thank you!

notro, 2015-09-15:
fbtft_device fails to find the spi master because it is loaded before spi-bcm2835:
```
[    4.153870] fbtft_device:  spi_busnum_to_master(0) returned NULL
[    4.161895] fbtft_device: failed to register SPI device
...
[    5.087098] spi spi0.0: setting up native-CS0 as GPIO 8
[    5.098428] spi spi0.1: setting up native-CS1 as GPIO 7
```
Try changing spi_bcm2708 to spi_bcm2835 in /etc/modules-load.d/modules.conf

computer007, 2015-09-15:
Thanks for fast replying!
That did the job.
However why is it needed to explicitly load "spi_bcm2835" whereas not with modprobe?
Thank you :)

notro, 2015-09-15:
I haven't got any experience with systemd, but that seems to be the ordering: first load modules-load.d modules, then run udevd which looks for a module for the SPI controller device:
```
[    4.057710] systemd[1]: Started udev Coldplug all Devices.
[    4.111212] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    4.131504] fbtft_device:  SPI devices registered:

[    4.485636] systemd-udevd[110]: starting version 215
[    5.087098] spi spi0.0: setting up native-CS0 as GPIO 8    (spi-bcm2835 writes these lines)
[    5.098428] spi spi0.1: setting up native-CS1 as GPIO 7
```
Maybe you can change the ordering with some systemd configuration file?

computer007, 2015-09-15:
Hum, don't know, that's out of my current skills. Anyway it works, so thank you again :).
I close the issue.

closed 2015-09-15 by computer007