notro/fbtft #324: need help with Adafruit 2.8" ILI9341 breakout board

ArduinoKid, 2015-09-27:
Currently I am trying to use an Adafruit 2.8" ILI9341 breakout board ( http://bit.ly/1xPvFbw ) with fbtft however I cant seem to get it working. If anyone can post instructions to get this display working in spi mode or can post a link to the instructions it would be greatly appreciated.

closed 2015-10-17 by ArduinoKid

pdp7, 2016-06-23:
@ArduinoKid did you ever get this working ok?

ArduinoKid, 2016-06-23:
@pdp7 I was eventually able to get this to work. I managed to use [this schematic](http://bit.ly/28RQgGy) from Adafruit, and [this guide](https://learn.adafruit.com/running-opengl-based-games-and-emulators-on-adafruit-pitft-displays?view=all#pitft-setup), also from Adafruit, to get this working. Since the pitft and the 2.8" breakout use the same display, i used the schematic to figure out what pins to connect the breakout to on the pi. I then used the guide to install the pitft software on the pi. After a reboot, the display was working fine. I have not looked into making use of the touch capabilities of the screen however, as it was unnecessary for my project. There was probably a better/easier way to get this to work, however I did not find it and this approach has worked for me.