notro/fbtft #325: 1.8" Sainsmart RPi 2, fbtft_request_gpios fail -16

andolfss, 2015-10-04:
Hi there, being new to all this, im not sure what is wrong, but I think all is installed properly, but when i run dmesgi get the following: 

fb_st7735r spi0.0: fbtft_request_gpios: gpio_request_one('dc'=24) failed with -16
fb_st7735r: probe of spi0.0 failed with error -16

I got a Sainsmart 1.8" TFT and a a Raspberry Pi 2. 

I'm not sure what I'll do, but ill try anything you'll suggest.

notro, 2015-10-04:
-16 is EBUSY, so that means someone has already claimed/requested the gpio.

andolfss, 2015-10-04:
So how do i Find who uses it? or should i just change to different GPIO?

notro, 2015-10-04:
The troubleshooting section shows how to do that: http://aron.ws/projects/lirc_rpi/

closed 2016-04-28 by notro