notro/fbtft #326: FB_IL9341 requests both DC and WR lines

sergeybeestar, 2015-10-07:
Hi,

Im using 8-bit SPI interface with a DC line to switch between command data. Trying the following command:

modprobe fbtft_device custom name=fb_ili9341 busnum=2 gpios=reset:4,dc:5

This is what I get in dmesg:
[  433.398999] fbtft_device:  SPI devices registered:
[  433.410001] fbtft_device:      spidev spi2.0 1000kHz 8 bits mode=0x03
[  433.421402] fbtft_device:  'fb' Platform devices registered:
[  433.433457] fbtft_device:  GPIOS used by 'fb_ili9341':
[  433.448914] fbtft_device:    'reset' = GPIO4
[  433.457247] fbtft_device:    'dc' = GPIO5
[  433.470309] fb_ili9341 fb_ili9341.0: Missing 'wr' gpio. Aborting.
[  433.481964] fbtft_device:  'fb' Platform devices registered:
[  433.494987] fb_ili9341: probe of fb_ili9341.0 failed with error -22
[  433.506116] fbtft_device:      fb_ili9341 id=0 pdata? yes

Why does it require wr? Or do I do something wrong?
Its a 2.2"  RPI screen, but using it on A13 sunxi compiled kernel.

notro, 2015-10-07:
You need the speed= argument to tell fbtft_device that this is a spi device.
Ref: http://lxr.free-electrons.com/source/drivers/staging/fbtft/fbtft_device.c#L1364

sergeybeestar, 2015-10-07:
Indeed thanks! was just to comment on it.

Now however having a big crash when it tries to load in SPI mode. Any idea what that could be?

  869.416322]
[  869.416325] IP: 0xc087d1a8:
d1a8 [  869.421039] d1a8  0000055c 0000055c 00000000 00000000 00000004 00000004 00000000 00000000 00000556 00000556 00000000 00000000 0000007f 0000007f 00000000 00000000

d1c8 [  869.436232] d1c8  00000000 00000000 00000000 00000000 00016053 00016053 00000000 00000000 00000193 00000193 00000000 00000000 00000000 00000000 00000000 00000000

d1e8 [  869.451425] d1e8  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 0004fb15 0004fb15 000001d6 000001d6 00016d1b 00016d1b 00000009 00000009

d208 [  869.466617] d208  0000011a 0000011a 00000000 00000000 00000000 00000000 00007a92 00007a92 00000000 00000000 00000000 00000000 00000d9d 00000d9d 00000000 00000000

d228 [  869.481809] d228  fffffffa fffffffa ffffe86e ffffe86e 00000901 00000901 00a42138 00a42138 0671f315 0671f315 0b964821 0b964821 0000f7eb 0000f7eb 00000000 00000000

d248 [  869.497002] d248  ffffe46e ffffe46e 00000000 00000000 00016d1b 00016d1b 00000000 00000000 00039619 00039619 00000000 00000000 ffffe46e ffffe46e 00000000 00000000

d268 [  869.512195] d268  fffffffa fffffffa fffffffa fffffffa eddaf79d eddaf79d 0000001a 0000001a f5a0a1ca f5a0a1ca 0000002e 0000002e f5a0a1ca f5a0a1ca 0000002e 0000002e

d288 [  869.527387] d288  cf2f5338 cf2f5338 ce2f2638 ce2f2638 cf82d330 cf82d330 ce2f2630 ce2f2630 00000000 00000000 00000000 00000000 0000000a 0000000a 00000000 00000000

FP: 0xcf83ccd4:
[  869.543795]
[  869.543797] FP: 0xcf83ccd4:
ccd4 [  869.548511] ccd4  205b0113 205b0113 38363820 38363820 3834382e 3834382e 5d333439 5d333439 c0020020 c0020020 60000113 60000113 ffffffff ffffffff 00000006 00000006

ccf4 [  869.563703] ccf4  cf83cd16 cf83cd16 9ae1acbb 9ae1acbb 0000006c 0000006c 4ae70638 4ae70638 000000ca 000000ca 00000001 00000001 00000113 00000113 cf83cd8c cf83cd8c

cd14 [  869.578896] cd14  cf82c980 cf82c980 0000000b 0000000b 00000000 00000000 00000000 00000000 00000001 00000001 c0047854 c0047854 cf83cde2 cf83cde2 c0755ee8 c0755ee8

cd34 [  869.594087] cd34  cf82c980 cf82c980 0000000b 0000000b 00000000 00000000 00000000 00000000 00000001 00000001 c0047854 c0047854 cf83cde2 cf83cde2 c0755ee8 c0755ee8

cd54 [  869.609280] cd54  c0031af4 c0031af4 c0047852 c0047852 00000000 00000000 cf83c000 cf83c000 00000001 00000001 c0755ee8 c0755ee8 c002f864 c002f864 c0755ee8 c0755ee8

cd74 [  869.624472] cd74  c0519568 c0519568 00000001 00000001 cf83c000 cf83c000 c0047852 c0047852 00000000 00000000 00000000 00000000 00000001 00000001 c0047854 c0047854

cd94 [  869.639664] cd94  cf83cde2 cf83cde2 c0755ee8 c0755ee8 c00127bc c00127bc cf83c2e8 cf83c2e8 0000000b 0000000b 00000008 00000008 bf000000 bf000000 63000004 63000004

cdb4 [  869.654857] cdb4  63373830 63373830 20306563 20306563 32353063 32353063 34613933 34613933 35306320 35306320 38373332 38373332 35652034 35652034 32333039 32333039

R0: 0xcf82d280:
[  869.671263]
[  869.671266] R0: 0xcf82d280:
d280 [  869.675980] d280  ffffffff ffffffff c0994a9c c0994a9c cf82d288 cf82d288 cf82d288 cf82d288 00000000 00000000 cf82d294 cf82d294 cf82d294 cf82d294 00000000 00000000

d2a0 [  869.691172] d2a0  00000000 00000000 00000000 00000000 00000001 00000001 cf82d2ac cf82d2ac cf82d2ac cf82d2ac cf82d2b4 cf82d2b4 cf82d2b4 cf82d2b4 cf7ced14 cf7ced14

d2c0 [  869.706365] d2c0  c002fdb0 c002fdb0 00000000 00000000 00000000 00000000 00000000 00000000 00000020 00000020 00000000 00000000 0000c350 0000c350 0000c350 0000c350

d2e0 [  869.721559] d2e0  00000000 00000000 00000000 00000000 ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff

d300 [  869.736752] d300  00000002 00000002 cf840000 cf840000 00000002 00000002 0420806c 0420806c 00000000 00000000 00000000 00000000 00000078 00000078 00000078 00000078

d320 [  869.751945] d320  00000078 00000078 00000000 00000000 c0523bcc c0523bcc 00000000 00000000 00000400 00000400 00400000 00400000 00000001 00000001 cf2f7478 cf2f7478

d340 [  869.767138] d340  00000000 00000000 cf82d344 cf82d344 cf82d344 cf82d344 00000000 00000000 4bc001e8 4bc001e8 000000ca 000000ca ab19bf2f ab19bf2f 0000000c 0000000c

d360 [  869.782330] d360  247d6d92 247d6d92 0000003b 0000003b 7932b259 7932b259 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

R5: 0xcf82d280:
[  869.798736]
[  869.798739] R5: 0xcf82d280:
d280 [  869.803453] d280  ffffffff ffffffff c0994a9c c0994a9c cf82d288 cf82d288 cf82d288 cf82d288 00000000 00000000 cf82d294 cf82d294 cf82d294 cf82d294 00000000 00000000

d2a0 [  869.818646] d2a0  00000000 00000000 00000000 00000000 00000001 00000001 cf82d2ac cf82d2ac cf82d2ac cf82d2ac cf82d2b4 cf82d2b4 cf82d2b4 cf82d2b4 cf7ced14 cf7ced14

d2c0 [  869.833837] d2c0  c002fdb0 c002fdb0 00000000 00000000 00000000 00000000 00000000 00000000 00000020 00000020 00000000 00000000 0000c350 0000c350 0000c350 0000c350

d2e0 [  869.849030] d2e0  00000000 00000000 00000000 00000000 ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff
[... 566 lines cut ...]

<0>Internal error: Oops: 17 [#30] PREEMPT ARM
[  874.023186] Internal error: Oops: 17 [#30] PREEMPT ARM
<d>Modules linked in:[  874.030146] Modules linked in: fbtft_device(+) fbtft_device(+) fb_ili9341 fb_ili9341 fbtft fbtft syscopyarea syscopyarea sysfillrect sysfillrect sysimgblt sysimgblt fb_sys_fops fb_sys_fops s [... cut ...]
r7 : c087d228  r6 : 00000000  r5 : cf82d300  r4 : 00000000
[  874.126282] r7 : c087d228  r6 : 00000000  r5 : cf82d300  r4 : 00000000
r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : cf82d300
[  874.138009] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : cf82d300
Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  874.150431] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Control: 10c5387d  Table: 4e234019  DAC: 00000015
[  874.162072] Control: 10c5387d  Table: 4e234019  DAC: 00000015

notro, 2015-10-07:
No

sergeybeestar, 2015-10-07:
ok, so t works if before that I insert:

modprobe fbtft dma=0

but I get these messages all the time on the console:

[spi]: drivers/spi/spi_sunxi.c(L583) [  993.085698] [spi]: drivers/spi/spi_sunxi.c(L583) unknow dma direction = 0
unknow dma direction = 0

closed 2016-04-28 by notro