notro/fbtft #329: fbtft_device: failed to register SPI device

umesh4b4, 2015-10-13:
Hi,

I am using fb_ili9341 driver and followed the steps provided the below link https://github.com/notro/fbtft/wiki/BeagleBone-Black

But when check the log file using dmesg command i am getting an error like this. 
 fbtft_device:  spi_busnum_to_master(0) returned NULL
fbtft_device: failed to register SPI device

Also i cannot see anything on the screen. I am using ADAFRUIT 2.8 TFT LCD https://www.adafruit.com/products/2090

Any help appreciated.

Thanks in Advance.

notro, 2015-10-14:
> fbtft_device: spi_busnum_to_master(0) returned NULL

This means that there is no SPI master/controller device/driver registered for bus 0.

umesh4b4, 2015-10-19:
Hi notro,

I tried to display some image through frame buffer created using fb_ili9341 module. I don't see any data coming out of SPI bus. Here is my device tree overlay code.

/*
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License version 2 as
 * published by the Free Software Foundation.
 */
/dts-v1/;
/plugin/;

/ {
    compatible = "ti,beaglebone", "ti,beaglebone-black";

    /* identification */
    part-number = "ADAFRUIT-ILI9341";
    version = "00A0";

    /* state the resources this cape uses */
    exclusive-use =
        /* the pin header uses */
        "P9.22",    /* spi1_sclk */
        "P9.21",    /* spi1_d0 MISO */
        "P9.18",    /* spi1_d1 MOSI */
        "P9.17",    /* spi1_cs0 */
        "P9.15",    /* gpio1_16 - DC */
        "P9.12",    /* gpio1_28 - reset */
        /* the hardware ip uses */
        "gpio1_16",
        "gpio1_28",
        "spi0";

    fragment@0 {
        target = <&am33xx_pinmux>;
        __overlay__ {
            bb_spi0_pins: pinmux_bb_spi0_pins {
                pinctrl-single,pins = <
                    0x150 0x30  /* spi0_sclk.spi0_sclk, MODE0 | INPUT_PULLUP | SPI0_SCLK P9_22 */
                    0x154 0x30  /* spi0_d0.spi0_d0, MODE0 | INPUT_PULLUP | SPI0_MISO P9_21 */
                    0x158 0x10  /* spi0_d0.spi0_d1, MODE0 | OUTPUT_PULLUP | SPI0_MOSI P9_18 */
                    0x15c 0x10  /* spi0_cs0.spi0_cs0, MODE0 | OUTPUT_PULLUP | SPI0_CS0 P9_17 */
                >;
            };

            lcd_ctrl_pinmux: lcd_ctrl_pins {
                pinctrl-single,pins = <
                    0x040 0x17  /* P9.15 gpio1_16  OUTPUT_PULLUP | MODE7 */
                    0x078 0x17  /* P9.12 gpio1_28  OUTPUT_PULLUP | MODE7 */
                >;
            };
        };
    };

    fragment@1 {
        target = <&spi0>;
        __overlay__ {
            status = "okay";
            pinctrl-names = "default";
            pinctrl-0 = <&bb_spi0_pins>;

            #address-cells = <1>;
            #size-cells = <0>;

            lcd@0{
                compatible = "ilitek,ili9341";
                reg = <0>;

                pinctrl-names = "default";
                pinctrl-0 = <&lcd_ctrl_pinmux>;

                spi-max-frequency = <48000000>;
                rotate = <270>;
                bgr;
                fps = <50>;
                buswidth = <8>;
                regwidth = <8>;
                reset-gpios = <&gpio2 28 0>;
                dc-gpios = <&gpio2 16 0>;

                init = <0x10000EF 0x03 0x80 0x02
        0x10000CF 0x00 0xC1 0x30
        0x10000ED 0x64 0x03 0x12 0x81   
        0x10000E8 0x85 0x00 0x78
        0x10000CB 0x39 0x2C 0x00 0x34 0x02
        0x10000F7 0x20
        0x10000EA 0x00 0x00
        0x10000C0 0x23
        0x10000C1 0x10
        0x10000C5 0x3E 0x28
        0x10000C7 0x86
        0x1000036 0x48
        0x100003A 0x55
        0x10000B1 0x00 0x18
        0x10000B6 0x08 0x82 0x27
        0x10000F2 0x00
        0x1000026 0x01
        0x10000E0 0x0F 0x31 0x2B 0x0C 0x0E 0x08 0x4E 0xF1 0x37 0x07 0x10 0x03 0x0E 0x09 0x00
        0x10000E1 0x00 0x0E 0x14 0x03 0x11 0x07 0x31 0xC1 0x48 0x08 0x0F 0x0C 0x31 0x36 0x0F
        0x1000011
        0x2000078
        0x1000029>;

                debug = <3>;
            };
        };
    };
};

Is there anything i need to change in DTS file?

Thanks

notro, 2015-10-19:
You don't need fbtft_device if you use a DT overlay. It's mostly for non-DT platforms.
Check that the device is really registered: /sys/bus/spi/devices/
Use dmesg to see driver loading messages: https://github.com/notro/fbtft/wiki/BeagleBone-Black#enable-overlay

closed 2016-04-28 by notro