notro/fbtft #332: Sainsmart 2.8 240x320 -- nothing working :(

r349, 2015-10-17:
I've tried everything to get this touchscreen to work using the ILTD28 wiki. So far, as far as I've gotten is just a white screen that turns on and off when I reboot my Pi (B+). I'm wondering if you shed some light on this stupid touch screen.

I've wired it using the parallel databus gpio pinouts but that yields nothing. I've tried that pinout on it's own, that pinout in combo with the SPI interface wiring, just the SPI wiring on it's own, but still no marked difference. 

Here's what happens when I run these commands-
sudo modprobe fbtft_device name=flexfb gpios=reset:25,dc:24
lsmod output:
Module                  Size  Used by
flexfb                 14438  0 
fbtft_device           35066  0 
fbtft                  34327  2 flexfb,fbtft_device
fb_sys_fops             1573  1 fbtft
syscopyarea             3113  1 fbtft
sysfillrect             3592  1 fbtft
sysimgblt               2286  1 fbtft
ads7846                12520  0 
hwmon                   4420  1 ads7846

dmesg prompts the following...

fbtft_device:  SPI devices registered:
[  872.177708] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[  872.177728] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[  872.177741] fbtft_device:  'fb' Platform devices registered:
[  872.177787] fbtft_device:      soc:fb id=-1 pdata? no
[  872.177872] fbtft_device: Deleting spi0.0
[  872.187400] spi spi0.0: setting up native-CS0 as GPIO 8
[  872.187605] fbtft_device:  GPIOS used by 'flexfb':
[  872.187628] fbtft_device:    'reset' = GPIO25
[  872.187642] fbtft_device:    'dc' = GPIO24
[  872.187653] fbtft_device:  SPI devices registered:
[  872.187673] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[  872.187763] fbtft_device:      flexfb spi0.0 32000kHz 8 bits mode=0x00
[  881.870506] flexfb: module is from the staging directory, the quality is unknown, you have been warned.
[  881.871363] flexfb: unknown parameter 'debug' ignored
[  881.871392] flexfb: unknown parameter 'rotate' ignored
[  882.419198] graphics fb1: flexfb frame buffer, 240x320, 150 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 32 MHz
[  893.111686] bcm2708_fb soc:fb: Unknown ioctl 0x40187a22

I've tried rotating, etc. and I've run every command I can find by googling this issue but so far nothing but just a white screen. Any ideas? Have I missed something?

notro, 2015-10-31:
> I've tried everything to get this touchscreen to work using the ILTD28 wiki.

Can you give me the url, searching yielded no result.

r349, 2015-11-05:
http://www.sainsmart.com/sainsmart-2-8-2-8-inch-tft-lcd-240x320-arduino-due-mega2560-r3-raspberry-pi-1.html

JeremyIglehart, 2015-12-17:
@r349 - I believe that the link for this device you're using is going to a 2.4" screen...

I have this screen: http://www.sainsmart.com/sainsmart-2-8-2-8-inch-tft-lcd-240x320-arduino-due-mega2560-r3-raspberry-pi.html

And I've been banging my head against the wall for more than three days now with no luck whatsoever.

@notro - I'm very eager to get this device working.  And though I'm not nearly as smart as you are on this issue - I am writing a tutorial to show people how I was able to get this whole screen setup from the beginning in a very detailed form to help make this easier.

I know that this screen was sold at Microcenter and there are a BUNCH of them in the clearance because people have bought them and then returned them because they couldn't figure out how to make them work.

I am volunteering to write a tutorial for this screen beginning to end in an extra detailed fashion, I just need a little assistance along the way to get everything working - I will include pictures and video to show people how to do the pinout and everything.

Would you be willing to assist me along the way?

notro, 2015-12-19:
All I can say is that you need to find the init sequence for the display, which seem to be a problem with many of these cheap displays. If the display could be used out of the box, they could charge more.
The link you gave has a link to Documents which is a zip file containing UTFT, but there is no mention of which utft model to use. There's a pdf of supported models, but no sainsmart displays are listed.

r349, 2015-12-20:
I've totally given up on this screen, mainly because I don't think it connects. @JeremyIglehart I do have the same screen, found as much info on it as possible and just could not get it to happen. I got mine from the clearance bin at Microcenter, funny that you mention it- it's back there now. 

Main reason why I gave up was that the pinout labels on the screen seemed to override each other and no matter the documentation of the pinouts I found and how to connect them to the Pi, there was always at least one pin that overlapped another function.

notro, 2016-04-28:
Closing old issues, reopen if needed.

closed 2016-04-28 by notro

gretzky, 2016-04-29:
Thanks for all you do bud

Closed #332 [https://github.com/notro/fbtft/issues/332] .