notro/fbtft #335: Tontech 480 x 320 LCD being incredibly slow

woolfman007, 2015-10-26:
Hi guys, 

I'm fairly new to the topic, so I'm certainly lacking a profund understanding of the technical concepts; anyway here's what I discovered:

I got the said Tontech LCD and made some tests, which showed that the fps I can get with this is super-slow. dmesg tells me, that the data transfer rate is about 1200kB / s, even though the SPI runs @128 MHz. I'll paste the relevant dmesg output at the end. 

I had to download the latest overlay.dtb, since I would get a white screen with the linux stock one (I've read about that already in a couple of posts).

What my experiements boiled down, was outputting a mpg file directly to /dev/fb1 with mplayer, like done by others for benchmarking reasons. The video impression was quite horrible (and the file I used was 320 x 240!). You could actually see / feel the image blt propagating from top to bottom in the single frames. From looking at it, I would guess that this was no more than 4 - 5 fps. 

Hope anyone can help me...

Thanks a lot & best, Wolfgang

------ dmesg output ------

[    4.612040] fb_s6d02a1: module is from the staging directory, the quality is unknown, you have been warned.
[    4.616147] fb_s6d02a1 spi0.0: fbtft_request_one_gpio: 'reset-gpios' = GPIO15
[    4.616199] fb_s6d02a1 spi0.0: fbtft_request_one_gpio: 'dc-gpios' = GPIO25
[    4.616249] fb_s6d02a1 spi0.0: fbtft_request_one_gpio: 'led-gpios' = GPIO18
[    4.616391] fb_s6d02a1 spi0.0: fbtft_verify_gpios()
[    4.616404] fb_s6d02a1 spi0.0: fbtft_init_display_dt()
[    4.616419] fb_s6d02a1 spi0.0: init: write_register: B0 00
[    4.616663] fb_s6d02a1 spi0.0: init: write_register: 11
[    4.616746] fb_s6d02a1 spi0.0: init: msleep(255)
[    4.877862] fb_s6d02a1 spi0.0: init: write_register: B3 02 00 00 00
[    4.894583] fb_s6d02a1 spi0.0: init: write_register: C0 13 3B 00 02 00 01 00 43
[    4.909618] fb_s6d02a1 spi0.0: init: write_register: C1 08 16 08 08
[    4.926645] fb_s6d02a1 spi0.0: init: write_register: C4 11 07 03 03
[    4.946943] fb_s6d02a1 spi0.0: init: write_register: C6 00
[    4.959791] fb_s6d02a1 spi0.0: init: write_register: C8 03 03 13 5C 03 07 14 08 00 21 08 14 07 53 0C 13 03 03 21 00
[    4.977372] fb_s6d02a1 spi0.0: init: write_register: 35 00
[    4.990017] fb_s6d02a1 spi0.0: init: write_register: 36 A0
[    5.002453] fb_s6d02a1 spi0.0: init: write_register: 3A 55
[    5.015251] fb_s6d02a1 spi0.0: init: write_register: 44 00 01
[    5.027295] fb_s6d02a1 spi0.0: init: write_register: D0 07 07 1D 03
[    5.039686] fb_s6d02a1 spi0.0: init: write_register: D1 03 30 10
[    5.051750] fb_s6d02a1 spi0.0: init: write_register: D2 03 14 04
[    5.063766] fb_s6d02a1 spi0.0: init: write_register: 29
[    5.074909] fb_s6d02a1 spi0.0: init: write_register: 2C
[    5.085996] fb_s6d02a1 spi0.0: set_var()
[    5.364511] fb_s6d02a1 spi0.0: Display update: 1115 kB/s (268.798 ms), fps=0 (0.000 ms)
[    5.378129] fb_s6d02a1 spi0.0: fbtft_register_backlight()
[    5.389593] graphics fb1: fb_s6d02a1 frame buffer, 480x320, 300 KiB video memory, 4 KiB DMA buffer memory, fps=33, spi0.0 at 128 MHz
[    5.407134] fb_s6d02a1 spi0.0: fbtft_backlight_update_status: polarity=0, power=0, fb_blank=0

notro, 2015-10-26:
Writing directly to /dev/fb1 (no mmap) will probably give lousy performance, fbtft is not optimized for that.

Which kernel are you using?
The latest ```sudo rpi-update``` kernel should give ~10MB/s throughput.

debug=32 will print performance numbers to the kernel log for each display update. If you run a movie that updates the whole screen, you will get the measured fps in the kernel log (dmesg).

https://github.com/notro/fbtft/wiki#install
https://github.com/notro/fbtft/wiki/Performance#mz61581-pi-ext
https://github.com/raspberrypi/linux/pull/1132

woolfman007, 2015-10-26:
Indeed, apt-get update / upgrade multiplied the performance by more than 10! I'm deeply impressed! Now I get around 12MB xfer rate and decent fps. Is that related to a new kernel oder a new fptft driver?

thanks so much!

notro, 2015-10-26:
It's down to the spi controller driver now supporting DMA (spi-bcm2835). The official Pi kernel got support for that not long ago. Previously I maintained a prebuilt fbtft kernel that used an out-of-tree spi dma driver.

closed 2016-04-28 by notro