notro/fbtft #336: Waveshare 3.5 does not start with /etc/modules entry...

RasTschubai, 2015-10-27:
I'm a fairly new RPI B+ user and installed the **Jessie Raspian with Kernel 4.0.7**.
Following the instructions here I was able to activate my Waveshare with the modprobe commands. But when I enter the modules in the /etc/modules file, the display stays white. dmesg and lsmod told me no *fbtft* nor *flexbf* was loaded. 
`dmesg` shows an error:
````
[    4.602648] systemd[1]: Started Restore / save the current clock.
[    4.629752] systemd[1]: systemd-modules-load.service: main process exited, code=exited, status=1/FAILURE
[    4.645697] systemd[1]: Failed to start Load Kernel Modules.
[    4.651793] systemd[1]: Unit systemd-modules-load.service entered failed state.
````
spi seems to be included ...`lsmod`
````

pi@RasPi ~ $ lsmod
Module                  Size  Used by
cfg80211              499071  0
rfkill                 22468  2 cfg80211
8192cu                569588  0
spi_bcm2708             8696  0
uio_pdrv_genirq         3682  0
uio                     9989  1 uio_pdrv_genirq
i2c_dev                 6749  0
snd_bcm2835            22339  0
snd_pcm                92165  1 snd_bcm2835
snd_timer              23475  1 snd_pcm
snd                    67406  3 snd_bcm2835,snd_timer,snd_pcm
fuse                   92407  1
ipv6                  353813  22

````
As I said... with following commands the dispay works...
````
pi@RasPi ~ $ sudo modprobe flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
pi@RasPi ~ $ sudo modprobe fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 gpios=reset:25,dc:24
````
Perhaps its an issue with the *Jessie* system? Any idea?
Martin

dan200, 2015-10-29:
Also experienced this. My workaround was to add the modprobe instructions to /etc/rc.local instead

RasTschubai, 2015-10-29:
Ahh... now thats a workaround I can live with! It's alive!
Thank you Dan.
BTW, were you able to get the touch working? With the ads7846 nothing happens

closed 2016-04-28 by notro