notro/fbtft #338: TFT 3.5 dead after raspbian upgrade

maniscalcus, 2015-11-07:
Hi there! 
I'm pretty new to raspberry's world and i'm in trouble :-( 
I have a 3.5" inch tft lcd with touchscreen( unknow brand the only thing i know is that it's back is blue similar to the waveshare one and as it with 3.5 inch rpi lcd v3.0 written on it ) and from the moment that i upgrade from wheezy to jessie my lcd died(only white light and red led on). I tried to follow the suggest in https://github.com/notro/fbtft/issues/215
and actually when i modprobe, as suggested by notro first and sudobrew next,

sudo modprobe flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
sudo modprobe fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 gpios=reset:25,dc:24

my screen wake up and if i launch some graphic display application they work( but the touchscreen is fully inverted,but this is another quest..). Anyway when i add flexfb and fbtft in my modules file and reboot nothing happen( all seems to work fine in fact nothing in dmseg !) and the only way i have to make it is manually modprobe again..
is there any solution to my problem? I have to upgrade because i need to work with the latest version of network-manager which isn't avaible for wheezy and previous kernel...
Thanks a lot

notro, 2015-11-07:
Jessie uses systemd which has two directories, one for the modules to load and one for the parameters: https://github.com/notro/fbtft/wiki#make-it-permanent-debian-jessie-arch-linux-systemd

Texy, 2015-11-08:
Hi notro,
can you please give examples of the format to use for both of these files, eg for the hy28b would it be
/etc/modules-load.d/raspberrypi.conf   :
fbtft_device

and

/etc/modprobe.d/fbtft.conf  :
options fbtft_device name=hy28b speed=32000000 rotate=90

?

thanks
Texy

notro, 2015-11-08:
Hi Texy

Turns out systemd processes /etc/modules-load.d/ before it starts udevadm which autoloads modules for devices that have no matching builtin driver like spi-bcm2835.
So when fbtft_device is loaded, spi-bcm2835 hasn't been loaded yet.

This can be solved by manually loading it first:
```
spi-bcm2835
fbtft_device
```

Texy, 2015-11-21:
Apologies for the late reply. Yes this has fixed the issue, however I was only trying it to see if it could cure issue 339, which it doesn't.
Thanks,
Texy

barartm, 2015-11-25:
Hello Texy,

How do you get it work on jessie? I can not make it work :/ could you please share how?

Thanks alot

closed 2016-04-28 by notro