notro/fbtft #339: Jessie /FBTFT/Pi

fredaunger, 2015-11-10:
I am experiencing  a problem with the boot sequence booting to /dev/fb1/ and then when it reaches the sequence to process the user GUI it redirects to /dev/fb0 and outputs to HDMI. Hope this is enough info to help resolve the problem.
Fred Aunger

fredaunger, 2015-11-11:
Further to the above on a more detailed look at the boot sequence the following seems to happen.(I am using a 2.8" tuch screen) It starts the sequence by outputting to the HDMI interface then switches to the 2.8" screen and then finally switches back to the HDMI interface with the user GUI information.
I have used Paspian Wheezy, Jessie and two different Pi's(256meg and 512meg) and get the same result.
Fred Aunger

notro, 2015-11-11:
Does this help: https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian#where-to-display-x-windows

Texy, 2015-11-12:
I can confirm this issue, although I have  only witnessed it with the Jessie build. This problem isn't related to X-Windows, as the console is effected too. Just to be clear, the issue is with single core Pi's only - the Pi 2B does not have the problem. I can move an SD card from a Pi 2B where it works fine, to a Pi B and I can see the issue.
Texy

notro, 2015-11-21:
I tried this on a Pi1 b+ with a custom kernel and I got a console on /dev/fb1.
I can't look at it with a vanilla setup now, but you could add 'debug' to /boot/cmdline.txt and then use 'sudo journalctl' to print the very detailed boot log. Find the last line printed on /dev/fb1 and look at what comes next in the log. Maybe you can spot the service that switches the console (if any).

The systemd service files are mostly loaded from /lib/systemd/system
Man: http://www.freedesktop.org/software/systemd/man/systemd.unit.html
Source: https://github.com/systemd/systemd/tree/master/units

Texy, 2015-11-22:
Hi Notro,
I couldn't make head nor tail of the journalctl logs, but you did prompt me to take a closer look at the point just before the output switched from the fb1 display. By default now the Raspbian image boots to desktop. You need to use raspi-config to select boot to Console.As the X settings hadn't been set for the display yet, it couldn't display anything. Consider this issue solved.
Thanks for you help, as always.,
Texy

closed 2016-04-28 by notro