notro/fbtft #340: 3.5 Inch Screen not working

shekit, 2015-11-11:
Hi,

I bought this screen(http://www.alibaba.com/product-detail/Raspberry-Pi-2-3-5-inch_60348819079.html) from a store here in New York and have been unable to get it to work. I havent been able to find any documentation about this screen. I am also unsure which screen option to select from the file. I have tried 'pitft', 'waveshare32b' and 'tinylcd35' however in all cases I am left with a blank white screen. In each case the raspPi also became unresponsive and I needed to reinstall the OS. I was hoping you could help me in identifying the screen or suggest which screen option to enter.

Thanks!

shekit, 2015-11-11:
Also I followed the steps suggested in this previous github issue (https://github.com/notro/fbtft/issues/215) by sudobrew and others. When I ran 'sudo modprobe ...' the screen did turn black at one stage but that's about it. When I rebooted the pi it went back to the white screen. Also 'dmesg | grep graphics' does not return anything. Any help would be great!

shekit, 2015-11-11:
Hi I got the screen to work. Finally! Phew. However the screen is really laggy. How do I increase the FPS. I tried bumping up the speed from 16000000 to 32000000 but the GUI looks really bad. I also added this line in the '/boot/config.txt':  dtoverlay=flexfb,fps=50,speed=32000000,debug=32 .

MunhozThiago, 2015-11-11:
Could you please describe the model of your LCD and the steps you took to make it work? Please put the whole commands.

shekit, 2015-11-12:
Hi,

The screen I am using is a generic brand. It has nothing mentioned and no documentation that I could find. Though this is how it looks (http://www.alibaba.com/product-detail/Raspberry-Pi-2-3-5-inch_60348819079.html) 

I followed the steps used by sudobrew in this issue (#215). I still havent got the touch working since I don't know the chip it uses. Also permanent settings are still not working. I have to ssh into the pi and use the sudo modprobe commands to get the LCD running.

shekit, 2015-11-12:
The main issue I'm having now is the screen is really laggy. It becomes really apparent when I'm viewing gifs. It's like a swipe passes through the full screen every time the screen updates. Could this be a screen issue or an issue with using a SPI screen? Is this something that is experienced with all small TFT screens?

I am unable to bump up the speed beyond 16000000 as it causes the screen to distort.I have also overclocked the RPi2 and tried increasing gpu_mem to 256 but there seems to be no visible improvement.

The monitor plugged in through the HDMI works fine with no lagginess.

Texy, 2015-11-12:
All screens are different - not all of them are 'fast' enough for your purpose. Whilst it does work with the pi, it doesn't work well enough.  Research what others find suitable and buy that.

andig, 2015-11-29:
@shekit I've had similar problems. As workaround I've booted the image coming with the display and took note of the parameters instead of guessing them. Wfm.

FreshXOpenSource, 2016-01-12:
@shekit : can you please provide the parameters to load the flexfb module in linux?

frahugo, 2016-01-23:
I made the mistake of ordering this noname screen.  @andig: you were able to make it work? What params should we use?  Thanks

andig, 2016-01-23:
imho each screen is different. I've finally installed the distro that came with the display docs, took note of the parameters and voila- those also work on recent raspbian. I could check what those were, but most likely that won't help you any.

frahugo, 2016-01-23:
Thx @andig. I'll try to find an image (didn't get one with the screen).

andig, 2016-01-23:
In my case I've eailed support and they sent download links per mail. Best luck...

Tom-Neverwinter, 2016-02-22:
http://www.aliexpress.com/item/1pc-3-5-inch-B-B-LCD-Touch-Screen-Display-Module-320-x-480-for-Raspberry/32408089216.html

I purchased one of these screens, I'm assuming I'm sol based on the speeds these screens are capable of.

stevstrong, 2017-01-11:
Here a way how to increase the speed of your display screen:
http://stm32duino.com/viewtopic.php?f=9&t=1654&p=22332#p22315

closed 2021-08-24 by github-actions[bot]