notro/fbtft #342: ILI9341V compatability with fb_ili9341.c

umesh4b4, 2015-11-20:
Hi Notro,

I have an LCD which is having ili9341V controller and no DC/RS pin on it. I have changed the initialise sequence as per my LCD in fb_ili9341. But I could not able to see any output. I see only backlight turned on. I appreciate for your suggestions on what things i need to change to get my LCD to work.

Thanks in Advance.

notro, 2015-11-20:
Maybe it's configured for 9-bit mode. The 9th bit being DC. The IM0-3 pins on the controller sets the mode.

umesh4b4, 2015-11-23:
Yes it's configured for 9-bit mode. But what changes i need to do in fb_ili9341.c driver file to work for my LCD. There are no IM0-3 pins on my lcd breakout board. 

Thanks

notro, 2015-11-23:
Use buswidth=9 with fbtft_device or buswidth = <9>; if you use Device Tree.

umesh4b4, 2015-11-24:
Thanks Notro

I am using Device Tree and changed buswidth = 9 but now i am getting following error message 

[  529.584449] fb_ili9341 fb_ili9341.0: no default functions for regwidth=8 and buswidth=9

when looked into fbtft-core.c i found the following condition
if (display->regwidth == 8 && display->buswidth == 9 && par->spi) {
		par->fbtftops.write_register = fbtft_write_reg8_bus9;

And when i looked into fbtft.h here we are passing NULL for spi_device parameter.

fbtft_probe_common(_display, NULL, pdev);

I have also tried with fbtft_device but i got same error message as above.

notro, 2015-11-24:
Is it a SPI or parallel bus (pdev) connected display?

umesh4b4, 2015-11-24:
Its SPI device. One more thing fb_ili9341 driver is working fine with Adafruit Display in 8 bit mode. I have just changed buswidth to 9 in the Device tree file that i used with Adafruit display.

notro, 2015-11-24:
I don't understand why it doesn't work. spi_device is passed in:
```
#define FBTFT_REGISTER_DRIVER(_name, _compatible, _display)                \
                                                                           \
static int fbtft_driver_probe_spi(struct spi_device *spi)                  \
{                                                                          \
        return fbtft_probe_common(_display, spi, NULL);                    \
}                                                                          \
```

umesh4b4, 2015-11-24:
Yes, I already saw it. when I use buswidth = 8 with Adafruit display its working without any issue. But Issue is only when buswidth is 9. Here is my DTS file

/*
* Copyright (C) 2013 Matt Ranostay <mranostay@gmail.com>
*
* This program is free software; you can redistribute it and/or modify
* it under the terms of the GNU General Public License version 2 as
* published by the Free Software Foundation.
*/
/dts-v1/;
/plugin/;

/*
 * Pin assignments
 *
 * Module      Connector
 * LITE    <-  P8.19
 * MISO    ->  P9.29
 * SCK     <-  P9.31
 * MOSI    <-  P9.30
 * TFT_CS  <-  P9.28
 * CARD_CS <-  NC
 * D/C     <-  P9.25
 * RESET   <-  P9.27
 *
 */

/ {
	compatible = "ti,beaglebone", "ti,beaglebone-black";
	part-number = "ADAFRUIT-ILI9341";
	version = "00A0";

	/* state the resources this cape uses */
	exclusive-use =
		/* the pin header uses */
		"P8.19",	/* bl: ehrpwm2A */
		"P9.27",	/* lcd: gpio3_19 */
		"P9.25",	/* lcd: gpio3_21 */
		"P9.31",	/* spi: spi1_sclk */
		"P9.29",	/* spi: spi1_d0 */
		"P9.30",	/* spi: spi1_d1 */
		"P9.28",	/* spi: spi1_cs0 */
		/* the hardware IP uses */
		"gpio3_19",
		"gpio3_21",
		"ehrpwm2A",
		"spi1";

	fragment@0 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			pwm_backlight_pins: pinmux_pwm_backlight_pins {
				pinctrl-single,pins = <
					0x020  0x4      /* gpmc_ad8.gpio0_22 | MODE4 */
				>;
			};

			bone_adafruit_lcd_pins: pinmux_bone_adafruit_lcd_pins {
				pinctrl-single,pins = <
					0x1a4 0x17      /* mcasp0_fsr.gpio3_19, OUTPUT_PULLUP | MODE7 */
					0x1ac 0x17      /* mcasp0_ahclkx.gpio3_21, OUTPUT_PULLUP | MODE7 */
				>;
			};

			bone_adafruit_spi1_pins: pinmux_adafruit_spi1_pins {
				pinctrl-single,pins = <
					0x190 0x33	/* mcasp0_aclkx.spi1_sclk, INPUT_PULLUP | MODE3 */
					0x194 0x33	/* mcasp0_fsx.spi1_d0, INPUT_PULLUP | MODE3 */
					0x198 0x13	/* mcasp0_axr0.spi1_d1, OUTPUT_PULLUP | MODE3 */
					0x19c 0x13	/* mcasp0_ahclkr.spi1_cs0, OUTPUT_PULLUP | MODE3 */
				>;
			};
		};
	};

	fragment@1 {
		target = <&epwmss2>;
		__overlay__ {
			status = "okay";
		};
	};

	fragment@2 {
		target = <&ehrpwm2>;
		__overlay__ {
			status = "okay";
		};
	};

	fragment@3 {
		target = <&spi1>;

		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;

			status		= "okay";
			pinctrl-names	= "default";
			pinctrl-0	= <&bone_adafruit_spi1_pins>;

			lcd@0 {
				#address-cells = <1>;
				#size-cells = <0>;

				compatible = "ilitek,ili9341";
				reg = <0>;

				spi-max-frequency = <64000000>;
				spi-cpol;
				spi-cpha;
				
				rotate = <270>;
               			bgr;
               			fps = <50>;
               			buswidth = <8>;
                		regwidth = <8>;

				pinctrl-names = "default";
				pinctrl-0 = <&bone_adafruit_lcd_pins>;

				reset-gpios = <&gpio4 19 0>;
				dc-gpios = <&gpio4 21 0>;
			};
		};
	};

	fragment@4 {
		target = <&ocp>;

		/* avoid stupid warning */
		#address-cells = <1>;
		#size-cells = <1>;

		__overlay__ {
			adafruit-tft-backlight {
				compatible	= "pwm-backlight";
				pinctrl-names	= "default";
				pinctrl-0	= <&pwm_backlight_pins>;

				pwms = <&ehrpwm2 0 500000 0>;

				pwm-names = "backlight";
				brightness-levels = <0 4 8 16 32 64 128 255>;
				default-brightness-level = <7>; /* index to the array above */
			};
		};
	};
};

umesh4b4, 2015-11-30:
Hi Notro,

I am now got frame buffer created and i have figure out the problem. Just look at init function in fbtft.h. Here we are initialising spi device and platform device. When it is creating Platform device i am getting error message. However Frame buffer device is created but I am not getting any data on my LCD.

static int __init fbtft_driver_module_init(void)                           \
{                                                                          \
	int ret;                                                           \
									   \
	ret = spi_register_driver(&fbtft_driver_spi_driver);               \
	if (ret < 0)                                                       \
		return ret;                                                \
	return platform_driver_register(&fbtft_driver_platform_driver);    \
}

notro, 2015-11-30:
This is driver registration not device probing.

fb_ili9341 is a (loadable) kernel module that through fbtft_driver_module_init() registers one spi driver and one platform driver.
fbtft_device or Device Tree creates a spi or platform device that is passed to the fbtft_driver_probe_{spi,pdev} probe function.

If you add ```debug = <7>;``` property to DT you'll get a lot of details in the kernel log.

ivanroberto, 2015-12-10:
Hi Notro,

I'm have the problem like the umesh4b4, but i'm using fbtft_device.
My display controller is ILI9340 and it was configurated (in factory) as 3-wire SPI, so it doesn't have D/C pin. Part number is: TM028HDZ25. I need to change SPI to 9 bits, but i can't do that...

I try to type this line to load de module:
sudo modprobe fbtft_device custom name=fb_ili9340 gpios=reset:25,led:18 speed=32000000 buswidth=9 bgr=1

But, no luck...

I have another display with ILI9340 4-wire SPI (with D/C pin), and this command works flawless
sudo modprobe fbtft_device name=adafruit22a

I'm pretty shure that these displays is almost the same, the only diference is 4-wire to 3-wire SPI.

Can you help me???

Thanks

notro, 2015-12-11:
> But, no luck...

This isn't very descriptive, but assuming the driver loads ok, I would try and lower the speed.

closed 2016-04-28 by notro