notro/fbtft #343: Driver support for RGB interface(C.H.I.P.)?

peterlin82, 2015-11-22:
Reference 
https://www.kickstarter.com/projects/1598272670/chip-the-worlds-first-9-computer/description
and
https://github.com/NextThingCo/CHIP-Hardware/blob/master/ALPHA%20CHIP%20v0_21%20PINOUT.png

C.H.I.P. is RGB interface.

Reference https://github.com/notro/fbtft/blob/master/fbtft_device.c
I only found MCU,SPI interface.
Does fbtft support for RGB interface?

thanks a lot.

notro, 2015-11-22:
> Does fbtft support for RGB interface?

No it doesn't.

peterlin82, 2015-11-22:
Is it possible in future?

notro, 2015-11-23:
No, this interface has different implementations on each SoC and driving it by bitbanging gpios is too slow. Besides the PocketCHIP has an LCD which is most likely connected to the RGB interface, meaning there's already a driver for that interface.

peterlin82, 2015-11-23:
notro,thank you very much.

pdp7, 2015-12-28:
fyi - it looks like it should be possible to use fbtft on CHIP with a SPI display:
https://bbs.nextthing.co/t/i2c-oled-display-library-for-c-h-i-p/900/3
```
Sure, it would be interesting to try SPI. SPI2 pins are 27-30 on the U14 header. By default, these pins are probably assigned to CSI in the device tree. If so, sun5i-r8-chip.dtb should be changed and recompiled (no recompilation of the kernel or modules necessary). This would enable those pins for SPI peripherals.
```

closed 2016-04-28 by notro