notro/fbtft #344: XPT2046 touchscreen controller trouble - configuration or not?

UncleGollum, 2015-11-22:
I have a rpi B+ with 3,2" display (fb_ili9340) and XPT2046 touch controller. Previously 3.18.10, now 4.0.7 kernel.

Display works as waveshare32b, but no touchscreen by default. Then load overlay in /boot/config.txt:
```
dtoverlay=ads7846,speed=500000,penirq=17,swapxy=1
```
device appears in lsmod and dmesg
```
ads7846 10636 0
```
```
[ 8.135143] ads7846 spi0.1: touchscreen, irq 411 
[ 8.281637] input: ADS7846 Touchscreen as /devices/platform/soc/20204000.spi/spi_master/spi0/spi0.1/input/input0 
[ 14.208572] fbtft_device: ads7846 spi0.1 500kHz 8 bits mode=0x00 
[ 14.294753] fbtft_device: ads7846 spi0.1 500kHz 8 bits mode=0x00
```
Now, then i tap screen, interrupts count increasing
```
cat /proc/interrupts | grep ads 411: 433 pinctrl-bcm2835 17 ads7846
```

But nothing happens then i use evtest and ts_calibrate.

notro, 2015-11-24:
This is strange. When the driver interrupt function (ads7846_hard_irq) is called it checks to see if the gpio is low and if so the threaded handler (ads7846_irq) is called and the chip is read.
Check with ```dmesg``` that you don't get an error.

If there's no error, do you get a couple of interrupts per tap or many?
I'm wondering if the interrupt line could be floating. You could try the ```penirq_pull=2``` overlay parameter to pull up the gpio.

This file shows the drivers pen_down state: /sys/bus/spi/devices/spi0.1/pen_down
Cat it while touching the panel.

Source: http://lxr.free-electrons.com/source/drivers/input/touchscreen/ads7846.c

UncleGollum, 2015-11-24:
Thanks for reply!

dmesg has no errors.
is many interrupts per tap, and i don't think what it my hands are shaking so. Sometimes result is adequate (1 or 2), sometimes dozens, usually then i choose another place on screen.
File /sys/bus/spi/devices/spi0.1/pen_down always have "0"

And, my overlay string now looks:
```
dtoverlay=ads7846,speed=500000,penirq=17,swapxy=1,penirq_pull=2
```
is this correct? 
Maibe GPIO settings is wrong?
```
gpio -g readall
 +-----+-----+---------+------+---+--B Plus--+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 |   IN | 1 |  3 || 4  |   |      | 5V      |     |     |
 |   3 |   9 |   SCL.1 |   IN | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT0 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT0 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 1 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |  OUT | 1 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |  OUT | 1 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI | ALT0 | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO | ALT0 | 0 | 21 || 22 | 1 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK | ALT0 | 0 | 23 || 24 | 1 | ALT0 | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | ALT0 | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+--B Plus--+---+------+---------+-----+-----+
```

notro, 2015-11-24:
I just connected a touch display and I got interrupts (varying number per touch), but the device didn't show up in evtest (no /dev/input event0). So I connected a keyboard and voila the touch device suddenly appeared along side the keyboard in evtest. When I removed the keyboard the touch panel was still there.

UncleGollum, 2015-11-25:
Still nothing

Connect USB keyboard, touchscreen became "event2" (0 is keys, 1 toucharea of keyboard).
Evtest work perfect with 0 and 1 and nothing from 2
```
crw-rw---T 1 root input 13, 64 Jan  1  1970 event0
crw-rw---T 1 root input 13, 65 Jan  1  1970 event1
crw-rw---T 1 root input 13, 66 Jan  1  1970 event2
crw-rw---T 1 root input 13, 63 Jan  1  1970 mice
crw-rw---T 1 root input 13, 32 Jan  1  1970 mouse0
crw-rw---T 1 root input 13, 33 Jan  1  1970 mouse1
lrwxrwxrwx 1 root root       6 Jan  1  1970 touchscreen -> event2
```

UncleGollum, 2016-03-22:
Still nothing, but one more thing:
As described in many manuals in internet, i trying enable waveshare32b overlay, and PI hangs

closed 2017-02-22 by notro