notro/fbtft #345: fbtft_device and ads7846_device not creating /dev file

Patosga, 2015-11-23:
Cheers,
I have one of those cheap lcd screens with a XPT2046 and a ili9341, which I bought from ebay.

Whenever I run modprobe for ads7846_device or fbtft_device I get everything ok in the log, except the last messages. I expect to see a message of device created like the wiki examples but I get nothing.

Im using a RPi2 model b and I'm using the spi-bcm2708 as a spi driver.

Command:
insmod ads7846_device.ko gpio_pendown=16 verbose=3 cs=1 speed=2000000 keep_vre
f_on=1 pressure_max=255 x_plate_ohms=60 x_min=250 x_max=3780 y_min=160 y_mac=393
0 model=7843

Output

[ 1124.079509] ads7846_device: ads7846_device_init()
[ 1124.087319] ads7846_device: SPI devices registered:
[ 1124.092278] ads7846_device:    spidev spi0.1 500kHz 8 bits mode=0x00
[ 1124.098631] ads7846_device:    fb_ili9341 spi0.0 16000kHz 8 bits mode=0x00
[ 1124.105564] ads7846_device:
[ 1124.108360] ads7846_device: Settings:
[ 1124.112093] ads7846_device:   model = 7843
[ 1124.116185] ads7846_device:   gpio_pendown = 16
[ 1124.120778] ads7846_device:   swap_xy = 0
[ 1124.124783] ads7846_device:   x_min = 250
[ 1124.128785] ads7846_device:   x_max = 3780
[ 1124.132951] ads7846_device:   y_min = 160
[ 1124.136957] ads7846_device:   y_max = 4095
[ 1124.141115] ads7846_device:   x_plate_ohms = 60
[ 1124.145640] ads7846_device:   pressure_min = 0
[ 1124.150075] ads7846_device:   pressure_max = 255
[ 1124.154750] ads7846_device:   keep_vref_on = 1
[ 1124.159188] ads7846_device:   vref_delay_usecs = 0
[ 1124.164041] ads7846_device:   vref_mv = 0
[ 1124.168045] ads7846_device:   settle_delay_usecs = 0
[ 1124.173073] ads7846_device:   penirq_recheck_delay_usecs = 0
[ 1124.178724] ads7846_device:   y_plate_ohms = 0
[ 1124.183229] ads7846_device:   debounce_max = 0
[ 1124.187667] ads7846_device:   debounce_tol = 0
[ 1124.192193] ads7846_device:   debounce_rep = 0
[ 1124.196646] ads7846_device: Deleting spi0.1
[ 1124.201351] ads7846_device: SPI devices registered:
[ 1124.206258] ads7846_device:    fb_ili9341 spi0.0 16000kHz 8 bits mode=0x00
[ 1124.213208] ads7846_device:    ads7846 spi0.1 2000kHz 8 bits mode=0x00
[ 1124.219729] ads7846_device:
#
Command for fbtft_device:
modprobe fbtft_device custom name=fb_ili9341 gpios=reset:25,dc:24,led:18 speed
=16000000 bgr=1 rotate=90

Output:
[ 1472.919417] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[ 1472.930755] fbtft_device:  SPI devices registered:
[ 1472.935578] fbtft_device:      ads7846 spi0.1 2000kHz 8 bits mode=0x00
[ 1472.942217] fbtft_device:  'fb' Platform devices registered:
[ 1472.947886] fbtft_device:      bcm2708_fb id=-1 pdata? no
[ 1472.953603] fbtft_device:  GPIOS used by 'fb_ili9341':
[ 1472.958757] fbtft_device:    'reset' = GPIO25
[ 1472.963209] fbtft_device:    'dc' = GPIO24
[ 1472.967306] fbtft_device:    'led' = GPIO18
[ 1472.971552] fbtft_device:  SPI devices registered:
[ 1472.976347] fbtft_device:      ads7846 spi0.1 2000kHz 8 bits mode=0x00
[ 1472.982930] fbtft_device:      fb_ili9341 spi0.0 16000kHz 8 bits mode=0x00
# 

Like a said above i can not see any indication of a file/device created in /dev. The examples above are some of the combinations I tried... Without any information from the driver I can not diagnose what is happening.

Thanks in advance.

notro, 2015-11-23:
> ads7846_device model=7843

ads7846_device only adds a spi device and with this model you have to load the module manually because there's no alias for it: modprobe ads7846

fbtft_device also just adds a device, but the module should load automatically. Try: modprobe fb_ili9341

Patosga, 2015-11-23:
Thanks a lot notro! Now I can see the devices alive.

In the touchscreen whenever I do cat and touch the screen I can see random data (probably because it is not ascii) being printed.
I will try to implement tslib tomorrow or when I get the frame buffer to work and report back.

Although, with the ili9341 I'm only getting a white screen, even when I cat a huge file to the frame buffer.

The commands I'm doing for the screen is the same as the previous post. I only added the modprobe fb_ili9341.

Nontheless, I will post the log here:

Commands:

# modprobe spi-bcm2708
# modprobe fbtft_device custom name=fb_ili9341 gpios=reset:25,dc:24,led:18 speed
=16000000 bgr=1 rotate=90
# modprobe fb_ili9341

Log:

[   61.674756] bcm2708_spi bcm2708_spi.0: master is unqueued, this is deprecated
[   61.683285] bcm2708_spi bcm2708_spi.0: SPI Controller at 0x3f204000 (irq 80)
[   71.625695] fbtft: module is from the staging directory, the quality is unknown.
[   71.640729] fbtft_device: module is from the staging directory, the quality is .
[   71.652018] fbtft_device:  SPI devices registered:
[   71.656841] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[   71.663270] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   71.669631] fbtft_device:  'fb' Platform devices registered:
[   71.675461] fbtft_device:      bcm2708_fb id=-1 pdata? no
[   71.680944] fbtft_device: Deleting spi0.0
[   71.685748] fbtft_device:  GPIOS used by 'fb_ili9341':
[   71.690959] fbtft_device:    'reset' = GPIO25
[   71.695333] fbtft_device:    'dc' = GPIO24
[   71.699426] fbtft_device:    'led' = GPIO18
[   71.703638] fbtft_device:  SPI devices registered:
[   71.708434] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   71.714832] fbtft_device:      fb_ili9341 spi0.0 16000kHz 8 bits mode=0x00
[   81.178209] fb_ili9341: module is from the staging directory, the quality is un.
[   81.570767] graphics fb1: fb_ili9341 frame buffer, 320x240, 150 KiB video memorz

I checked the connections 3 times and tried without the rotate 90.
All I get is a white screen :(

Any ideia at all is very very welcome.

Thanks a lot for the help so far.

notro, 2015-11-23:
You can use evtest to test input devices.
A white display means it's not initialized. It should be black (pixels are zero). The reason for this is usually a wiring problem or that the default init sequence doesn't match the display.

Source: http://lxr.free-electrons.com/source/drivers/staging/fbtft/fb_ili9341.c

Patosga, 2015-11-24:
Again, thanks a lot.

I followed your suggestion and used evtest. Now I can see ABS_X ABS_Y and pressure value every time I click.

Tomorrow I will go deeper with the ili9341 and I will update this post

Patosga, 2015-11-24:
I tried again today and the display worked the first time I tried... Maybe there was something wrong with the wiring.

I got everything to work with tslib.

Many thanks for the help notro.

closed 2015-11-24 by Patosga