notro/fbtft #347: Can't configure noname dispay

AlexEdunov, 2015-12-03:
I use [noname display from AliExpress](http://www.aliexpress.com/item/Best-Price-Original-3-5-LCD-TFT-Touch-Screen-Display-for-Raspberry-Pi-2-Model-B/32443379727.html).

I followed [this tutorial](http://www.circuitbasics.com/setup-lcd-touchscreen-raspberry-pi/) and I didn't get the desired result. My display is still white.

As I understood, "supported display" means that you know how to configure fbtft_device_display structure, in other words all the display will work if the configuration is correct.

    /* Supported displays in alphabetical order */
    static struct fbtft_device_display displays[] = {
        ...
    }

So, can I configure it manually and how to find out the correct values? On the product description page there is some information on the appointment of pins.

Or maybe I just need to use correct value of fbtft_device name?

FreshXOpenSource, 2015-12-23:
I looks like you have the same display as in issue #340

See https://github.com/notro/fbtft/issues/340

FreshXOpenSource, 2016-01-06:
After further research we got the display partially running using the following values extracted from the SPI dmesg

```SP=10000000
modprobe fbtft_device debug=3  name=flexfb speed=${SP} gpios=reset:25,dc:24
modprobe flexfb width=480 height=320 buswidth=8 init=-1,0xB0,0x80,-1,0xC0,0x0A,0x0A,-1,0xC1,0x45,0x07,-1,0xC2,0x33,-1,0xC5,0x00,0x42,0x80,-1,0xB1,0xD0,0x11,-1,0xB4,0x02,-1,0xB6,0x00,0x02,0x3B,-1,0xB7,0x07,-1,0x36,0x2B,-1,0xF0,0x36,0xA5,0xD3,-1,0xE5,0x80,-1,0xE5,0x01,-1,0xB3,0x00,-1,0xE5,0x00,-1,0xF0,0x36,0xA5,0x53,-1,0xE0,0x00,0x35,0x33,0x00,0x00,0x00,0x00,0x35,0x33,0x00,0x00,0x00,-1,0x3A,0x55,-1,0x11,-2,250,-1,0x29,-3
```

The same display seems to be described here 

https://www.raspberrypi.org/forums/viewtopic.php?p=433719

Unfortunately our version flickers and does not seem to get the correct TFT_VERTICAL :
- using width=480 height=320 results in a distorted image with flickering
- using width=320 height=480 results in flickering 

Any ideas how to proceed from here?

Kernel is 4.0.7-v7+ from notro, dmesg :

[ 4325.895858] flexfb: module is from the staging directory, the quality is unknown, you have been warned.
[ 4325.897573] flexfb spi0.0: fbtft_request_gpios: 'reset' = GPIO25
[ 4325.897613] flexfb spi0.0: fbtft_request_gpios: 'dc' = GPIO24
[ 4325.897634] flexfb spi0.0: flexfb_verify_gpios_dc()
[ 4325.897653] flexfb spi0.0: fbtft_init_display()
[ 4325.897671] flexfb spi0.0: fbtft_reset()
[ 4326.017884] flexfb spi0.0: init: write(0xB0) 0x80
[ 4326.018032] flexfb spi0.0: init: write(0xC0) 0x0A 0x0A
[ 4326.018136] flexfb spi0.0: init: write(0xC1) 0x45 0x07
[ 4326.018240] flexfb spi0.0: init: write(0xC2) 0x33
[ 4326.018332] flexfb spi0.0: init: write(0xC5) 0x00 0x42 0x80
[ 4326.018421] flexfb spi0.0: init: write(0xB1) 0xD0 0x11
[ 4326.018508] flexfb spi0.0: init: write(0xB4) 0x02
[ 4326.018597] flexfb spi0.0: init: write(0xB6) 0x00 0x02 0x3B
[ 4326.018686] flexfb spi0.0: init: write(0xB7) 0x07
[ 4326.018774] flexfb spi0.0: init: write(0x36) 0x08
[ 4326.018865] flexfb spi0.0: init: write(0xF0) 0x36 0xA5 0xD3
[ 4326.018955] flexfb spi0.0: init: write(0xE5) 0x80
[ 4326.019043] flexfb spi0.0: init: write(0xE5) 0x01
[ 4326.019131] flexfb spi0.0: init: write(0xB3) 0x00
[ 4326.019221] flexfb spi0.0: init: write(0xE5) 0x00
[ 4326.019313] flexfb spi0.0: init: write(0xF0) 0x36 0xA5 0x53
[ 4326.019413] flexfb spi0.0: init: write(0xE0) 0x00 0x35 0x33 0x00 0x00 0x00 0x00 0x35 0x33 0x00 0x00 0x00
[ 4326.019516] flexfb spi0.0: init: write(0x3A) 0x55
[ 4326.019605] flexfb spi0.0: init: write(0x11)
[ 4326.019657] flexfb spi0.0: init: mdelay(250)
[ 4326.269702] flexfb spi0.0: init: write(0x29)
[ 4326.589681] flexfb spi0.0: Display update: 937 kB/s (319.879 ms), fps=0 (0.000 ms)
[ 4326.589702] flexfb spi0.0: fbtft_register_backlight()
[ 4326.589714] flexfb spi0.0: fbtft_register_backlight(): led pin not set, exiting.
[ 4326.592557] graphics fb1: flexfb frame buffer, 320x480, 300 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 8 MHz

closed 2016-04-28 by notro