notro/fbtft #350: flexfb using DMA?.

Ferite, 2015-12-09:
Hi.

I have a SpotPear 3.5" display working with Raspberry 2 (Linux raspberrypi 3.18.9-v7+). The driver is loaded from /etc/modules this way:

spi-bcm2835
flexfb width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
fbtft_device debug=3 rotate=90 name=flexfb speed=48000000 gpios=reset:25,dc:24 txbuflen=32768 fps=50

It seems howerver that it is not using DMA (dmesg shows "[ 5.919606] fbtft_device: bcm2708_fb id=-1 pdata? no"). My questions are:

Can flexfb use DMA?. If so, How can I force flexfb to use spi-bcm2835 and DMA?

Thanks.

notro, 2015-12-10:
> Linux raspberrypi 3.18.9-v7+

You have to update the kernel to 4.1.6 or later to get spi-bcm2835 dma support.
flexfb will also trigger dma transfers. The spi driver takes care of that.

Ferite, 2015-12-15:
Hi.

Thanks. There is a dramatic improvement with DMA. By the way, I was wondering if it is the same setting the default display to fb1 in 99-fbturbo.conf or fbtft.conf. The only difference I can find is that fbtft.conf does not show the "Module g2d_23 not found" warning.

Thanks.

notro, 2015-12-15:
> setting the default display to fb1 in 99-fbturbo.conf

This is what I do when I test displays.
This explains the warning: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=67334

closed 2016-04-28 by notro