notro/fbtft #351: Overlay from fb_ili9340.c

ChainedHope, 2015-12-10:
Is there a way to use a modified fb_ili9340.c with a corresponding -overlay.dtb file? I changed fb_ili9340.c to support the 3.5'' pitft's resolution and im able to get it working by using the commands:

sudo modprobe fbtft_device name=pitft
sudo FRAMEBUFFER=/dev/fb1 startx

but id like to be able to use an -overlay.dtb file in the config.txt and change the 99-fbturbo.conf to use framebuffer 1 instead of framebuffer 0. I figured that the default pitft28-resisitve-overlay.dtb would automatically use the compiled driver and be good to go but it doesnt work.

Any help?

notro, 2015-12-11:
If you are referring to the adafruit 3.5" pitft, this is the overlay they use: https://github.com/adafruit/Adafruit-Pi-Kernel-o-Matic/blob/pitft/pitft35r-overlay.dts

ChainedHope, 2015-12-13:
I was actually asking because I can use the pitft28-resistive-overlay.dtb file included with Raspbian to run the screen at 240x320 so I figured I could change the driver file to 320x480 and use the included overlay. Guess thats not a possibility?

notro, 2015-12-14:
Are you saying that you want to run the 2.8" screen at 320x480?

ChainedHope, 2016-01-02:
https://github.com/notro/fbtft/wiki/LCD-Shields
It says there under the PiTFT 3.5'' that you can change the driver to work with it since its unsupported. I have changed it and recompiled the kernel. I want to use the overlay file that is attached to the driver which is the pitft28-resistive-overlay.dtb.

I can use the commands above to get it to run, but cant use the overlay. The question is why I cant use the overlay.

notro, 2016-01-02:
You could try the overlay adafruit has provided (pitft35r-overlay.dts): https://github.com/adafruit/Adafruit-Pi-Kernel-o-Matic/tree/pitft
There are precompiled dtbs in the boot/overlays folder

closed 2016-04-28 by notro