notro/fbtft #352: Unable to make it permanent in jessie

barartm, 2015-12-12:
Hello, when i do:

sudo modprobe flexfb nobacklight regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3 width=480 height=320
sudo modprobe fbtft_device name=flexfb speed=16000000 gpios=reset:25,dc:24

My display works, however when i reboot my rpi2 it seems that is unable to load the driver or something at boot:

pi@raspberrypi ~ $ dmesg | grep graphics
pi@raspberrypi ~ $ dmesg | grep spi
[    4.661316] spi spi0.0: setting up native-CS0 as GPIO 8
[    4.674243] spi spi0.1: setting up native-CS1 as GPIO 7
[    4.704807] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[    4.704817] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[    4.704937] fbtft_device: Deleting spi0.0
[    4.705458] spi spi0.0: setting up native-CS0 as GPIO 8
[    4.705674] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[    4.705682] fbtft_device:      flexfb spi0.0 16000kHz 8 bits mode=0x00
pi@raspberrypi ~ $ con2fbmap 1 1
ioctl FBIOPUT_CON2FBMAP: Invalid argument
pi@raspberrypi ~ $ 

My /etc/modules-load.d/fbtft.conf has this:

spi-bcm2835
fbtft_device

And My /etc/modprobe.d/fbtft.conf has this:

options flexfb regwidth=16 nobacklight init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3 width=480 height=320
options fbtft_device name=flexfb speed=16000000 gpios=reset:25,dc:24

How can i make it permanent on jessie? It seems that following the instructions on wiki doesnt work for me

Here are more information of my system:

pi@raspberrypi ~ $ cat /etc/debian_version
8.0
pi@raspberrypi ~ $ lsb_release -da
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie

pi@raspberrypi ~ $ uname -r
4.1.13-v7+

notro, 2015-12-12:
Does it help to add flexfb to /etc/modules-load.d/fbtft.conf?
```
spi-bcm2835
flexfb
fbtft_device
```

barartm, 2015-12-13:
Thats it! is working perfectly :smiley: Thanks!

I want to rotate the display 180, i changed the value fbtft_device.rotate=180 in /boot/cmdline.txt but it does not make a difference, if i try 90 or 270 it show the display (without rotation) but only like 3/4 of the screen and the other 1/4 show only a white bar, do i missing something?

notro, 2015-12-13:
flexfb doesn't have support for rotation. fbtft uses the rotate value to flip width/height if 90 or 270, but the controller setup is left to the driver.
Register 0x36 is usually responsible for rotation/flip/mirror/bgr: ```-1,0x36,0x28```
Look in the controller datasheet to find the value.
If you don't know the controller, you can look at the other fbtft drivers which sets 0x36 in set_par() and try those values. The MIPI compatible controllers have this register in common I think.

closed 2016-04-28 by notro