notro/fbtft #353: Errors loading fbtft on the Intel Edison

ShawnHymel, 2015-12-24:
I managed to get fbtft compiled with the Edison kernel using the Yocto Project and [wallacezq's recipe](https://github.com/wallacezq/fbtft-yocto-recipe). I have an [Adafruit 1.8" LCD](https://www.adafruit.com/products/358) connected to the Edison (SPI bus 5.0). I try to load the module with

    modprobe fbtft dma=0
    modprobe fbtft_device name=sainsmart18 busnum=5 speed=5000000 bgr=1 gpios=reset:44,dc:45

but `dmesg` gives me the following:

```
[   63.099949] fbtft_device:  SPI devices registered:
[   63.099982] fbtft_device:      ads7955 spi5.0 25000kHz 16 bits mode=0x00
[   63.100002] fbtft_device:      spidev spi5.1 25000kHz 8 bits mode=0x00
[   63.100016] fbtft_device:  'fb' Platform devices registered:
[   63.100106] fbtft_device: Deleting spi5.0
[   63.100904] BUG: sleeping function called from invalid context at /media/Project/Yocto/edison-src/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/3.10.17-r0/linux/mm/slub.c:926
[   63.101049] in_atomic(): 1, irqs_disabled(): 1, pid: 292, name: modprobe
[   63.101118] CPU: 1 PID: 292 Comm: modprobe Tainted: G           O 3.10.17-yocto-standard #5
[   63.101133] Hardware name: Intel Corporation Merrifield/BODEGA BAY, BIOS 542 2015.01.21:18.19.48
[   63.101147]  f5640590 00000124 f5197cf0 c18ca73a f5197d14 c126d689 c1ac832c 00000001
[   63.101202]  00000001 00000124 f5640890 f5d55284 00000000 f5197d48 c13203d9 f5197d38
[   63.101252]  c18d33bd c18cfab6 c15c7014 00000020 f5d5532c 000080d0 f5197d48 f5d55284
[   63.101303] Call Trace:
[   63.101342]  [<c18ca73a>] dump_stack+0x16/0x18
[   63.101371]  [<c126d689>] __might_sleep+0xe9/0x110
[   63.101401]  [<c13203d9>] kmem_cache_alloc_trace+0x39/0x1a0
[   63.101429]  [<c18d33bd>] ? add_preempt_count+0x7d/0xf0
[   63.101453]  [<c18cfab6>] ? _raw_spin_unlock_irqrestore+0x26/0x60
[   63.101479]  [<c15c7014>] ? setup+0x5d4/0x840
[   63.101504]  [<c15c7014>] setup+0x5d4/0x840
[   63.101530]  [<c1270ddb>] ? get_parent_ip+0xb/0x40
[   63.101555]  [<c18d34c5>] ? sub_preempt_count+0x95/0xf0
[   63.101581]  [<c15c00fa>] spi_setup+0x4a/0x120
[   63.101609]  [<c1586650>] ? klist_devices_get+0x20/0x20
[   63.101634]  [<c1586507>] ? bus_find_device+0x67/0x90
[   63.101660]  [<c15c0301>] spi_add_device+0xe1/0x1d0
[   63.101688]  [<c15c0eff>] spi_new_device+0x7f/0xc0
[   63.101725]  [<f842c492>] fbtft_device_init+0x492/0x1000 [fbtft_device]
[   63.101754]  [<c1319cac>] ? __vunmap+0x7c/0xf0
[   63.101780]  [<f842c000>] ? 0xf842bfff
[   63.101805]  [<c120017a>] do_one_initcall+0xaa/0x170
[   63.101828]  [<c120017a>] ? do_one_initcall+0xaa/0x170
[   63.101856]  [<c18c8967>] ? set_section_ro_nx+0x54/0x59
[   63.101885]  [<c129e443>] load_module+0x1113/0x16c0
[   63.101912]  [<c132b015>] ? kernel_read+0x35/0x50
[   63.101948]  [<c129eb6d>] SyS_finit_module+0x7d/0xc0
[   63.101975]  [<c130661b>] ? vm_mmap_pgoff+0x7b/0xa0
[   63.102007]  [<c18d0178>] syscall_call+0x7/0xb
[   63.102034] spi spi5.0: setting default chip values
[   63.102081] ------------[ cut here ]------------
[   63.102108] WARNING: at /media/Project/Yocto/edison-src/out/linux64/build/tmp/work/edison-poky-linux/linux-yocto/3.10.17-r0/linux/kernel/softirq.c:160 local_bh_enable_ip+0x98/0xe0()
[   63.102120] Modules linked in: fbtft_device(O+) fbtft(O) fb_sys_fops syscopyarea sysfillrect sysimgblt usb_f_acm u_serial g_multi libcomposite bcm_bt_lpm bcm4334x(O)
[   63.102208] CPU: 1 PID: 292 Comm: modprobe Tainted: G           O 3.10.17-yocto-standard #5
[   63.102222] Hardware name: Intel Corporation Merrifield/BODEGA BAY, BIOS 542 2015.01.21:18.19.48
[   63.102234]  c1ac6794 00000000 f5197c68 c18ca73a f5197c90 c12408be c1acfc6b c1ac6794
[   63.102285]  000000a0 c1248c98 c1248c98 c1522eb1 00000000 f6c4b024 f5197ca0 c1240982
[   63.102334]  00000009 00000000 f5197cac c1248c98 f6c4b0a4 f5197cb4 c18cfa7f f5197cf0
[   63.102384] Call Trace:
[   63.102413]  [<c18ca73a>] dump_stack+0x16/0x18
[   63.102442]  [<c12408be>] warn_slowpath_common+0x5e/0x80
[   63.102467]  [<c1248c98>] ? local_bh_enable_ip+0x98/0xe0
[   63.102491]  [<c1248c98>] ? local_bh_enable_ip+0x98/0xe0
[   63.102519]  [<c1522eb1>] ? intel_mid_dma_alloc_chan_resources+0x71/0x1a0
[   63.102545]  [<c1240982>] warn_slowpath_null+0x22/0x30
[   63.102570]  [<c1248c98>] local_bh_enable_ip+0x98/0xe0
[   63.102594]  [<c18cfa7f>] _raw_spin_unlock_bh+0x1f/0x30
[   63.102620]  [<c1522eb1>] intel_mid_dma_alloc_chan_resources+0x71/0x1a0
[   63.102645]  [<c18bb39a>] ? klist_put+0x3a/0x90
[   63.102672]  [<c152075b>] dma_chan_get+0x3b/0xc0
[   63.102697]  [<c15215c5>] __dma_request_channel+0x195/0x370
[   63.102722]  [<c1501668>] ? pci_get_subsys+0x38/0x40
[   63.102745]  [<c15c5aa0>] ? u32_reader+0x60/0x60
[   63.102770]  [<c15c6ec9>] setup+0x489/0x840
[   63.102796]  [<c1270ddb>] ? get_parent_ip+0xb/0x40
[   63.102822]  [<c18d34c5>] ? sub_preempt_count+0x95/0xf0
[   63.102849]  [<c15c00fa>] spi_setup+0x4a/0x120
[   63.102876]  [<c1586650>] ? klist_devices_get+0x20/0x20
[   63.102901]  [<c1586507>] ? bus_find_device+0x67/0x90
[   63.102927]  [<c15c0301>] spi_add_device+0xe1/0x1d0
[   63.102954]  [<c15c0eff>] spi_new_device+0x7f/0xc0
[   63.102990]  [<f842c492>] fbtft_device_init+0x492/0x1000 [fbtft_device]
[   63.103018]  [<c1319cac>] ? __vunmap+0x7c/0xf0
[   63.103042]  [<f842c000>] ? 0xf842bfff
[   63.103066]  [<c120017a>] do_one_initcall+0xaa/0x170
[   63.103089]  [<c120017a>] ? do_one_initcall+0xaa/0x170
[   63.103117]  [<c18c8967>] ? set_section_ro_nx+0x54/0x59
[   63.103145]  [<c129e443>] load_module+0x1113/0x16c0
[   63.103170]  [<c132b015>] ? kernel_read+0x35/0x50
[   63.103205]  [<c129eb6d>] SyS_finit_module+0x7d/0xc0
[   63.103232]  [<c130661b>] ? vm_mmap_pgoff+0x7b/0xa0
[   63.103264]  [<c18d0178>] syscall_call+0x7/0xb
[   63.103282] ---[ end trace b72fd31158b70fba ]---
[   63.127052] fbtft_device:  GPIOS used by 'sainsmart18':
[   63.127075] fbtft_device:    'reset' = GPIO44
[   63.127090] fbtft_device:    'dc' = GPIO45
[   63.127102] fbtft_device:  SPI devices registered:
[   63.127121] fbtft_device:      spidev spi5.1 25000kHz 8 bits mode=0x00
[   63.127138] fbtft_device:      fb_st7735r spi5.0 5000kHz 8 bits mode=0x00
[   63.265662] intel_mid_ssp_spi_unified 0000:00:07.1: ERROR : DMA buffers already mapped
```

I do not get a /dev/fb? device, so it seems the module is running into an issue with DMA buffers. As far as I can tell, between the recipe's patches and `dma=0`, I should be disabling DMA on fbtft, but it does not seem to be the case. I do not want DMA because as far as I know, the Edison does not have DMA on SPI.

How do I prevent fbtft from using DMA? Any help would be appreciated!

pdp7, 2015-12-28:
@ShawnHymel I ran into a similar problem with the BeagleBone Black.  @notro suggested this change to disable DMA: https://github.com/notro/fbtft/issues/337#issuecomment-153534170

Robert C Nelson has now created a flag in SPI driver used by the BeagleBone Black: https://github.com/RobertCNelson/bb-kernel/blob/cef1661f6d27847b7a1f37ab6db2aec397e6388c/patches/beaglebone/dts/0007-spi-omap2-mcspi-ti-pio-mode.patch

ShawnHymel, 2015-12-29:
@pdp7 Thanks! It looks like DMA_MIN_BYTES is in spi-omap2-mcspi, which I don't believe exists for the Edison source. I'll see if I can find something similar.

sukeshak, 2016-01-06:
Hi @ShawnHymel 
Any update on your investigation? Did you get it working?
I am in the same boat and trying to get 128x128 1.5" using SSD1351 to work.

ShawnHymel, 2016-01-09:
No dice. I've posted this question on the Intel forums. Apparently, DMA has been enabled on SPI for the Edison. I tried disabling it in a recent kernel build, but I'm still getting the same error. I'm betting I'm not turning off DMA correctly,  but I haven't seen anything on which options to disable in the kernel.

sukeshak, 2016-01-14:
Thanks @ShawnHymel for the reply. In case I get lucky, will update here... Fingers crossed :)

andy-shev, 2017-01-02:
@ShawnHymel, check my **eds** branch. It has now all patches I developed recently to support SPI/DMA for Edison on vanilla kernel. I checked my stuff with Adafruit 2.8" and SparkFun OLED. *fbtft* related stuff has been submitted today to public mailing list, Cc'ing @notro.

ShawnHymel, 2017-01-06:
@andy-shev Awesome, thanks!

andy-shev, 2021-08-16:
It can be closed as "Fixed upstream". The kernels (approx. from v5.10.y) works out of the box on Edison WRT fbtft driver. Yeah, there were fixes here and there, but one should use stable kernels.

ShawnHymel, 2021-08-16:
Fixed upstream. Thanks, @andy-shev !

closed 2021-08-16 by ShawnHymel