notro/fbtft #361: waveshare spotpear 4 inch on raspberry pi 2

K33F, 2016-01-13:
Hi. i have try every tutorial on the net to solve my problem but didnt work, i have a pi 2 with raspbian, ive done the raspi-config to activate the SPI, i downloaded and install the notro firmware, did every update, did a sudo modprobe flexfb regwidth=16 nobacklight init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3 width=480 height=320
and
sudo modprobe fbtft_device name=flexfb speed=16000000 gpios=reset:25,dc:24
the screen became black, so ive writing this setting in /etc/modules, after that ive write in /boot/cmdline.txt at the end fbcon=map:1 fbcon=font:ProFont6x11, change in the sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf fb0 to fb1, did a restart and the screen always stays white, ive changed again the fb1 to fb0 tried the modprobe on the line of flexfb and the screen became black , did a con2fbmap 1 1 to put it on the lcd didnt work, what i am missing here anybody have a clue?

notro, 2016-01-15:
See https://github.com/notro/fbtft/wiki#make-it-permanent-debian-jessie-arch-linux-systemd

K33F, 2016-01-23:
The issue was cause because of noobs, when i download and install raspbian from the official site, the screen work with the same configuration thnx for the reply hope this help someone else thnx

closed 2016-01-23 by K33F