notro/fbtft #362: Bootsplash on Jessie

franckyg, 2016-01-13:
Hi,

I'm trying to get bootsplash on raspbian Jessie for 2 days without success.

I followed this : https://github.com/notro/fbtft/wiki/Bootsplash

Replacing spi_bcm2708 by spi_2835.

But I can't get plymouth-drm, is no more available with Jessie. So I continued with plymouth.

My tft is working, I can see picture with fbi, video with mplayer. Driver is correctly load automatically (I can see backlight turning on when starting) but no splashscreen.

Any idea to help me ?
Thanks
Franck

Krakawet, 2016-02-05:
Try to focus first on making plymouth working, you can run it from the console for testing purpose.
Once you've done with plymouth, work on it's loading at boot !
If it's correctly launched and still not displaying, try check if the correct screen is selected at boot (the fbcon map10 thing). 

You may make it work in another way :
Use HDMI as output, and duplicating the display on the touchscreen with fbcp.
For early splashscreen/bootsplash with screen mirroring, take at look at technico.top

alidaf, 2016-11-22:
I'm in the same boat. The "Bootsplash" wiki page is out of date by referring to spi_bcm2708. When you say that it can be tested by running from the console, does that mean that a keyboard must be attached, or can the console output be triggered on the display by typing the commands via ssh?

Krakawet, 2016-12-10:
Sorry alidaf, I can't say how it would work through ssh. 
But running "sudo plymouth-set-default-theme joy" can give you clue, even on ssh.

I've manage to do this under Jessie, here's some of my implied files (adapt to your own cases). 

/etc/initramfs-tools/modules : 
```
spi_bcm2835
fbtft
fb_s6d02a1

```
/boot/cmdline.txt
`logo.nologo splash plymouth.ignore-serial-consoles dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait`

there is no more need of the "spi" executable described in the bootsplash wiki, nor any other initramfs script (at least for my screens !)

alidaf, 2016-12-12:
Thanks for that Krakawet. Still no joy here so I've abandoned it for now. I've submitted a bug report to Debian so I'll wait to hear from that. I need to move on and start on other aspects of the project that have been held up by this.

pozzugno, 2017-08-07:
I'm in the same situation. I'd like to have a splash screen on PiTFT early during boot process.

So do you confirm I don't need to install plymouth-drm (that isn't available on jessie)?

kjyv, 2017-08-29:
I think you might have an issue with this change in raspian: https://github.com/raspberrypi-ui/plymouth/commit/dfb02aa5daa7193398d0801a92d63f78d08295a4
You can check for the message in /var/log/plymouth.log if you add plymouth.debug to /boot/cmdline.txt. I rebuilt the deb package to change the code back to original and got a splash screen on /dev/fb1

closed 2021-08-24 by github-actions[bot]