notro/fbtft #364: Help with ssd1306 and Device Trees

aleksandarzivkovic, 2016-01-21:
Hello,

for some time already I'm trying to enable "OLED W click" (SPI SSD1306 OLED display) on my Raspberry Pi A+ by using image generated by Buildroot 2015.08. I had it working by using fbtft driver (from https://github.com/notro/fbtft/wiki) via kernel command line (drivers statically built). I was able to see it as /dev/fb1 and happily drawing lines there.
Then I have upgraded to newer kernel (https://github.com/raspberrypi/linux/tree/592503752b6951972f161f04280683c5af38d173) and now I'm unsuccessfully trying to do the same thing with Device Trees mechanism. I'm using device tree overlay from here https://github.com/kenrestivo/linux/blob/3d61a22b05cbf7ac2f6b620dcd45c2fb49fb9442/arch/arm/boot/dts/overlays/adafruit13m-overlay.dts (with adjusted GPIO lines) but can never get driver loaded.
I assume that dtb overlay is successfully loaded because when running command on Raspberry Pi ```dtc -I fs /proc/device-tree``` I can see:
```
...
spi@7e204000 {
		reg = <0x7e204000 0x1000>;
		dmas = <0x6 0x6 0x6 0x7>;
		interrupts = <0x2 0x16>;
		pinctrl-0 = <0x7>;
		compatible = "brcm,bcm2835-spi";
		cs-gpios = <0x0 0x0>;
		clocks = <0x5>;
		status = "okay";
		#address-cells = <0x1>;
		phandle = <0x13>;
		#size-cells = <0x0>;
		dma-names = "tx", "rx";
		pinctrl-names = "default";
		linux,phandle = <0x13>;

		adafruit13m@0 {
				fps = <0x14>;
				reg = <0x0>;
				pinctrl-0 = <0x26>;
				debug = <0x0>;
				compatible = "solomon,ssd1306";
				reset-gpios = <0xd 0x4 0x0>;
				spi-max-frequency = <0x5b8d80>;
				dc-gpios = <0xd 0x12 0x0>;
				phandle = <0x27>;
				buswidth = <0x8>;
				pinctrl-names = "default";
				linux,phandle = <0x27>;
		};

		spidev@0 {
				reg = <0x0>;
				compatible = "spidev";
				spi-max-frequency = <0x7a120>;
				status = "disabled";
				#address-cells = <0x1>;
				#size-cells = <0x0>;
		};

		spidev@1 {
				reg = <0x1>;
				compatible = "spidev";
				spi-max-frequency = <0x7a120>;
				status = "disabled";
				#address-cells = <0x1>;
				#size-cells = <0x0>;
		};
...
```
I have also included debug flags in command line (```debug dyndbg="func of_platform_bus_create +p" dyndbg="func bus_add_driver +p" dyndbg="func device_add +p" dyndbg="func driver_probe_device +p" dyndbg="func really_probe +p" dyndbg="func driver_bound +p" dyndbg="func deferred_probe_work_func +p" dyndbg="func driver_deferred_probe_add +p" dyndbg="func driver_deferred_probe_add +p" dyndbg="func driver_deferred_probe_del +p"```) and can see nothing from ssd1306:
```
# dmesg | grep fb
[    0.141332] device: 'soc:fb': device_add
[    1.381392] device: 'fbcon': device_add
[    2.027190] bus: 'platform': add driver bcm2708_fb
[    2.032061] bus: 'platform': driver_probe_device: matched device soc:fb with driver bcm2708_fb
[    2.040735] bus: 'platform': really_probe: probing driver bcm2708_fb with device soc:fb
[    2.061607] device: 'fb0': device_add
[    2.082578] driver: 'bcm2708_fb': driver_bound: bound to device 'soc:fb'
[    2.089336] bus: 'platform': really_probe: bound device soc:fb to driver bcm2708_fb
[    3.636455] bus: 'spi': add driver fb_ssd1306
[    3.640924] bus: 'platform': add driver fb_ssd1306
```
Here you can see that ```fb0``` device successfully goes through initialization and at last through bounding. But not for ssd1306.
Besides this driver I have added rotary-encoder driver to my RPi and it works correctly so I don't suspect to device tree include mechanism:
```
# dmesg | grep "rotary-encoder"
[    3.217678] bus: 'platform': add driver rotary-encoder
[    3.222886] bus: 'platform': driver_probe_device: matched device soc:encoder with driver rotary-encoder
[    3.232598] bus: 'platform': really_probe: probing driver rotary-encoder with device soc:encoder
[    3.257933] driver: 'rotary-encoder': driver_bound: bound to device 'soc:encoder'
[    3.265427] bus: 'platform': really_probe: bound device soc:encoder to driver rotary-encoder
```

I have no idea where to look at at this point. Can you give me a hint or advice? Is it possible that fbtft doesn't have proper support for device tree?

Regards,
Aleksandar Zivkovic

notro, 2016-01-22:
What about spi-bcm2835? DT spi devices are [registered](http://lxr.free-electrons.com/ident?i=of_register_spi_devices) when the spi master driver is registered.
```
~$ ls -l /sys/devices/platform/soc/3f204000.spi/
total 0
lrwxrwxrwx 1 root root    0 Jan 22 21:08 driver -> ../../../../bus/platform/drivers/spi-bcm2835
-rw-r--r-- 1 root root 4096 Jan 22 21:08 driver_override
-r--r--r-- 1 root root 4096 Jan 22 21:08 modalias
lrwxrwxrwx 1 root root    0 Jan 22 21:08 of_node -> ../../../../firmware/devicetree/base/soc/spi@7e204000
drwxr-xr-x 3 root root    0 Jan 22 21:08 spi_master
lrwxrwxrwx 1 root root    0 Jan 22 21:08 subsystem -> ../../../../bus/platform
-rw-r--r-- 1 root root 4096 Jan 22 21:08 uevent

~$ ls -l /sys/bus/spi/devices/spi*
lrwxrwxrwx 1 root root 0 Jan 22 21:05 /sys/bus/spi/devices/spi0.0 -> ../../../devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.0
lrwxrwxrwx 1 root root 0 Jan 22 21:05 /sys/bus/spi/devices/spi0.1 -> ../../../devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1
```

aleksandarzivkovic, 2016-01-22:
Yes, that helps a lot!
Actually in my kernel configuration SPI_BCM2835 was not enabled. Only SPI_BCM2708. When I enabled SPI_BCM2835 I immediately got ```/dev/fb1``` which I missed so much. It's the confusion for me about Broadcom chip naming and corresponding kernel driver naming.
Still don't have lines on my screen but will figure out myself from this point on.
You helped a lot. Thank you very much!

closed 2016-01-22 by aleksandarzivkovic