notro/fbtft #365: Boot up (systemd) fails to load modules for Waveshare 3.5 Spotpear, but manual modprobe works

alexbarcelo, 2016-01-23:
I don't understand what is happening. I have read #215 and found a lot of good advice, but I have the following problem:

`systemd-modules-load.service` fails. The `/etc/modules` contains the following:

    flexfb width=480 height=320 regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3

    fbtft_device debug=3 rotate=0 name=flexfb speed=16000000 gpios=reset:25,dc:24

And the `systemd` error (found through the syslog):

    Jan 23 21:41:22 touchberrypi systemd-modules-load[85]: Failed to find module 'flexfb width=480 height=320 regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3'
    Jan 23 21:41:22 touchberrypi systemd-modules-load[85]: Failed to find module 'fbtft_device debug=3 rotate=0 name=flexfb speed=16000000 gpios=reset:25,dc:24'

However, I can just login through ssh and do:

    $ sudo modprobe flexfb width=480 height=320 regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
    $ sudo modprobe fbtft_device debug=3 rotate=0 name=flexfb speed=16000000 gpios=reset:25,dc:24

and everything works. I have fiddled with adding the SPI module in the modprobe, but it has no impact (I have already enabled it through the `raspi-config`). I started with the official Raspberry Raspbian Jessy Lite image and I have updated the rpi-firmware (because of #215 trend).

I am almost ready to do some dirty patching at rc.local, but I would like to set it up "the right way". Unfortunately, I am stuck. I don't understand the difference between the `/etc/modules` file and the `modprobe` command approach.

notro, 2016-01-23:
The module arguments goes into a file of it's own: https://github.com/notro/fbtft/wiki#make-it-permanent-debian-jessie-arch-linux-systemd

alexbarcelo, 2016-01-24:
Ok, now I'm ashamed for not reading thoroughly the wiki :( sorry about that, abused Google and failed to read first.

However, I have still some issue between `systemd` and regular modprobe. I have put in `/etc/modprobe.d/fbtft.conf`:

    options flexfb width=480 (...)
    options fbtft_device name=flexfb debug=3 rotate=0 speed=16000000 gpios=reset:25,dc:24

And, the `/etc/modules-load.d/fbtft.conf` contains:

    spi-bcm2835
    flexfb
    fbtft_device

The `systemd-modules-load` shows the following (in the syslog):

    Jan 24 08:31:51 touchberrypi systemd-modules-load[86]: Inserted module 'spi_bcm2835'
    Jan 24 08:31:51 touchberrypi systemd-modules-load[86]: Inserted module 'flexfb'
    Jan 24 08:31:51 touchberrypi systemd-modules-load[86]: Failed to insert 'fbtft_device': Invalid argument
    Jan 24 08:31:51 touchberrypi systemd-modules-load[86]: Inserted module 'i2c_dev'

(the last one is in /etc/modules and is unrelated to all this)

The kernel message are in the syslog, a bit further down, the related to fbtft_device are:

    Jan 24 08:31:51 touchberrypi kernel: [    3.541700] fbtft_device:  SPI devices registered:
    Jan 24 08:31:51 touchberrypi kernel: [    3.541708] fbtft_device:  'fb' Platform devices registered:
    Jan 24 08:31:51 touchberrypi kernel: [    3.541737] fbtft_device:      soc:fb id=-1 pdata? no
    Jan 24 08:31:51 touchberrypi kernel: [    3.541768] fbtft_device:  spi_busnum_to_master(0) returned NULL
    Jan 24 08:31:51 touchberrypi kernel: [    3.541771] fbtft_device: failed to register SPI device

It seems as if there is some SPI error, but the `spi-bcm2835` seem to have correctly been loaded prior to `fbtft_device`, so I don't know what I'm missing.

Now that the options are correctly set, I can make the screen work by simply doing:

    sudo modprobe fbtft_device

(with no further options) from an ssh connection. Am I missing something? Race condition? Hardware initialization issues? Or some mix-up in module loading order?

Thanks a lot for the software and for all your help.

On a sidenote, the link of the wiki you just sent me has the Donations link broken, just saying ;)

notro, 2016-01-24:
> spi_busnum_to_master(0) returned NULL

This means spi-bcm2835 hasn't been registered as spi master driver (yet).
I don't know why this happens to you though. You could try and add ```debug``` to /boot/cmdline.txt. That would give you loads of systemd messages. Maybe you'll get more info about excatly when the modules are loaded.

alexbarcelo, 2016-01-25:
Yeah, it seems you are right, seems as if the `fbtft_device` was being loaded as the first module. Looking into the `kernel` output from the syslog I obtain:

    Jan 25 08:16:56 touchberrypi kernel: [   12.070768] random: nonblocking pool is initialized
    Jan 25 08:16:56 touchberrypi kernel: [   18.871445] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
    Jan 25 08:16:56 touchberrypi kernel: [   18.898751] fbtft_device:  SPI devices registered:
    Jan 25 08:16:56 touchberrypi kernel: [   18.919310] fbtft_device:  'fb' Platform devices registered:
    Jan 25 08:16:56 touchberrypi kernel: [   18.940461] fbtft_device:      soc:fb id=-1 pdata? no
    Jan 25 08:16:56 touchberrypi kernel: [   18.960783] fbtft_device:  spi_busnum_to_master(0) returned NULL
    Jan 25 08:16:56 touchberrypi kernel: [   18.981784] fbtft_device: failed to register SPI device
    Jan 25 08:16:56 touchberrypi kernel: [   21.492737] i2c /dev entries driver
    Jan 25 08:16:56 touchberrypi kernel: [   21.922317] bcm2708_spi 3f204000.spi: DMA channel 2 at address 0xf3007200 with irq 77
    Jan 25 08:16:56 touchberrypi kernel: [   21.922360] bcm2708_spi 3f204000.spi: DMA channel 4 at address 0xf3007400 with irq 20
    Jan 25 08:16:56 touchberrypi kernel: [   21.957303] ads7846 spi0.1: touchscreen, irq 497
    Jan 25 08:16:56 touchberrypi kernel: [   21.958667] input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input2
    Jan 25 08:16:56 touchberrypi kernel: [   21.959099] bcm2708_spi 3f204000.spi: SPI Controller at 0x3f204000 (irq 80)
    Jan 25 08:16:56 touchberrypi kernel: [   21.959123] bcm2708_spi 3f204000.spi: SPI Controller running in dma mode
    Jan 25 08:16:56 touchberrypi kernel: [   21.966757] bcm2708_i2c 3f804000.i2c: BSC1 Controller at 0x3f804000 (irq 79) (baudrate 100000)

However, the lines in the fbtft.conf file are as follows:

    spi-bcm2835
    flexfb
    fbtft_device

Just thinking, does `systemd` honour the order of them? I also tried to split them into two files, a `000-foo.conf` and a `999-bar.conf` but the behaviour is the same.

The lines of the `systemd-modules-load` service are consistent with the file ordering though.

I will keep looking into it. But using rc.local in order to perform the "late `modprobe`" gives good results ("it works"), so I have that fallback just in case.

notro, 2016-01-25:
Which kernel are you using? I don't remember when the switch to spi-bcm2835 was made, but I see that you use spi_bcm2708.

closed 2016-04-28 by notro