notro/fbtft #366: Problems with touch in python / pygame

suppenbowl, 2016-01-24:
Hi,

first of all thank you for your great work, my 3.5" touchscreen is working perfectly with X Windows.
In X windows, the display AND touch are perfectly working.

But when i try to use the touch-feature in python, the display is working, but touch always delivers 0,319 as coordinates. Always.

When i run evtest, the touch seems to be okay. Any ideas ?
Oh, just in case it is important, the touchscreen is /dev/input/event0.

Here are the relevant lines of the python-code:

os.putenv ("SDL_VIDEODRIVER" , "fbcon")
os.putenv ("SDL_FBDEV"              , "/dev/fb1")
os.putenv ("SDL_MOUSEDRV"     , "TSLIB")
os.putenv ("SDL_MOUSEDEV"     , "/dev/input/event0")

     if event.type == pygame.MOUSEBUTTONDOWN:
                        pos = (pygame.mouse.get_pos() [0], pygame.mouse.get_pos() [1])

Kind regards,

Stefan

notro, 2016-01-26:
Maybe you need to calibrate. See https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian#tslib-sdl-pygame

suppenbowl, 2016-01-27:
Hi,

thank you for your repley, but i already followed the steps. 
When using ts_test everything is perfect. 
The only problem i have is when accessing the Touchscreen from python / pygame.

Best regards,

Stefan

Krakawet, 2016-02-05:
Are you sure that event0 is your touchscreen ?
You can check this point with : 
cat /dev/input/event0
If you see no result when touching the screen, then your touchscreen may be linked elsewhere (see "ls /dev/input").
Mine is /dev/input/touchscreen as i setted up a udev rule, look at https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian (All Hail to Notro !)

pabloruizpons, 2016-04-09:
some news here? I have the same problem...
Setted up in udev as /dev/input/touchscreen but is also a link to event0
When calibrated and tested with ts_calibrate and ts_test it works PERFECT, but when I use the code in python, the pointer goes "crazy" with lots of points around the screen.

pabloruizpons, 2016-04-09:
Fixed creating a file
`#!/bin/bash

#enable wheezy package sources
echo "deb http://archive.raspbian.org/raspbian wheezy main
" > /etc/apt/sources.list.d/wheezy.list

#set stable as default package source (currently jessie)
echo "APT::Default-release \"stable\";
" > /etc/apt/apt.conf.d/10defaultRelease

#set the priority for libsdl from wheezy higher then the jessie package
echo "Package: libsdl1.2debian
Pin: release n=jessie
Pin-Priority: -10
Package: libsdl1.2debian
Pin: release n=wheezy
Pin-Priority: 900
" > /etc/apt/preferences.d/libsdl

#install
apt-get update
apt-get -y --force-yes install libsdl1.2debian/wheezy`

saving it into a file (fix.sh for example) then chmod +x fix.sh
and executing it

notro, 2017-02-22:
Closing issue since there has been no activity in a long time. Reopen if needed.

closed 2017-02-22 by notro

rGunti, 2017-10-20:
Just tested it on a recent version of Raspbian with a Watterott 2.7" Display. Can confirm that the @pabloruizpons works.

Noted, because there where no confirmations for this fix in this thread.

mr-varga, 2018-05-10:
After this fix i have this problem:

`pygame.error: File is not a Windows BMP file`

toddboyland, 2020-01-30:
I ran into the same issue w/ pygame on a Raspberry Pi.
In the OS, the touchscreen works fine.
When the python/pygame script it run the touchsreen is extremely erratic.
I tried wheezy, jessie, stretch, and buster (lite, and full OS) on Raspberry Pi version 3's & Pi 4 hardware platforms.

My solution ended up being related to how the .py script was started.
When I manually executed the .py script from ssh (as sudo due to permission issue), the touchscreen was erratic. 
Example: /usr/bin/python3 /home/pi/foldername/script.py

When I schedule the script using crontab -e (as pi user), the touchscreen was erratic.
Example entry in cron: @reboot /usr/bin/python3 /home/pi/foldername/script.py

When I configure autostart the touchscreen finally works well.
Configure autostart: (must be performed as pi user, as sudo causes changes in file permissions)
mkdir /home/pi/.config/autostart
nano /home/pi/.config/autostart/script_name.desktop

[Desktop Entry]
Type=Application
Name=script_name
Exec=/usr/bin/python3 /home/pi/foldername/script.py

Hopefully this helps someone else, as it took a long time to narrow down.