notro/fbtft #369: FBTFT Module Failing to load at startup

theLuthier, 2016-02-04:
I’m using a 2.2″ SPI TFT display with my RPi2. In conjunction with RetroPie 3.2, everything worked fine with FBTFT and FBCP working at startup. I’ve since installed Retropie 3.4 and I keep running into 2 problems:

1. FBTFT module fails to load when added to /etc/modules.

[image]

I’ve managed to work around it by adding “modprobe fbtft_device name=adafruit22a” to /etc/rc.local, however it only works about 50% of the time.

2. Following the trend, FBCP only runs at startup about 50% of the time. I've tried adding “/usr/local/bin/fbcp &” seemingly everywhere I've seen suggested online: rc.local, init.d/asplashscreen, my own custom startup script as recommend in the wiki (http://www.stuffaboutcode.com/2012/06/raspberry-pi-run-program-at-start-up.html).

However as soon as I get to the console, “modprobe fbtft_device” loads fine as does “fbcp&”. So this tells me that perhaps there’s an issue with when the module loads and when/if the fbcp command runs.

Any help is _very_ appreciated at this point.

PS - Huge thanks to notro for making this possible in the first place!

notro, 2016-02-04:
Have you followed this: https://github.com/notro/fbtft/wiki#make-it-permanent-debian-jessie-arch-linux-systemd

theLuthier, 2016-02-04:
Thanks for the repy, notro. Yes I have, here's the output from dmesg:

[image]

notro, 2016-02-04:
This kernel seem to be using the out-of-tree [spi-bcm2708](https://github.com/notro/spi-bcm2708/blob/master/spi-bcm2708.c) driver, so you need to specify spi-bcm2708 instead of spi-bcm2835.

theLuthier, 2016-02-04:
That *seems* to have fixed it! Though dmesg still gives an error:

[image]

I'm used to this working sometimes and not others, so I'll give it a couple days and update if this appears to be a permanent fix. Thanks a ton for the help!

notro, 2016-02-04:
If you add ```debug``` to /boot/cmdline.txt, you'll get a lot of systemd debug messages in the kernel log. Maybe that can aid you in tracking down the reason for the error.

theLuthier, 2016-02-04:
Oh nice, thanks for the suggestion - I'll certainly try it.

notro, 2016-04-28:
Closing old issues, reopen if needed.

closed 2016-04-28 by notro

Zhairgling, 2017-03-09:
Hi did you solved this error ? I've the same : 

[image]