notro/fbtft #371: raspbian fbdev error

canamla, 2016-02-06:
hi Notro!
i have some problems with fbtft in X mode, console mode works well
startx doesn't load it's trying trying and tryin some times i can see mouse cursor and logo, then again all black

i saw error :
gdbus.error org.freedesktop.policykit1.error.failed an authentication agent already exists for the given subject

 solution to the problem was to edit the file /etc/xdg/autostart/lxpolkit.desktop
NotShowIn=GNOME;KDE;LXDE

for  time then im waiting the rapsberry standing more slower  
in the ssh in parallel session wrote:  cannot allocate memory 
and sudo reboot didn't work

in the  /var/log/Xorg.0.log
(EE)  FBDEV(0): FBIOBLANK: Invalid argument

[  1466.360]
X.Org X Server 1.12.4
Release Date: 2012-08-27
[  1466.362] X Protocol Version 11, Revision 0
[  1466.363] Build Operating System: Linux 3.2.0-2-mx5 armv7l Debian
[  1466.366] Current Operating System: Linux raspberrypi 4.0.7+ #1 PREEMPT Sat Jul 11 20:08:42 CEST 2015 armv6l
[  1466.366] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boar$
[  1466.378] Build Date: 11 February 2015  09:31:17PM
[  1466.382] xorg-server 2:1.12.4-6+deb7u6 (Julien Cristau <jcristau@debian.org>)
[  1466.384] Current version of pixman: 0.33.1
[  1466.388]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[  1466.388] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[  1466.405] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Feb  5 22:50:41 2016
[  1466.417] (==) Using config directory: "/etc/X11/xorg.conf.d"
[  1466.421] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[  1466.430] (==) ServerLayout "TFT"
[  1466.430] (**) |-->Screen "ScreenTFT" (0)
[  1466.430] (**) |   |-->Monitor "MonitorTFT"
[  1466.439] (**) |   |-->Device "DeviceTFT"
[  1466.440] (==) Automatically adding devices
[  1466.440] (==) Automatically enabling devices
[  1466.464] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[  1466.464]    Entry deleted from font path.
[  1466.465] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[  1466.465]    Entry deleted from font path.
[  1466.465] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[  1466.465]    Entry deleted from font path.
[  1466.465] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[  1466.465]    Entry deleted from font path.
[  1466.472] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[  1466.473]    Entry deleted from font path.
[  1466.473] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[  1466.473]    Entry deleted from font path.
[  1466.473] (WW) The directory "/var/lib/defoma/x-ttcidfont-conf.d/dirs/TrueType" does not exist.
[  1466.473]    Entry deleted from font path.
[  1466.473] (==) FontPath set to:
        /usr/share/fonts/X11/Type1,
        built-ins
[  1466.474] (==) ModulePath set to "/usr/lib/xorg/modules"
[  1466.474] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[  1466.474] (II) Loader magic: 0xb6f46cf0
[  1466.474] (II) Module ABI versions:
[  1466.474]    X.Org ANSI C Emulation: 0.4
[  1466.474]    X.Org Video Driver: 12.1
[  1466.474]    X.Org XInput driver : 16.0
[  1466.475]    X.Org Server Extension : 6.0
[  1466.475] (II) LoadModule: "extmod"
[  1466.496] (II) Loading /usr/lib/xorg/modules/extensions/libextmod.so
[  1466.512] (II) Module extmod: vendor="X.Org Foundation"
[... 151 lines cut ...]
[  1467.704] (II) evdev: G-Tech Wireless Dongle: Adding scrollwheel support
[  1467.704] (**) evdev: G-Tech Wireless Dongle: YAxisMapping: buttons 4 and 5
[  1467.704] (**) evdev: G-Tech Wireless Dongle: EmulateWheelButton: 4, EmulateWheelInertia: 10, EmulateWheelTime$
[  1467.705] (**) Option "config_info" "udev:/sys/devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3:1.1/0003$
[  1467.705] (II) XINPUT: Adding extended input device "G-Tech Wireless Dongle" (type: KEYBOARD, id 7)
[  1467.705] (**) Option "xkb_rules" "evdev"
[  1467.706] (**) Option "xkb_model" "pc105"
[  1467.706] (**) Option "xkb_layout" "gb"
[  1467.708] (II) evdev: G-Tech Wireless Dongle: initialized for relative axes.
[  1467.709] (WW) evdev: G-Tech Wireless Dongle: ignoring absolute axes.
[  1467.711] (**) G-Tech Wireless Dongle: (accel) keeping acceleration scheme 1
[  1467.711] (**) G-Tech Wireless Dongle: (accel) acceleration profile 0
[  1467.712] (**) G-Tech Wireless Dongle: (accel) acceleration factor: 2.000
[  1467.712] (**) G-Tech Wireless Dongle: (accel) acceleration threshold: 4
[  1467.715] (II) config/udev: Adding input device G-Tech Wireless Dongle (/dev/input/mouse0)
[  1467.716] (II) No input driver specified, ignoring this device.
[  1467.716] (II) This device may have been added with another device file.
[  1467.773] (EE) FBDEV(0): FBIOBLANK: Invalid argument
[  1634.779] (II) evdev: G-Tech Wireless Dongle: Close
[  1634.820] (II) UnloadModule: "evdev"
[  1634.821] (II) evdev: G-Tech Wireless Dongle: Close
[  1634.821] (II) UnloadModule: "evdev

i think it is conflict in configuration but i have no idea,  i need help
this is my first console experience )

notro, 2016-02-07:
Do you have any disk space left? ```df -h```

canamla, 2016-02-08:
~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  3.5G   11G  26% /
devtmpfs        237M     0  237M   0% /dev
tmpfs            49M  248K   48M   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs            97M     0   97M   0% /run/shm
/dev/mmcblk0p1   56M   20M   37M  35% /boot

canamla, 2016-02-08:
25 minutes and ssh session -  dead
cannot allocate memory

notro, 2016-02-09:
I don't know what the problem can be. The Raspberry Pi forum is probably a better place to get help troubleshooting this.

canamla, 2016-02-09:
Please answer me, for last question
 i think that it can  be seeing the callibrator

> 

Install

`cd /tmp
wget http://tronnes.org/downloads/xinput-calibrator_0.7.5-1_armhf.deb
sudo dpkg -i -B xinput-calibrator_0.7.5-1_armhf.deb
rm xinput-calibrator_0.7.5-1_armhf.deb`

Configure xinput-calibrator to autostart with X windows.

`sudo wget -O /etc/X11/Xsession.d/xinput_calibrator_pointercal https://raw.github.com/tias/xinput_calibrator/master/scripts/xinput_calibrator_pointercal.sh
echo "sudo /bin/sh /etc/X11/Xsession.d/xinput_calibrator_pointercal" | sudo tee -a /etc/xdg/lxsession/LXDE-pi/autostart`

> 

how can i well disable this changes with all routes for this ap

closed 2017-02-22 by notro