notro/fbtft #372: Display power on/off

rogercorrea, 2016-02-06:
Please,
I need control display to power on/off.
I'm using Waveshare 3.5 LCD.
It's possible execute command using python to on/off display?
The my idea is to use my presence sensor to determine of the display state.
Thanks!

notro, 2016-02-07:
I don't think there's any backlight control on that display. If enabled you should have a folder under ```/sys/class/backlight/```.
Is there a led= in your gpios= argument to fbtft_device?

Krakawet, 2016-02-08:
If electronic don't afraid you, you may achieve your projet using a transistor as a switch to control the current supply from a gpio pin ?

rogercorrea, 2016-02-10:
Yes, but I imagined one solution more simples (but it seems not).
Maybe deactivate VCC pins of the Raspberry were able to shut display.
It's possible?

notro, 2016-02-13:
If you cut power, the display controller would loose it's setup, and the only way to get it back is to reload the driver.

rogercorrea, 2016-02-13:
Yes, I know.
For now I am implemented one method to force sleep mode using "sudo pm-suspend", but display backlight don't shut.
I don't know if is sufficient to energy saver.

I tried remove and reload drivers but my Raspbian crashed and only back with reboot.

muesli4, 2016-02-17:
In the specification I read that the ILI9341 does support low-power states, are those supported? Is there some way to stop refreshing the screen? (Does my display survive longer if not refreshed?)

notro, 2016-02-18:
> In the specification I read that the ILI9341 does support low-power states, are those supported?

No, but as far as I remember those savings are so small (~1mA?) that they only makes sense when the display is driven by a very low power device. Raspberry Pi for instance draws way to much power for this to make any sense. It's the backlight that draws much power (100-150mA?).

> Is there some way to stop refreshing the screen?

The display controller keeps refreshing the panel all the time as any display would.
If you don't update the framebuffer, nothing is transferred over SPI though.

> Does my display survive longer if not refreshed?

At least it lasts longer if it is powered off, but it all depends on the panel quality. For instance the rpi-display by Watterott uses a industrial grade panel, and I'll expect that to last longer. But I'm just guessing here.

Snyder1977, 2016-03-22:
Is there any chance to power on/off the backlight of this SainSmart 5" HDMI Display?

[http://hardware-libre.fr/2015/07/review-guide-sainsmart-5-inch-hdmi-touch-display/](url)

I would be great if the backlight could be controlled by software, Any advice would be very much appreciated.

closed 2017-02-22 by notro