notro/fbtft #373: New Tontec MZ61581 model laggy / buggy ?

Krakawet, 2016-02-07:
Hello everyone !
I'm getting stuck with the new model of the MZ61581-PI-EXT labelled 2015.12.12.

I followed the tontec official setup guide : basically a wheezy with the tontec version of mz61581-overlay, https:/www.itontec.com/mz61581-overlay.dtb
It works but give poor refresh rate (4-5fps). Here is the debug output when testing as described in the performance section :
[    5.134410] fb_s6d02a1 spi0.0: Display update: 1427 kB/s (210.015 ms), fps=0 (0.000 ms)
[    5.214428] graphics fb1: fb_s6d02a1 frame buffer, 480x320, 300 KiB video memory, 4 KiB DMA buffer memory, fps=33, spi0.0 at 128 MHz
[  246.433115] fb_s6d02a1 spi0.0: Display update: 1876 kB/s (159.828 ms), fps=5 (179.977 ms)
using fps=50 as dtparam option didn't do the trick.

It seems that using the original wheezy 2015-05-05 mz51681-overlay.dtb give the same result.

I noticed that doing a apt-get update/upgrade on the wheezy or using notro's firmware give a good refresh rate, but the image is bad-looking: it looks like semi transparent small horizontal lines where added to the original image. See how this bug is looking : http://technico.top/bug.jpg

[    4.071614] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    4.094943] fb_s6d02a1: module is from the staging directory, the quality is unknown, you have been warned.
[    4.114872] fbtft_of_value: width = 320
[    4.126691] fbtft_of_value: height = 480
[    4.138003] fbtft_of_value: buswidth = 8
[    4.148963] fbtft_of_value: debug = 3
[    4.159683] fbtft_of_value: rotate = 270
[    4.169773] fbtft_of_value: fps = 30
[    4.179945] fb_s6d02a1 spi0.0: fbtft_request_one_gpio: 'reset-gpios' = GPIO15
[    4.196668] fb_s6d02a1 spi0.0: fbtft_request_one_gpio: 'dc-gpios' = GPIO25
[    4.219552] fb_s6d02a1 spi0.0: fbtft_request_one_gpio: 'led-gpios' = GPIO18
[    4.253710] fb_s6d02a1 spi0.0: fbtft_verify_gpios()
[    4.276918] fb_s6d02a1 spi0.0: fbtft_init_display_dt()
[    4.292323] fb_s6d02a1 spi0.0: fbtft_reset()
[    4.459994] fb_s6d02a1 spi0.0: init: write_register: B0 00
[    4.472096] fb_s6d02a1 spi0.0: init: write_register: 11
[    4.483316] fb_s6d02a1 spi0.0: init: msleep(255)
[    4.754557] fb_s6d02a1 spi0.0: init: write_register: B3 02 00 00 00
[    4.766771] fb_s6d02a1 spi0.0: init: write_register: C0 13 3B 00 02 00 01 00 43
[    4.779983] fb_s6d02a1 spi0.0: init: write_register: C1 08 16 08 08
[    4.792118] fb_s6d02a1 spi0.0: init: write_register: C4 11 07 03 03
[    4.804172] fb_s6d02a1 spi0.0: init: write_register: C6 00
[    4.815367] fb_s6d02a1 spi0.0: init: write_register: C8 03 03 13 5C 03 07 14 08 00 21 08 14 07 53 0C 13 03 03 21 00
[    4.831605] fb_s6d02a1 spi0.0: init: write_register: 35 00
[    4.842848] fb_s6d02a1 spi0.0: init: write_register: 36 A0
[    4.854039] fb_s6d02a1 spi0.0: init: write_register: 3A 55
[    4.865205] fb_s6d02a1 spi0.0: init: write_register: 44 00 01
[    4.876555] fb_s6d02a1 spi0.0: init: write_register: D0 07 07 1D 03
[    4.888302] fb_s6d02a1 spi0.0: init: write_register: D1 03 30 10
[    4.899719] fb_s6d02a1 spi0.0: init: write_register: D2 03 14 04
[    4.911066] fb_s6d02a1 spi0.0: init: write_register: 29
[    4.921541] fb_s6d02a1 spi0.0: init: write_register: 2C
[    4.931964] fb_s6d02a1 spi0.0: set_var()
[    4.966594] fb_s6d02a1 spi0.0: Display update: 11748 kB/s (25.536 ms), fps=0 (0.000 ms)
[    4.979926] fb_s6d02a1 spi0.0: fbtft_register_backlight()
[    4.991644] graphics fb1: fb_s6d02a1 frame buffer, 480x320, 300 KiB video memory, 4 KiB DMA buffer memory, fps=33, spi0.0 at 128 MHz
[    5.009070] fb_s6d02a1 spi0.0: fbtft_backlight_update_status: polarity=0, power=0, fb_blank=0
[    5.023106] bcm2708_spi 3f204000.spi: SPI Controller at 0x3f204000 (irq 80)
[    5.035558] bcm2708_spi 3f204000.spi: SPI Controller running in dma mode
[   55.658737] fb_s6d02a1 spi0.0: Display update: 12473 kB/s (24.050 ms), fps=20 (50.000 ms)

Did someone have any clue one making this working ?

Krakawet, 2016-02-07:
One more hint : 
True black and true red color are displayed in a correct way.
Every other color is showing these color variation.
Seen on test image : https://i.ytimg.com/vi/qqAtlyQTMn8/maxresdefault.jpg -> http://technico.top/mire.jpg

notro, 2016-02-07:
> graphics fb1: fb_s6d02a1 frame buffer, 480x320, 300 KiB video memory, 4 KiB DMA buffer memory, fps=33, spi0.0 at 128 MHz

If you have a kernel that uses the spi-bcm2835 SPI controller driver, you need to increase the DMA buffer to 16 or 32kB: https://github.com/notro/fbtft/wiki#install

https://github.com/raspberrypi/linux/blob/rpi-4.1.y/arch/arm/boot/dts/overlays/mz61581-overlay.dts

When it comes to your bug, compare the ```init: write_register: ``` sequence to see if there's a difference in values between the kernels.

Krakawet, 2016-02-09:
Mouhahaha it's working !
The solution was to use a bigger DMA buffer (the slow part), and/or to limit the SPI speed (the distorsion part).

Notro, you deserve your own religion ! (and you should reactivate your donation link ;)

closed 2016-02-09 by Krakawet