notro/fbtft #374: Odroid W with docking board and 2.2inch 320x240 QVGA color TFT LCD assembled

perrycannon, 2016-02-07:
I have an Hardkernel Odroid W connected a Odroid W docking board with a 2.2inch 320x240 QVGA color TFT LCD assembled attached.

I have followed the steps in the wiki to get the tft lcd to work. I am using Jessie debain updated. I have been unsuccessful. The tft lcd display will not work. Please help me.

Thank you
Perry

Krakawet, 2016-02-07:
Can you please give more informations ?
Please copy there every tft related line when doing "lsmod" or "dmesg", and give us the URL of the wiki setup ,

perrycannon, 2016-02-07:
How can I copy dmesg?

Thank you
Perry

Krakawet, 2016-02-07:
The easiest way is to open a ssh terminal from your main computer.
If you are on windows, download, by example kitty (www.9bis.net/kitty/?page=Download).
Type your raspberry IP Adress, port 22 and let's go to login.

From now, you will access your raspberry from your main PC, allowing you to directly copy/paste stuff with a simple right click.

perrycannon, 2016-02-08:
This is the wiki i used. https://github.com/notro/fbtft/wiki 
Below is what I could find in dmesg and lsmod
dmesg[    0.524259] console [ttyAMA0] enabled[    0.528648] bcm2835-mbox 2000b880.mailbox: mailbox enabled[    0.596846] bcm2708-dmaengine 20007000.dma: DMA legacy API manager at f2007000, dmachans=0xf35[    0.605728] bcm2708-dmaengine 20007000.dma: Initialized 7 DMA channels (+ 1 legacy)[    0.614245] bcm2708-dmaengine 20007000.dma: Load BCM2835 DMA engine driver[    0.621240] bcm2708-dmaengine 20007000.dma: dma_debug:0
LSMODModule                   Size  Used bycfg80211               499834  0rfkill                   22491  2 cfg80211joydev                   9913  0evdev                   11364  2r8188eu                467496  0bcm2835_gpiomem         3703  0spi_bcm2835           7948  0snd_bcm2835          22317  0snd_pcm                 92581  1 snd_bcm2835snd_timer               23454  1 snd_pcmsnd                     68161  3 snd_bcm2835,snd_timer,snd_pcmuio_pdrv_genirq      3690  0uio                     10002  1 uio_pdrv_genirqi2c_dev                  6730  0fuse                    91981  1ipv6                   360374  24
I really appreciate your help. I don't know what how I can get the tft display to function.
Thank youPerry

Krakawet, 2016-02-08:
Ok, as we don't see any fbtft, fbtft_device or fb_* stuff, it seems the setup procedure failed quite early. Let's assume we will do new attempts on fresh raspbian image.
If you don't mind working on a non-updated wheezy, this guide is still looking relevant : [http://odroid.com/dokuwiki/doku.php?id=en:odroidwtftlcd](http://odroid.com/dokuwiki/doku.php?id=en:odroidwtftlcd)

In other case, take your jessie image :
_sudo rpi-update
sudo apt-get update
sudo apt-get upgrade_
Write "dtparam=spi=on" in /boot/config.txt ,reboot and try the :
_sudo modprobe fbtft_device name=adafruit22a
con2fbmap 1 1_
If you see something, that's fine, you can follow the "making it permanent section", in other cases, please paste here the last dmesg lines.
There are some variation around the adafruit 2.2" drivers, "_sudo modprobe fbtft_device name=adafruit22_" may also be tried. Note that one of the old adafruit 2,2" is no more supported, hope you didn't get this one on your board !

perrycannon, 2016-02-09:
Thank you for your help. The steps you provided worked. I used the latest NOOBS, then updated.

Now I need to learn how to rotate, and get the GUI on the tft. I was able to use the CLI on the tft, startx sent the GUI to the HDMI display.

Thank you for your help.
Perry

Krakawet, 2016-02-09:
Rotation should not be hard to achieve, as it's an option when loading fbtft_device. I assume the default rotation is 270°, dmesg may confirm it. In that case, try :
sudo modprobe fbtft_device name=adafruit22a rotate=90

For displaying, you may change /usr/share/X11/xorg.conf.d/99-fbturbo.conf
and change "Option fbdev /dev/fb0" to "Option  fbdev /dev/fb1".

closed 2017-02-22 by notro