notro/fbtft #376: "Click released" not detected when the touch is short?.

Ferite, 2016-02-13:
Hi Notro and commnunity.

Hi have a WaveShare  SpotPear 3.2". I have some issue and I am hopping to get some advice.

It seems like when you press tochscreen for a very short time, a "click pressed" event is detected, but not the release. Thus, if this happen on a button, for instance, It looks like the button keeps being pressed, despite you have removed your finger from the touchscreen. This behavior is a little bit tricky of reproduce, but it happens time to time. Is there a setting I should change to fix this?. The touchscreen settings in /usr/share/X11/xorg.conf.d/10-evdev.conf looks this way:

Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "Calibration"   "240 3800 3900 240"
EndSection

Thanks and best regards.

Krakawet, 2016-02-16:
For debug purpose and/or direct touchscreen read, you can : 
cat /dev/input/touchscreen
Note that the touchscreen may be a registered with a different name !

From there you will see exactly which events codes are sent, and possibly read them from this location.

Ferite, 2016-02-22:
Thanks @Krakawet . This is a good start point.

closed 2017-02-22 by notro