notro/fbtft #377: Adafruit18_green distorted image

jfs10, 2016-02-14:
I have an adafruit18_green connected to a Rpi Model A running Jessie. I am using python and pygame to display a hello world message on screen. The pointer appears elongated and the text almost looks as if it is 3d. Going from left to right, the characters gets larger.

I had initially tried adafruit18, but I just got a white screen. Adafruit18_green seems to work. Below is the from dmesg:

> [   62.740531] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
> [   62.750133] fbtft_device:  SPI devices registered:
> [   62.750178] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
> [   62.750196] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
> [   62.750207] fbtft_device:  'fb' Platform devices registered:
> [   62.750255] fbtft_device:      soc:fb id=-1 pdata? no
> [   62.750319] fbtft_device: Deleting spi0.0
> [   62.750842] spi spi0.0: setting up native-CS0 as GPIO 8
> [   62.751027] fbtft_device:  GPIOS used by 'adafruit18_green':
> [   62.751051] fbtft_device:    'reset' = GPIO25
> [   62.751063] fbtft_device:    'dc' = GPIO24
> [   62.751073] fbtft_device:    'led' = GPIO18
> [   62.751082] fbtft_device:  SPI devices registered:
> [   62.751098] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
> [   62.751112] fbtft_device:      fb_st7735r spi0.0 4000kHz 8 bits mode=0x00
> [   62.786721] fb_st7735r: module is from the staging directory, the quality is unknown, you have been warned.
> [   63.764366] graphics fb1: fb_st7735r frame buffer, 160x128, 40 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 4 MHz
> 

Any ideas why the displayed images/text appear distorted?

Thank you

notro, 2016-02-14:
There are three differences between **adafruit18** and **adafruit18_green**:
* Speed 32MHz vs. 4MHz.
* Custom set_addr_win: http://lxr.free-electrons.com/ident?i=adafruit18_green_tab_set_addr_win
* BGR/RGB config

See: http://lxr.free-electrons.com/source/drivers/staging/fbtft/fbtft_device.c#L249

Try adafruit18 and lower the speed: speed=4000000

jfs10, 2016-02-15:
Setting the device to adafruit18 and setting the speed to 4000000 worked
perfectly.

Thank you

jfs10, 2016-02-17:
The screen is working, but I have to do a ctrl-c to get my python test
script to progress. The script just writes hello there on the screen. I run
the script and nothing displays until I hit ctrl-c. Then I have to hit
ctrl-c to finish the script.

I have yet to try showing a movie.

Thanks

Jonathan Swaby

closed 2017-02-22 by notro