notro/fbtft #378: Adafruit22 - write failed

jfs10, 2016-02-14:
I have an old adafruit 2.2 tft attached to a  Rpi B+ running Jessie. I did have this display working on an arduino and a Rpi, but that was a while ago. I did try it on a Rpi A recently, and got the same results I am getting now. I have checked and rechecked the wiring. I am hoping that the error that I am getting is related to wiring.

Below is the output from dmesg:

> [  105.612673] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
> [  105.619750] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
> [  105.621520] fbtft_device:  SPI devices registered:
> [  105.621561] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
> [  105.621583] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
> [  105.621596] fbtft_device:  'fb' Platform devices registered:
> [  105.621648] fbtft_device:      soc:fb id=-1 pdata? no
> [  105.621715] fbtft_device: Deleting spi0.0
> [  105.632745] spi spi0.0: setting up native-CS0 as GPIO 8
> [  105.633000] fbtft_device:  GPIOS used by 'adafruit22':
> [  105.633025] fbtft_device:    'reset' = GPIO25
> [  105.633038] fbtft_device:    'led' = GPIO23
> [  105.633049] fbtft_device:  SPI devices registered:
> [  105.633067] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
> [  105.633085] fbtft_device:      fb_hx8340bn spi0.0 32000kHz 8 bits mode=0x00
> [  105.676845] fb_hx8340bn: module is from the staging directory, the quality is unknown, you have been warned.
> [  105.803170] fb_hx8340bn spi0.0: write() failed and returned -22
> [  105.803213] fb_hx8340bn spi0.0: write() failed and returned -22
> [  105.956822] fb_hx8340bn spi0.0: write() failed and returned -22
> [  105.956867] fb_hx8340bn spi0.0: write() failed and returned -22
> [  105.956890] fb_hx8340bn spi0.0: write() failed and returned -22
> [  105.981465] fb_hx8340bn spi0.0: write() failed and returned -22
> [  105.981508] fb_hx8340bn spi0.0: write() failed and returned -22
> [  105.995129] fb_hx8340bn spi0.0: write() failed and returned -22
> [  105.995176] fb_hx8340bn spi0.0: write() failed and returned -22
> [  106.009027] fb_hx8340bn spi0.0: write() failed and returned -22
> [  106.009073] fb_hx8340bn spi0.0: write() failed and returned -22
> [  106.009096] fb_hx8340bn spi0.0: write() failed and returned -22
> [  106.009114] fb_hx8340bn spi0.0: write() failed and returned -22
> [  106.009274] fb_hx8340bn spi0.0: fbtft_update_display: write_vmem failed to update display buffer
> [  106.009297] fb_hx8340bn spi0.0: write() failed and returned -22
> [  106.009316] fb_hx8340bn spi0.0: write() failed and returned -22
> [  106.009334] fb_hx8340bn spi0.0: write() failed and returned -22
> [  106.031004] graphics fb1: fb_hx8340bn frame buffer, 220x176, 75 KiB video memory, 16 KiB DMA buffer memory, fps=20, spi0.0 at 32 MHz
> 

Any ideas?

thanks

notro, 2016-02-14:
That's a 9-bit SPI display. Add fbtft_device option debug=7. That should print some function names and we'll find out if it tries to use native 9-bit or if it emulates using 8-bit.
It looks to me that it uses native 9-bit, but the spi-bcm2835 driver doesn't support that. spi-bcm2708 does support 9-bit, unless it's the DMA capable version.

jfs10, 2016-02-15:
The command that I issued:
 modprobe fbtft_device name=adafruit22 rotate=90 speed=16000000 debug=7

The results from dmesg:
[  250.645557] fbtft_device: module is from the staging directory, the
quality is unknown, you have been warned.
[  250.665125] fbtft_device:  SPI devices registered:
[  250.665178] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[  250.665197] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[  250.665209] fbtft_device:  'fb' Platform devices registered:
[  250.665259] fbtft_device:      soc:fb id=-1 pdata? no
[  250.665327] fbtft_device: Deleting spi0.0
[  250.673850] spi spi0.0: setting up native-CS0 as GPIO 8
[  250.674064] fbtft_device:  GPIOS used by 'adafruit22':
[  250.674088] fbtft_device:    'reset' = GPIO25
[  250.674102] fbtft_device:    'led' = GPIO23
[  250.674113] fbtft_device:  SPI devices registered:
[  250.674130] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[  250.674145] fbtft_device:      fb_hx8340bn spi0.0 16000kHz 8 bits
mode=0x00
[  250.709280] fb_hx8340bn: module is from the staging directory, the
quality is unknown, you have been warned.
[  250.712738] fb_hx8340bn spi0.0: fbtft_gamma_parse_str() str=
[  250.712784] fb_hx8340bn spi0.0: 1 3 0E 5 0 2 09 0 6 1 7 1 0 2 2
3 3 17 8 4 7 05 7 6 0 3 1 6 0 0
[  250.712869] fb_hx8340bn spi0.0: fbtft_request_gpios_match('reset')
[  250.712928] fb_hx8340bn spi0.0: fbtft_request_gpios: 'reset' = GPIO25
[  250.712950] fb_hx8340bn spi0.0: fbtft_request_gpios_match('led')
[  250.712986] fb_hx8340bn spi0.0: fbtft_request_gpios: 'led' = GPIO23
[  250.713006] fb_hx8340bn spi0.0: fbtft_verify_gpios()
[  250.713023] fb_hx8340bn spi0.0: fbtft_reset()
[  250.978536] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: c1 ff 83 40
[  250.978595] fb_hx8340bn spi0.0: fbtft_write_spi(len=8): c1 00 ff 01 83
01 40 01
[  250.978622] fb_hx8340bn spi0.0: write() failed and returned -22
[  250.978641] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: 11
[  250.978660] fb_hx8340bn spi0.0: fbtft_write_spi(len=2): 11 00
[  250.978677] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.246137] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: ca 70 00 d9
[  251.246191] fb_hx8340bn spi0.0: fbtft_write_spi(len=8): ca 00 70 01 00
01 d9 01
[  251.246218] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.246239] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: b0 01 11
[  251.246262] fb_hx8340bn spi0.0: fbtft_write_spi(len=6): b0 00 01 01 11 01
[  251.246279] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.246298] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: c9 90 49 10 28 28
10 00 06
[  251.246319] fb_hx8340bn spi0.0: fbtft_write_spi(len=18): c9 00 90 01 49
01 10 01 28 01 28 01 10 01 00 01 06 01
[  251.246336] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.282183] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: b5 35 20 45
[  251.282238] fb_hx8340bn spi0.0: fbtft_write_spi(len=8): b5 00 35 01 20
01 45 01
[  251.282268] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.282289] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: b4 33 25 4c
[  251.282309] fb_hx8340bn spi0.0: fbtft_write_spi(len=8): b4 00 33 01 25
01 4c 01
[  251.282326] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.302136] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: 3a 05
[  251.302188] fb_hx8340bn spi0.0: fbtft_write_spi(len=4): 3a 00 05 01
[  251.302213] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.302233] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: 29
[  251.302252] fb_hx8340bn spi0.0: fbtft_write_spi(len=2): 29 00
[  251.302271] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.322156] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: 36 a8
[  251.322209] fb_hx8340bn spi0.0: fbtft_write_spi(len=4): 36 00 a8 01
[  251.322234] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.322260] fb_hx8340bn spi0.0: fbtft_update_display(start_line=0,
end_line=175)
[  251.322280] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: 2a 00 00 00 db
[... 12 lines cut ...]
[  251.322607] fb_hx8340bn spi0.0: fbtft_write_spi(len=16384): 00 01 00 01
00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00
01 00 01 ...
[  251.322628] fb_hx8340bn spi0.0: fbtft_update_display: write_vmem failed
to update display buffer
[  251.322659] fb_hx8340bn spi0.0: Display update: 191941 kB/s (0.394 ms),
fps=0 (0.000 ms)
[  251.322681] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: 26 01
[  251.322700] fb_hx8340bn spi0.0: fbtft_write_spi(len=4): 26 00 01 01
[  251.322718] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.322738] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: c2 60 71 01 0e 05
02 09 31 0a
[  251.322760] fb_hx8340bn spi0.0: fbtft_write_spi(len=20): c2 00 60 01 71
01 01 01 0e 01 05 01 02 01 09 01 31 01 0a 01
[  251.322778] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.322797] fb_hx8340bn spi0.0: fbtft_write_reg8_bus9: c3 67 30 61 17 48
07 05 33
[  251.322817] fb_hx8340bn spi0.0: fbtft_write_spi(len=18): c3 00 67 01 30
01 61 01 17 01 48 01 07 01 05 01 33 01
[  251.322834] fb_hx8340bn spi0.0: write() failed and returned -22
[  251.342210] fb_hx8340bn spi0.0: fbtft_register_backlight()
[  251.351782] graphics fb1: fb_hx8340bn frame buffer, 220x176, 75 KiB
video memory, 16 KiB DMA buffer memory, fps=20, spi0.0 at 16 MHz
[  251.351827] fb_hx8340bn spi0.0: fbtft_backlight_update_status:
polarity=1, power=0, fb_blank=0

ConnieNew, 2016-02-27:
I don't think it is the wiring.

I am actually getting the same issue...though I am not using the adafruit22 device, I am using something that worked before on my older rPis but are failing on a B+.  I have a revision 1 original rPi b and my tft board works great using adafruit22a and notros old drivers (i am too worried to upgrade and lose functionality)
Using this same board on an rPi B+ with latest firmware produces the same write errors as you have.

ConnieNew, 2016-02-27:
my device works now!
needed the adafruit22a device rather than the adafruit22
try
sudo modprobe fbtft_device name=adafruit22a rotate=90 speed=16000000 
good luck

notro, 2016-02-27:
There was a fix for 9-bit displays in 4.3 which is needed when using spi-bcm2835: https://github.com/torvalds/linux/commit/cabb5b2a2866829177df2f1cee70f0dd42ee6ae0

4.4 will be the default soon, can be tried now with: ```sudo BRANCH=next rpi-update```

jfs10, 2016-02-27:
> spi-bcm2835: torvalds/linux@cabb5b2

That worked. thank you

closed 2017-02-22 by notro