notro/fbtft #379: has anyone found settings that make touchscreen work with Seeed / 52Pi "PiShow 2.8"?

jpwynn, 2016-02-20:
I can get the display working fine, with the following in /boot/config.txt

```
dtparam=spi=on
dtparam=i2c1=on
dtparam=i2c_arm=on
dtoverlay=rpi-display,rotate=90,fps=20
```

However, I cannot get the touchscreen to respond (using sudo evtest) at all.

dmesg does not report any errors.

evtest startts off okay, but touching the screne does not generat einterrupts:

```
$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	ADS7846 Touchscreen
Select the device event number [0-0]: 0
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "ADS7846 Touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max     4095
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max     4095
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max      255
Properties:
Testing ... (interrupt to exit)
```

The Seeed's absolutely horrid docs give an example of using /etc/modules the suggest ads7846 with the parameters below, but they do not work. I tried 2 touchscreens to try to ensure it wasn't a bad device.

/etc/modules contains:
```
snd-bcm2835
fbtft dma
fbtft_device name=rpi-display rotate=270
stmpe_ts
ads7846
ads7846_device model=7846 cs=0 gpio_pendown=25 speed=2000000 keep_vref_on=1 x_plate_ohms=60 pressure_max=255 x_min=200 x_max=3900 y_min=200 y_max=3900
```

I also tried commenting ads7846 and ads7846_device out and instead adding another dtoverlay line to the /boot/config.tst for the touchscreen:

```
dtoverlay=ads7846,penirq=25,pmax=255,ymin=190,ymax=3850,xmax=3850,xmin=230,cs=0,swapxy=0,xohms=60,penirq_pull=2
```

That however creates a conflict with the display, and dmesg has several errors:

```
[    2.916135] spi-bcm2835 3f204000.spi: chipselect 0 already in use
[    2.916158] spi_master spi0: spi_device register error /soc/spi@7e204000/rpi-display@0
[    3.374979] ads7846 spi0.1: failed to request/setup pendown GPIO25: -16
```

notro, 2016-02-20:
> spi-bcm2835 3f204000.spi: chipselect 0 already in use

The display already uses spi0.0, so cs=0 won't work.

-16 is -EBUSY which means someone has claimed the gpio already.
The rpi-display overlay adds an ads7846 device so when you try to add another that's probably why it fails.

You can look at /proc/interrupts to verify that touch interrupts are working.

https://github.com/raspberrypi/linux/blob/rpi-4.1.y/arch/arm/boot/dts/overlays/rpi-display-overlay.dts

alpharesearch, 2016-06-18:
> dtparam=spi=on
dtparam=i2c1=on
dtparam=i2c_arm=on
dtoverlay=rpi-display,rotate=90,fps=20

I did add this to my boot.txt, looks like the light turns on but I only get a black screen... what other steps are needed to use this display?

alpharesearch, 2016-06-22:
If I just have this line:
> dtoverlay=ads7846,penirq=25,xohms=100

and uncomment this line like this:

> #dtoverlay=rpi-display,rotate=90,fps=20

The touchscreen starts showing values with evtest but no screen output. But as soon as I take the # out the screen starts working but the touch part stops working. So I guess I need to use the integrated ads7846 driver from the rpi-display overlay... but if I just have the second line evtest doesn't detect any touch screen? 

```
[    4.154813] ads7846 spi0.1: touchscreen, irq 191
[    4.155691] input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input1
[    4.158455] spi-bcm2835 3f204000.spi: chipselect 1 already in use
[    4.158481] spi_master spi0: spi_device register error /soc/spi@7e204000/ads7846@1
[    4.158516] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/ads7846@1
[    4.159682] bcm2708_i2c 3f804000.i2c: BSC1 Controller at 0x3f804000 (irq 83) (baudrate 100000)
[    4.194955] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    4.297831] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    4.310354] fb_ili9341: module is from the staging directory, the quality is unknown, you have been warned.
[    4.311197] pinctrl-bcm2835 3f200000.gpio: pin gpio18 already requested by 3f203000.i2s; cannot claim for spi0.0
[    4.311216] pinctrl-bcm2835 3f200000.gpio: pin-18 (spi0.0) status -22
[    4.311232] pinctrl-bcm2835 3f200000.gpio: could not request pin 18 (gpio18) from group gpio18  on device pinctrl-bcm2835
[    4.311245] fb_ili9341 spi0.0: Error applying setting, reverse things back
```

alpharesearch, 2016-06-22:
I modified and merged the two overlays together, the touch screen is working but the X axis is flipped. If I touch on the right side the mouse pointer jumps to the left site and vise versa. Any idea how to turn just the X axis around without changing the Y axis?

>dtoverlay=pishow28,rotate=90,xmin=300,xmax=3800,ymin=700,ymax=3400,pmax=255,xohms=60,swapxy

```
/*
 * Device Tree overlay for pishow 2.8
 *
 */

/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			status = "okay";

			spidev@0{
				status = "disabled";
			};

			spidev@1{
				status = "disabled";
			};
		};
	};

	fragment@1 {
		target = <&gpio>;
		__overlay__ {
			rpi_display_pins: rpi_display_pins {
				brcm,pins = <18 23 24 25>;
				brcm,function = <1 1 1 0>; /* out out out in */
				brcm,pull = <0>; /* none */
			};
		};
	};

	fragment@2 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			rpidisplay: rpi-display@0{
				compatible = "ilitek,ili9341";
				reg = <0>;
				pinctrl-names = "default";
				pinctrl-0 = <&rpi_display_pins>;

				spi-max-frequency = <32000000>;
				rotate = <270>;
				bgr;
				fps = <30>;
				buswidth = <8>;
				reset-gpios = <&gpio 23 0>;
				dc-gpios = <&gpio 24 0>;
				led-gpios = <&gpio 18 1>;
				debug = <0>;
			};

			rpidisplay_ts: rpi-display-ts@1 {
				compatible = "ti,ads7846";
				reg = <1>;

				spi-max-frequency = <2000000>;
				interrupts = <25 2>; /* high-to-low edge triggered */
				interrupt-parent = <&gpio>;
				pendown-gpio = <&gpio 25 0>;
				/* driver defaults */
				ti,x-min = /bits/ 16 <0>;
				ti,y-min = /bits/ 16 <0>;
				ti,x-max = /bits/ 16 <0x0FFF>;
				ti,y-max = /bits/ 16 <0x0FFF>;
				ti,pressure-min = /bits/ 16 <0>;
				ti,pressure-max = /bits/ 16 <0xFFFF>;
				ti,x-plate-ohms = /bits/ 16 <400>;
			};
		};
	};
	__overrides__ {
		speed =   <&rpidisplay>,"spi-max-frequency:0";
		rotate =  <&rpidisplay>,"rotate:0";
		fps =     <&rpidisplay>,"fps:0";
		debug =   <&rpidisplay>,"debug:0";
		swapxy =  <&rpidisplay_ts>,"ti,swap-xy?";
		xmin =    <&rpidisplay_ts>,"ti,x-min;0";
		ymin =    <&rpidisplay_ts>,"ti,y-min;0";
		xmax =    <&rpidisplay_ts>,"ti,x-max;0";
		ymax =    <&rpidisplay_ts>,"ti,y-max;0";
		pmin =    <&rpidisplay_ts>,"ti,pressure-min;0";
		pmax =    <&rpidisplay_ts>,"ti,pressure-max;0";
		xohms =   <&rpidisplay_ts>,"ti,x-plate-ohms;0";
	};
};

```

notro, 2016-06-22:
The driver doesn't support inverting axis: https://www.kernel.org/doc/Documentation/devicetree/bindings/input/ads7846.txt
You need to do it in userspace: https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian#displays-with-a-touchpanel

alpharesearch, 2016-06-23:
@notro Thanks for your hints, this helped me get the touch working now. The pishow 2.8 unit uses a different pin:

```
                interrupts = <22 2>; /* high-to-low edge triggered */
                interrupt-parent = <&gpio>;
                pendown-gpio = <&gpio 22 0>;
```

There are 4 push buttons on the unit, is there any way to add the buttons to the device Tree overlay file as normal keyboard buttons?

notro, 2016-06-23:
See this example: https://github.com/raspberrypi/linux/blob/rpi-4.4.y/arch/arm/boot/dts/overlays/tinylcd35-overlay.dts

alpharesearch, 2016-06-23:
@notro thanks again for the example and your support. Just as a reference here is a fully working file for the PiShow 2.8 touch display:
```
/*
 * Device Tree overlay for pishow 2.8
 *
 */

/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

    fragment@0 {
	target = <&spi0>;
	__overlay__ {
	    status = "okay";

	    spidev@0{
		status = "disabled";
	    };

	    spidev@1{
		status = "disabled";
	    };
	};
    };

    fragment@1 {
	target = <&gpio>;
	__overlay__ {
	    pishow_display_pins: pishow_display_pins {
		brcm,pins = <18 23 24 25>;
		brcm,function = <1 1 1 0>; /* out out out in */
		brcm,pull = <0>; /* none */
	    };
            pishow_ts_pins: pishow_ts_pins {
		brcm,pins = <22>;
		brcm,function = <0>; /* in */
	    };
	    keypad_pins: keypad_pins {
		brcm,pins = <16 20 21 26>;
		brcm,function = <0>; /* in */
		brcm,pull = <1>; /* down */
	    };
	};
    };

    fragment@2 {
	target = <&spi0>;
	__overlay__ {
	    /* needed to avoid dtc warning */
	    #address-cells = <1>;
	    #size-cells = <0>;

	    pishow28: rpi-display@0{
		compatible = "ilitek,ili9341";
		reg = <0>;
		pinctrl-names = "default";
		pinctrl-0 = <&pishow_display_pins>,
			    <&pishow_ts_pins>;

		spi-max-frequency = <32000000>;
		rotate = <270>;
		bgr;
		fps = <30>;
		buswidth = <8>;
		reset-gpios = <&gpio 23 0>;
		dc-gpios = <&gpio 24 0>;
		led-gpios = <&gpio 18 1>;
		debug = <0>;
	    };

	    pishow28_ts: rpi-display-ts@1 {
		compatible = "ti,ads7846";
		reg = <1>;

		spi-max-frequency = <2000000>;
		interrupts = <22 2>; /* high-to-low edge triggered */
		interrupt-parent = <&gpio>;
		ti,keep-vref-on = <1>;
		pendown-gpio = <&gpio 22 0>;
		/* driver defaults */
		ti,x-min = /bits/ 16 <230>;
		ti,y-min = /bits/ 16 <200>;
		ti,x-max = /bits/ 16 <3900>;
		ti,y-max = /bits/ 16 <3700>;
		ti,pressure-min = /bits/ 16 <0>;
		ti,pressure-max = /bits/ 16 <0xFFFF>;
		ti,x-plate-ohms = /bits/ 16 <80>;
		ti,swap-xy = <1>;
	    };
	};
    };
    fragment@3 {
	target-path = "/soc";
	__overlay__ {
	    keypad: keypad {
		compatible = "gpio-keys";
		#address-cells = <1>;
		#size-cells = <0>;
		pinctrl-names = "default";
		pinctrl-0 = <&keypad_pins>;
		#status = "disabled";
		autorepeat;
		button@16 {
		    label = "GPIO KEY_UP";
		    linux,code = <103>;
		    gpios = <&gpio 16 0>;
		};
		button@26 {
		    label = "GPIO KEY_DOWN";
		    linux,code = <108>;
		    gpios = <&gpio 26 0>;
		};
		button@20 {
		    label = "GPIO KEY_LEFT";
		    linux,code = <105>;
		    gpios = <&gpio 20 0>;
		};
		button@21 {
		    label = "GPIO KEY_RIGHT";
		    linux,code = <106>;
		    gpios = <&gpio 21 0>;
		};
	    };
	};
    };
    __overrides__ {
	speed =   <&pishow28>,"spi-max-frequency:0";
	rotate =  <&pishow28>,"rotate:0";
	fps =     <&pishow28>,"fps:0";
	debug =   <&pishow28>,"debug:0";
	swapxy =  <&pishow28_ts>,"ti,swap-xy;0";
	xmin =    <&pishow28_ts>,"ti,x-min;0";
	ymin =    <&pishow28_ts>,"ti,y-min;0";
	xmax =    <&pishow28_ts>,"ti,x-max;0";
	ymax =    <&pishow28_ts>,"ti,y-max;0";
	pmin =    <&pishow28_ts>,"ti,pressure-min;0";
	pmax =    <&pishow28_ts>,"ti,pressure-max;0";
	xohms =   <&pishow28_ts>,"ti,x-plate-ohms;0";
	#keypad =  <&keypad>,"status";
    };
};
```
Plus the /etc/X11/xorg.conf.d/99-ads7846-cal.conf file:
```
Section "InputClass"
    Identifier "Touchscreen"
    MatchProduct "ADS7846 Touchscreen"
    Option "EmulateThirdButton" "1"
    Option "EmulateThirdButtonButton" "3"
    Option "EmulateThirdButtonTimeout" "1500"
    Option "EmulateThirdButtonMoveThreshold" "30"
    Option "InvertX" "1"
    Option "InvertY" "1"
    Option "SwapAxes" "0"
    Option "Calibration" "215 3800 129 3676"
EndSection
```
Please add what every you want to the official library if you can. Thanks.

closed 2017-02-22 by notro