notro/fbtft #380: Having an issue accessing fbtft drivers

camnauman, 2016-02-27:
I have a Raspberry Pi Model B and have been trying for a week to be able to use my adafruit 1.8 tft screen. From my understanding notro fbtft drivers have been added into the rpi-update, but I still cannot figure out how to get my Pi to access the drivers. Any help would be greatly appreciated.

Ardillo, 2016-02-28:
Well I'm busy with the same topic.

First do the update

`wget http://goo.gl/1BOfJ -O /usr/bin/rpi-update && chmod +x /usr/bin/rpi-update`
`rpi-update`

Then uncomment the line in /boot/config.txt to enable SPI. I'm using Arch Linux, so this could be a bit different.
`device_tree_param=spi=on`

Make sure that the following is in /etc/modules-load.d/raspberrypi.conf
`spi-bcm2835`

After a reboot you should be able to load the module using:
`modprobe fbtft_device name=adafruit18`
I'm using a sainsmart18, but for the basics they are the same I guess.

This worked for me.

camnauman, 2016-02-29:
So I did all this and am just getting a white screen on the TFT I have checked to make sure the pins are connected correctly. What should the screen by outputting?

Ardillo, 2016-03-01:
Ok, that's  normal

If everything goes right you will have a device called **fb1** in /dev

You can test it by following: https://github.com/notro/fbtft-spindle/wiki/FBTFT-image

`sudo apt-get install -y mplayer`
`wget http://fredrik.hubbe.net/plugger/test.mpg`
`mplayer -nolirc -vo fbdev2:/dev/fb1 -vf scale=320:-3 test.mpg`

There is a lot more to find on how to redirect a terminal to that fb1 device.

closed 2017-02-22 by notro