notro/fbtft #382: Pi Zero and freetronicsoled128

multivac61, 2016-03-10:
I´m having issues using the SPI bus with fbtft on my Pi Zero to control the Freetronics OLED display. I followed the tutorial from [here](http://www.freetronics.com.au/pages/oled128-quickstart-guide-raspberry-pi#.VuGxxkLEQzZ) without the old kernel update trick etc.

When looking at both the MOSI and SCLK signals coming from the Zero I can see that they are both very distorted when running `FRAMEBUFFER=/dev/fb1 startx`. When I however run the SPI test code from [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md#troubleshooting) I get much cleaner signals. 

Could the issue be related to the drivers?

notro, 2016-03-12:
Try lowering the speed (default 20MHz): ```sudo modprobe fbtft_device name=freetronicsoled128 speed=8000000```
The spidev test code runs at 500kHz.

multivac61, 2016-03-13:
Of course the distorted waveform had something to do with the SPI speed! Now I am able to update the screen flawlessly at 15MHz from the Pi zero. Thanks @notro!

closed 2017-02-22 by notro