notro/fbtft #384: HY28B rotate 270 not working

tpitman, 2016-03-30:
I have the latest Raspian Jessie and a HP28B display running on a raspberry pi.

The instructions on setting this up talk about the display will be oriented so that the hdmi connector will be at the top of the display.

I have rotate=90 on the dtoverlay command like the instructions say.

The display is upside down or rotated 180 from what I expect.

The touch screen is acting like the hdmi should be at the top so it is rotated 180 from the display.

I have tried changing 90 to 270 and the display does not change at all, so either 90 is not working (most likely) or 270 is not working.

Please advise.

Texy, 2016-04-02:
I agree - with the latest raspbian (of 2016-03-18) the display is fixed at 270 degrees, ie with the text 'upside down' when the hdmi connector is at the top. It was working with previous raspbian releases.
It does not appear to be a problem with the .dtb file as I copied from an earlier version - actually it appears to be stable from August 2015.
Texy

notro, 2016-04-02:
Debugging aids:
The value that made it into fbtft: /sys/class/graphics/fb1/rotate
The value that made it into the dtb: /proc/device-tree/soc/spi@7e204000/<something>/rotate
Firmware log (DT overlay): sudo vcdbg log msg

tpitman, 2016-04-04:
Does any of this give a solution to the rotation problem or am I still
waiting on something?

notro, 2016-04-04:
It's not a solution but things to look at to find out where the rotation value gets lost.
This should give you the hex value of the rotation parameter that fbtft is reading from the Device Tree:
```
xxd -ps /proc/device-tree/soc/spi@7e204000/hy28b@0/rotation
```
I haven't got the DT overlay loaded, so the path could be wrong.

tpitman, 2016-04-04:
Are you guys also able to reproduce this issue and are working on it or are
you relying on me to give this debug information so you can solve it? I am
happy to provide it, but don't want to duplicate if you are already doing
it.

notro, 2016-04-04:
I haven't got time to look into it now

Texy, 2016-04-05:
Do you mean 'rotate', not 'rotation'? If so, this is the response
pi@raspberrypi:~ $ xxd -ps /proc/device-tree/soc/spi@7e204000/hy28b@0/rotate
0000010e

Also
pi@raspberrypi:~ $ cat /sys/class/graphics/fb1/rotate
270

This might point to the problem, via sudo vcdbg log msg

001475.516: Loaded overlay 'hy28b'
001475.533: dtparam:  rotate=270
001478.162: Unknown dtparam ' rotate' - ignored

On the assumption that fbtft and the hy28b overlay file have both been stable for previous releases, I would assume this is a raspbian issue.....
Note that hy28a also brings up the same issue
001475.387: Loaded overlay 'hy28a'
001475.403: dtparam:  rotate=270
001478.031: Unknown dtparam ' rotate' - ignored

Needs reporting via github?
Texy

Texy, 2016-04-05:
OK, so I found the 'problem' - it doesn't like the space before the rotate parameter!
This is bad
dtoverlay=hy28b, rotate=90
This is good
dtoverlay=hy28b,rotate=90

I'm not sure if this was always like this.
Texy

Texy, 2016-04-05:
This thread explains it 
https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=139732

Texy

tpitman, 2016-04-05:
I do not have a space in front of rotate and yes I do mean rotate. Any
other things to try?

tpitman, 2016-04-05:
I did the sudo vcdbg log msg and I don't have the error rotate unknown.

I do have an unknown like this:

Unknown dtparam 'arm_freq' - ignored

But nothing else unknown.

My dtoverlay line in the /boot/config.txt is simply this:

dtoverlay=hy28b,rotate=270

nothing more nothing less no spaces.

Please advise.

Texy, 2016-04-05:
The rotation/rotate question was aimed for notro, not you.
So you are using the Raspbian Jessie image of 2016-03-18 ?
Not not via NOOBS?

Maybe deleted the complete dtoverlay line and manually copy & paste from my setup guide here 
https://www.raspberrypi.org/forums/viewtopic.php?f=45&t=105797&p=729810

Texy

tpitman, 2016-04-05:
I started with the noobs SD and then patched using this link:

https://www.raspberrypi.org/forums/viewtopic.php?f=45&t=105797

tpitman, 2016-04-05:
LOL, just realized that is the same page.

Texy, 2016-04-05:
I've never used NOOBS, I only use Raspbian. Maybe it's a NOOBS issue as it uses a different config file and method, I think, (I could be wrong).
Texy

tpitman, 2016-04-05:
No way. copy and paste worked. I tried typing it again slowly and still
didn't work. Must be magic in the copied text.

Texy, 2016-04-05:
Yeah - that was an option, that's why I suggested it.
Glad you got there in the end.
Texy

tpitman, 2016-04-05:
Your support is great. Thank you.

closed 2017-02-22 by notro