notro/fbtft #385: add to fbtft_device.c data about waveshare 3.5 spotbear from this working snippet

gwpl, 2016-04-04:
Hi!

Could someone make a patch adding data about waveshare35a (Waveshare SpotBear 3.5" LCD) to fbtft_device.c , based on this working (yes, I checked, and it also works for me) snippet:

https://archlinuxarm.org/forum/viewtopic.php?f=31&t=9208

Thanks!
Best regards,
Grzegorz

notro, 2017-02-22:
Closing issue since there has been no activity in a long time. Reopen if needed.

closed 2017-02-22 by notro

gwpl, 2017-02-23:
Was this configuration actually add to fbtft_device.cc file?

I keep looking forward officiall support for this class of displays (I own waveshare displays in multiple sizes).

notro, 2017-02-25:
No, only 22 and 32b are supported by https://github.com/torvalds/linux/blob/master/drivers/staging/fbtft/fbtft_device.c

If you're on a Pi, I suggest you use a Device Tree overlay instead: https://github.com/swkim01/waveshare-dtoverlays

gwpl, 2017-02-25:
Sounds good to me - i.e. good advice! Thanks!