notro/fbtft #387: No fb1 in /dev

camnauman, 2016-04-19:
I am trying to get a adafruit 1.8 screen to work. I am following these instructions and have had it work before. However I did a new install and after following these update steps there is no fb1 in my /dev folder. I was under the impression that it was created when the rpi is updated. Here are my update steps that I have done so far.

wget http://goo.gl/1BOfJ -O /usr/bin/rpi-update && chmod +x /usr/bin/rpi-update
rpi-update
and I have also done rpi-upgrade

Any help would be greatly appreciated!

torch2424, 2016-05-04:
I am also having the same issue actually. Any help would be much appreciated!

torch2424, 2016-05-04:
I can confirm fbtft is being loaded by my dmesg:

````
[    3.587803] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    3.594112] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[    3.596522] fbtft_device:  SPI devices registered:
[    3.596558] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[    3.596572] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[    3.596582] fbtft_device:  'fb' Platform devices registered:
[    3.596618] fbtft_device:      soc:fb id=-1 pdata? no
[    3.596685] fbtft_device: Deleting spi0.0
[    3.598530] spi spi0.0: setting up native-CS0 as GPIO 8
[    3.598674] fbtft_device:  GPIOS used by 'fb_ili9341':
[    3.598691] fbtft_device:    'reset' = GPIO23
[    3.598701] fbtft_device:    'led' = GPIO24
[    3.598709] fbtft_device:    'dc' = GPIO25
[    3.598717] fbtft_device:  SPI devices registered:
[    3.598730] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[    3.598741] fbtft_device:      fb_ili9341 spi0.0 32000kHz 8 bits mode=0x00
[    3.612300] fb_ili9341: module is from the staging directory, the quality is unknown, you have been warned.
```

but when I run ls /dev i get the following:

````
autofs
block
bus
cachefiles
char
console
cpu_dma_latency
disk
fb0
fd
full
gpiomem
hwrng
initctl
input
kmsg
log
loop-control
loop0
loop1
loop2
loop3
loop4
loop5
loop6
loop7
mem
memory_bandwidth
mmcblk0
mmcblk0p1
mmcblk0p2
mmcblk0p5
mmcblk0p6
mmcblk0p7
mmcblk0p8
network_latency
network_throughput
null
ptmx
pts
ram0
ram1
ram10
ram11
ram12
ram13
ram14
ram15
ram2
ram3
ram4
ram5
ram6
ram7
ram8
ram9
random
raw
shm
snd
spidev0.1
stderr
stdin
stdout
tty
tty0
tty1
tty10
tty11
tty12
tty13
tty14
tty15
tty16
tty17
tty18
tty19
tty2
tty20
tty21
tty22
tty23
tty24
tty25
tty26
tty27
tty28
tty29
tty3
tty30
tty31
tty32
tty33
tty34
tty35
tty36
tty37
tty38
tty39
tty4
tty40
tty41
tty42
tty43
tty44
tty45
tty46
tty47
tty48
tty49
tty5
tty50
tty51
tty52
tty53
tty54
tty55
tty56
tty57
tty58
tty59
tty6
tty60
tty61
tty62
tty63
tty7
tty8
tty9
ttyAMA0
ttyprintk
urandom
vc-cma
vc-mem
vchiq
vcio
vcs
vcs1
vcs3
vcsa
vcsa1
vcsa3
vcsm
zero
````

camnauman, 2016-05-04:
You have to go into the fbturbo 99 file in /usr/share/X11/xorg.conf.d
And in there is the fbturbo and change fb0 to fb1 and it should work

torch2424, 2016-05-04:
@camnauman I actually dont have that file. I am testing this out on https://github.com/recalbox/recalbox-os , which is essentially retropie but with support for a few more things. 

I cant find how they are doing they're x server, perhaps it has something to do with NOOBS?

But after working on it a bit more. I finally got the fb1 folder to show, but I had to use a dtoverlay in the /boot/config.txt to get it to appear. I compled my own to correspond with the reset,led,and dc gpio pins. And using the following command:

````
modprobe fbtft_device custom name=fb_ili9341 rotate=90 speed=32000000 fps=50 bgr=1 gpios=reset:23,led:24,dc:25
````

as root within a init script does NOT show fb1.

The screen I've been trying to use is this one: http://marcosgildavid.blogspot.com/2014/02/getting-ili9341-spi-screen-working-on.html

camnauman, 2016-05-04:
If you don't have that file do rpi-update. It includes notro's drivers
which includes the screen your using from what I can see. Take a look at
notro tft wiki there's a lot of info there. I may have told you the wrong
directory for the fbturbo file. I can get back to you tomorrow with the
exact directory

http://marcosgildavid.blogspot.com/2014/02/getting-ili9341-spi-screen-working-on.html

torch2424, 2016-05-04:
@camnauman I'll try that, thank you!

closed 2017-02-22 by notro