notro/fbtft #388: ArchLinux ARM linux-raspberrypi 4.4.7-4 breaks pitft

visibilityspots, 2016-04-20:
I'm running a pi with the ArchLinux ARM distro on it. When I upgrade to the latest kernel my display doesn't comes up anymore
```
==> Software upgrade (new version) :
core/linux-raspberrypi 4.1.8-1 -> 4.4.7-4
```
Is this a known issue or am I overlooking something?

HorstBaerbel, 2016-05-12:
Same here on a recent Raspbian with 4.4.9+ kernel

notro, 2016-05-12:
There has been changes to the DT overlay mechanism. No spaces allowed in dtoverlay=
I assume theres no /dev/fb1?
See if this shows any errors: sudo vcdbg log msg

visibilityspots, 2016-05-12:
in the /boot/config.txt I double checked the spaces and couldn't find any:

```bash
dtparam=spi=on,i2c_arm=on
dtoverlay=waveshare35a,w1-gpio-pullup,gpiopin=4,extpullup=1
```

as you assumed correctly no /dev/fb1 available

```bash
sudo ls /dev/fb1              
ls: cannot access '/dev/fb1': No such file or directory
```

and there are indeed some errors in the vcdbg log:

```bash
sudo /opt/vc/bin/vcdbg log msg
[sudo] password for jan: 
000877.877: HDMI:EDID error reading EDID block 0 attempt 0
000879.141: HDMI:EDID error reading EDID block 0 attempt 1
000880.400: HDMI:EDID error reading EDID block 0 attempt 2
000881.657: HDMI:EDID error reading EDID block 0 attempt 3
000882.915: HDMI:EDID error reading EDID block 0 attempt 4
000884.173: HDMI:EDID error reading EDID block 0 attempt 5
000885.431: HDMI:EDID error reading EDID block 0 attempt 6
000886.689: HDMI:EDID error reading EDID block 0 attempt 7
000887.946: HDMI:EDID error reading EDID block 0 attempt 8
000889.204: HDMI:EDID error reading EDID block 0 attempt 9
000890.226: HDMI:EDID giving up on reading EDID block 0
000892.914: HDMI:Setting property pixel encoding to Default
000892.934: HDMI:Setting property pixel clock type to PAL
000892.954: HDMI:Setting property content type flag to No data
000892.974: HDMI:Setting property fuzzy format match to enabled
001099.388: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001099.412: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
001099.801: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001101.070: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001102.334: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001103.597: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001104.860: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001106.123: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001107.386: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001108.649: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001109.912: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001111.175: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001112.201: hdmi: HDMI:EDID giving up on reading EDID block 0
001112.232: hdmi: HDMI: No lookup table for resolution group 0
001112.248: hdmi: HDMI: hotplug attached with DVI support
001112.309: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001112.675: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
001113.936: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
001115.199: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
001116.461: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
001117.723: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
001118.985: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
001120.247: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
001121.509: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
001122.771: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
001124.033: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
001125.058: hdmi: HDMI:EDID giving up on reading EDID block 0
001125.121: hdmi: HDMI: hotplug deassert
001125.136: hdmi: HDMI: HDMI is currently off
001125.151: hdmi: HDMI: changing mode to unplugged
001125.172: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
001125.962: *** Restart logging
001126.901: Read command line from file 'cmdline.txt'
dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 fsck.mode=force root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rw rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
001498.001: Loading 'kernel7.img' to 0x8000 size 0x5e6aa4
001501.641: Kernel trailer DTOK property says yes
001501.657: Kernel trailer DDTK property says yes
001504.168: Loading 'bcm2709-rpi-2-b.dtb' to 0x5eeaa4 size 0x327a
001569.160: dtparam: pwr_led_gpio=35
001576.836: dtparam: uart0_clkrate=3000000
001587.795: dtparam: spi=on
001594.402: dtparam: i2c_arm=on
001605.811: Failed to load overlay 'waveshare35a'
001665.382: dtparam: arm_freq=700000000
001702.170: dtparam: core_freq=250000000
001711.636: dtparam: cache_line_size=64
001732.905: Device tree loaded to 0x2fffc800 (size 0x37fa)
001736.110: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
006630.367: vchiq_core: vchiq_init_state: slot_zero = 0xf6c80000, is_master = 1
006634.335: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
```

HorstBaerbel, 2016-05-12:
Same "Failed to load overlay 'waveshare35a'" here. I got my dts/dtb from [here](https://github.com/swkim01/waveshare-dtoverlays). They worked for the 4.1 kernel.

notro, 2016-05-12:
Have you done this from the README:

or if linux 4.4 kernel or newer,
```
sudo cp waveshare-dtoverlays/waveshare32b-overlay.dtb /boot/overlays/waveshare32b.dtbo
```

visibilityspots, 2016-05-12:
Alright, copying over that overlay to the correct directory did the trick for me! I have a working display again with the latest kernel on arch linux ARM! 

Thanks a lot for helping me out.

HorstBaerbel, 2016-05-12:
I've tried this and it does NOT work for me. Still "Failed to load overlay 'waveshare35a'".

visibilityspots, 2016-05-12:
@HorstBaerbel could you maybe post the whole output of that vcdbg log msg command?

HorstBaerbel, 2016-05-12:
> 000726.392: HDMI:EDID version 1.3, 1 extensions, screen size 26x14 cm
000726.456: HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
000726.554: HDMI:EDID found preferred DMT detail timing format: 1366x768p @ 60 Hz (86)
000726.653: HDMI:EDID standard timings block x 8: 0x0101 0101 0101 0101 0101 0101 0101 0101 
000739.659: HDMI:EDID parsing v3 CEA extension 0
000739.698: HDMI:EDID monitor support - underscan IT formats:no, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:1
000739.758: HDMI:EDID found DMT detail timing format: 1366x768p @ 60 Hz (86)
000739.815: HDMI:EDID found DMT detail timing format: 1366x768p @ 60 Hz (86)
000739.896: HDMI:EDID found unknown detail timing format: 0x1080i hfp:88 hs:44 hbp:-116 vfp:2 vs:5 vbp:16 pixel clock:0 MHz
000739.930: HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz 
000739.964: HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz 
000739.997: HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz 
000740.058: HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48|96|192 kHz, sample size: 16|20|24 bits
000740.076: HDMI:EDID found HDMI VSDB length 5
000740.103: HDMI:EDID HDMI VSDB has physical address 1.0.0.0
000740.119: HDMI:EDID HDMI VSDB has no extension fields
000740.163: HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
000740.290: HDMI:EDID preferred mode is updated to DMT (2) 640x400p @ 85 Hz with pixel clock 31500000 Hz
000743.520: HDMI:Setting property pixel encoding to Default
000743.538: HDMI:Setting property pixel clock type to PAL
000743.559: HDMI:Setting property content type flag to No data
000743.577: HDMI:Setting property fuzzy format match to enabled
000781.505: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
000793.668: hdmi: HDMI:EDID version 1.3, 1 extensions, screen size 26x14 cm
000793.738: hdmi: HDMI:EDID features - videodef 0x80 !standby !suspend !active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
000793.841: hdmi: HDMI:EDID found preferred DMT detail timing format: 1366x768p @ 60 Hz (86)
000793.943: hdmi: HDMI:EDID standard timings block x 8: 0x0101 0101 0101 0101 0101 0101 0101 0101 
000806.950: hdmi: HDMI:EDID parsing v3 CEA extension 0
000806.991: hdmi: HDMI:EDID monitor support - underscan IT formats:no, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:1
000807.052: hdmi: HDMI:EDID found DMT detail timing format: 1366x768p @ 60 Hz (86)
000807.114: hdmi: HDMI:EDID found DMT detail timing format: 1366x768p @ 60 Hz (86)
000807.203: hdmi: HDMI:EDID failed to find a matching detail format for 0x1080i hfp:88 hs:44 hbp:-116 vfp:2 vs:5 vbp:16 pixel clock:0 MHz
000807.226: hdmi: HDMI:EDID calculated refresh rate is 0 Hz
000807.256: hdmi: HDMI:EDID guessing the format to be 0x1080i @24 Hz
000807.334: hdmi: HDMI:EDID found unknown detail timing format: 0x1080i hfp:88 hs:44 hbp:-116 vfp:2 vs:5 vbp:16 pixel clock:0 MHz
000807.374: hdmi: HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz 
000807.408: hdmi: HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz 
000807.444: hdmi: HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz 
000807.507: hdmi: HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48|96|192 kHz, sample size: 16|20|24 bits
000807.530: hdmi: HDMI:EDID found HDMI VSDB length 5
000807.559: hdmi: HDMI:EDID HDMI VSDB has physical address 1.0.0.0
000807.578: hdmi: HDMI:EDID HDMI VSDB has no extension fields
000807.623: hdmi: HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
000807.755: hdmi: HDMI:EDID preferred mode is updated to DMT (2) 640x400p @ 85 Hz with pixel clock 31500000 Hz
000807.779: hdmi: HDMI: hotplug attached with HDMI support
000807.850: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
000810.338: hdmi: HDMI: power_on to DMT mode 640x400@85
000813.510: hdmi: HDMI: Action callback added to queue to happen at frame 0
000825.081: *** Restart logging
000826.361: Read command line from file 'cmdline.txt'
dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=font:ProFont6x11 logo.nologo consoleblank=0
001071.916: Loading 'kernel.img' to 0x8000 size 0x3edbd0
001075.590: Kernel trailer DTOK property says yes
001075.604: Kernel trailer DDTK property says yes
001078.125: Loading 'bcm2708-rpi-b-plus.dtb' to 0x3f5bd0 size 0x2e3f
001119.380: dtparam: uart0_clkrate=3000000
001128.121: dtparam: spi=on
001146.905: Failed to load overlay 'waveshare35a'
001166.628: Loaded overlay 'ads7846'
001166.643: dtparam: cs=1
001167.603: dtparam: penirq=17
001169.814: dtparam: penirq_pull=2
001170.572: dtparam: speed=1000000
001171.561: dtparam: keep_vref_on=1
001173.663: Unknown dtparam 'keep_vref_on' - ignored
001173.675: dtparam: swa=true
001175.782: Unknown dtparam 'swa' - ignored
001175.822: dtparam: i2c_arm=on
001231.343: Loaded overlay 'w1-gpio-pullup'
001231.361: dtparam: gpiopin=4
001232.062: dtparam: extpullup=1
001282.321: dtparam: arm_freq=700000000
001284.374: Unknown dtparam 'arm_freq' - ignored
001284.410: dtparam: core_freq=250000000
001291.273: dtparam: cache_line_size=32
001305.852: Device tree loaded to 0x17fec900 (size 0x36be)
001307.115: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
003815.564: vchiq_core: vchiq_init_state: slot_zero = 0x57880000, is_master = 1
003819.693: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
003825.071: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000002

HorstBaerbel, 2016-05-12:
The touchscreen works btw (cat /dev/input/mouse0 displays garbage when touching it), so SPI-communication "should" work.

HorstBaerbel, 2016-05-17:
Hmm. I suspect it has nothing to do with the .dtb/.dtbo, because somebody else is using it with success. I'm using a RPi zero. Could that be a problem? If I can do something else to debug the problem, let me know...

notro, 2016-05-21:
> 001146.905: Failed to load overlay 'waveshare35a'

Do you have /boot/overlay/waveshare35a.dtbo ?
The DT overlay mechanism has changed recently, so previously it would have been waveshare35a-overlay.dtb

HorstBaerbel, 2016-05-23:
I can check in Wednesday earliest, but is the name important? I understand the extension has to be dtbo, but is the file name important?! I am pretty sure the file is \boot\overlays\waveshare35a.dtbo though. Will check when I get back home.
I read a bit about debugging Kernel overlays, but it sounded like I wouldn't want to go there... ;)

HorstBaerbel, 2016-05-24:
Wow. That was stupid. I renamed waveshare35a-overlay.dtbo to waveshare35a.dtbo and voila, it works. I was totally blind to that name thing. Sorry...

closed 2016-05-25 by visibilityspots