notro/fbtft #391: fbtft and parallel RGB of an imx6

notro, 2016-05-10:
@notro do you have any recommendations on where to look for a jump start on interfacing a TFT with the Parallel RGB of an imx6. we're looking to add a custom display to our linux device...seems like we could use a lot of the elements of FBTFT but you're messages suggest abandoning that as the maintainer is not accepting any more drivers.

[moved from https://github.com/notro/tinydrm/issues/2]

notro, 2016-05-10:
I assume your talking MIPI DBI type (controller with onboard display ram) here and not DPI (pixel clocked) which fbtft doesn't support.
I haven't talked about tinydrm with Greg, the staging maintainer, so he would probably accept a new fbtft driver (4.5 had no fbtft patches, but 4.6 had). But fbtft will eventually be removed, it's just not known when that will happen.
I'm not sure what to do with the DBI interface in tinydrm, because it would probably require a new subsystem for an intel808x type bus, which is a lot of work to get accepted. And DBI is slooow when driven with gpios, but the Pi for instance has hardware support for DBI so it could be sped up.

So my question is doesn't imx6 have a DPI interface? That would give a much better performance using almost the same amount of pins.

beardedbadger, 2016-05-10:
Thanks @notro….any thoughts good places to start with implementing the parallel rgb? Seems like it might be reasonable to just expand on what you’ve done here.

beardedbadger, 2016-05-10:
@notro Yes, in the case of the frame buffer question, its is re a LCD display with onboard ram.  The screens we are evaluating for prototypes all have onboard ram w controller (as most TFT modules do) which means it would be straightforward to implement a frame buffer driver.  

With that said, the iMX6 has the parallel rgb interface which allows you to take advantage of the IPU blocks on the SoC, hence our interest in connecting to it for its increased graphics performance.  

You are correct, the iMX6 does support a DPI mode.  The challenge that we're working with is how to tie that path to one of these displays. 

Having dug even deeper into the common display controllers,  it appears that there are certain modes that could allow you to set these controllers up to work with an external pixel clock.  I believe this is what is meant by the **DE Mode** in the data sheet for the [ILI9431](http://www.displaytech-us.com/sites/default/files/driver-ic-data-sheet/Ilitek-ILI9341.pdf).  Would you have the same reading of that?

notro, 2016-05-12:
Many of these controllers are MIPI compliant.
The MIPI specs are not open, but you might be able to find them online.
http://mipi.org/specifications/display-interface

"7.6. Display Data Format" shows that it supports MCU (DBI) interface and RGB (DPI) interface.
I have no experience with DPI so I can't help you with that.

The benefit of DPI is that there's no need to sync the framebuffer with the display RAM, which takes time. How much depends on the number of pixels, bytes per pixel and bus width/speed.
https://github.com/notro/fbtft/wiki/Performance

Before fbtft went into mainline, it had optimized parallel write functions for the Pi. A driver can override all the default fbtft callbacks like write().
https://github.com/notro/fbtft/blob/master/fbtft-io.c#L133

This driver overrides several callbacks: http://lxr.free-electrons.com/source/drivers/staging/fbtft/fb_watterott.c

closed 2017-02-22 by notro