notro/fbtft #393: ILI9488

flexiti, 2016-05-19:
Its working  with ILI9488 8 bit paralel bus ?

http://www.aliexpress.com/snapshot/7471223026.html?orderId=73901012832947

flexiti, 2016-06-20:
Now I have connected this display using fbtft 
[image]

 
This is done as folows:
`sudo modprobe fbtft_device name=flexpfb rotate=180 gpios=dc:110,reset:21,wr:1,cs:6,db00:3,db01:68,db02:71,db03:64,db04:2,db05:65,db06:66,db07:67`

`sudo modprobe flexfb width=320 height=480 buswidth=8 init=-1,0x11,-2,200,-1,0x36,-1,0x3A,0x05,-1,0x36,0x48,-1,0x13,-1,0x29,-1,0x2C,-3`
and test picture:
`sudo fbi -d /dev/fb8 -T 1 -noverbose -a example.jpg`
Video:
```
sudo apt-get install -y mplayer

sudo SDL_VIDEODRIVER=fbcon SDL_FBDEV=/dev/fb8 mplayer -vo sdl -framedrop example.mpg
```
Why it is work only as fb8 not fb1 ?
Why it is extremaly slow  (video and picture also) ?  ( 1 frame pre second)
Some hint?
 
ps. Here is example of this display working directly usign WiringOP
 http://forum.armbian.com/index.php/topic/1363-395-inch-display-test/
[image]

flexiti, 2016-06-23:
I found that with this type of connection (GPIO only) and at different ports so it will be difficult to get more fps.
This LCD with fbtft is unfortunately slower than adopted WiringPO which I have included above. (Fbtft - 1fps, Wiring 2-5fps)
However, using fbtft has many advantages such as. Jpg easy viewing.
But I have one problem - the program with uses fbtft hangs up after an hour or more (the same program with WiringPO operates 24 hours a day).
Can I somehow find out what is causing the crashes ?
(Helps remove fbtft and load it again - my program works, only display stops)

notro, 2016-06-23:
fbtft and parallel gpio is slow. I guess WiringPO writes directly to the hw registers, hence the speedup, whereas fbtft uses gpiolib: http://lxr.free-electrons.com/ident?i=fbtft_write_gpio8_wr
Before fbtft went into mainline Linux, it had optimization for the Pi: https://github.com/notro/fbtft/blob/master/fbtft-io.c#L133

As to why it crashes, I don't know. Anything in the kernel log? dmesg
How do you know that it's the driver that stops working?
Have you tried to stop your application and put the console on it: con2fbmap 1 8

flexiti, 2016-06-24:
@notro 
Thanks for the tip, I'll need to check and tests. I'll let you know

ps.
After running Linux, if I write _con2fbmap 1 8_ nothing happens, only as I do reboot I see the final logs.
What should I do to see something on the lcd after entering con2fbmap ? ( except reboot)

notro, 2016-06-25:
I re-read the thread and you mention using fbi to put an image on the screen. This is a good way to test the driver.

flexiti, 2016-06-28:
I did a test like below, after aprox 20 min system hangs
It looks after the start that it is a memory leak or am I doing something wrong?

[image]
[image]
[image]
[image]
(jpg aprox every 5 min)

melanrz write to me:

> With fbtft hang (freeze) because use gpu,disable gpu and you can have no issue.

this could be the reason?

Test:

```
int main()
 {
    for(;;)
    {
          system("fbi -d /dev/fb8 -T 1 -noverbose -a /home/mariuszb/start1.jpg ");
          delay(4000);
          system("fbi -d /dev/fb8 -T 1 -noverbose -a /home/mariuszb/start2.jpg "); 
          delay(4000);
    }                
}         
```

(start1.jpg and start2.jpg  - two images with a resolution of 320x480)
Test on OrangePi PC, Armbian 5.13

fbtft.conf (in modprobe.d)
`options fbtft_device name=flexpfb rotate=180 fps=2 gpios=dc:110,reset:21,wr:1,cs:6,db00:3,db01:68,db02:71,db03:64,db04:2,db05:65,db06:66,db07:67`

flexfb.conf (in modprobe.d)

> options flexfb width=320 height=480 buswidth=8 init=-1,0x11,-2,200,-1,0x36,-1,0x3A,0x05,-1,0x36,0x48,-1,0x13,-1,0x29,-1,0x2C,-3

fbtft.conf (in modules-load.d)

> fbtft_device
> flexfb

flexiti, 2016-06-29:
@notro  thanks for help and fbtft

closed 2016-07-05 by flexiti