notro/fbtft #394: 3.5 Waveshare SpotPear: Led pin not set, exiting

O-Luhishi, 2016-05-20:
Hi there, fairly new to Linux but I've had some good experience messing around with the raspberry pi. I bought a 3.5 Waveshare SpotPear A v3 LCD screen. And at first I decided that I would not mind using the pre-installed image which was fine. However, I then later decided that I wanted to install the LCD screen on an image I currently use. So I went about and followed a fair few of the tutorials out there. 
With no luck, I wasn't able to do it (a while back so probably inexperience), however I then decided to try it this time again and I knew everything I did was correct however whenever I started it up the FTB would stay white and flicker every so often rapidly. I then decided to reinstall the pre-installed image the supplier used and the exact same response was given to me by the raspberry/LCD screen.

After fiddling around I plugged in the HDMI cable to a monitor and then the boot up console (I think) decided to start. As this was strange enough as it is, I then was displayed with the message shown in the photo I have provided. "Led pin not set, exiting."

This seemed to happen both with my own version of raspbian and the image that was supplied from company of whom I purchased it from. 

Does anyone have any ideas or care to assist me in troubleshooting and fixing this problem? Or is it ultimately that the LCD is faulty and need returning? 

Thanks, O-L

[image]

notro, 2016-05-21:
You are probably using the custom argument for fbtft_device. Apparently it doesn't handle displays without gpio backlight control. You can work around it by setting a gpio for the backlight that isn't connected to anything: led:<gpio>
This should let the driver load.

O-Luhishi, 2016-05-21:
I think I understand, however I don't have any idea as to how I would going about to do this. Possible to point me in the direction?

Thanks

notro, 2016-05-21:
Sorry, I thought that this error would prevent probing of the driver, but it doesn't.
If the display doesn't work with the image provided by the seller, I would guess that it's broken.
You can try and lower the speed to see if that helps.

Sidenote:
I looked at flexfb and it has a **nobacklight** module argument, so it's possible to get rid of that error message.

closed 2017-02-22 by notro