notro/fbtft #396: Raspberry PI3B and SKU:363295: error -16

obor00, 2016-05-29:
When trying to install this Touchscreen: [http://www.raspberrypiwiki.com/index.php/3.5_inch_TFT_LCD_Touch_Screen_SKU:363295]
on a Raspberry PI3B with Raspbian Jessie, I get this error from flexfb:

> [ 1128.645124] flexfb spi0.0: fbtft_request_gpios: gpio_request_one('reset'=25) failed with -16
[ 1128.645357] flexfb: probe of spi0.0 failed with error -16
[ 1128.645398] fbtft_device: GPIOS used by 'flexfb':
[ 1128.645408] fbtft_device: 'reset' = GPIO25
[ 1128.645417] fbtft_device: 'dc' = GPIO24
[ 1128.645436] ads7846 spi0.1: ads7846 spi0.1 10kHz 8 bits mode=0x00
[ 1128.645473] spi spi0.0: flexfb spi0.0 16000kHz 8 bits mode=0x00

Any idea why this error "-16" comes from ?

notro, 2016-05-30:
-16 == -EBUSY
This means that someone else has already requested gpio25.

obor00, 2016-05-30:
So following the docs of the touchscreen, I replaced the GPIO value reset and DC (13 and 15). I added also cs.
I wrote the script: 

> rmmod -f flexfb fbtft fbtft_device ads7846
modprobe flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
modprobe fbtft_device debug=3 rotate=90 name=flexfb **speed=48000000 gpios=reset:13,dc:15,cs:24**
modprobe ads7846 model=7846 cs=1 gpio_pendown=17  keep_vref_on=1 swap_xy=1 pressure_max=255 x_plate_ohms=60 x_min=200 x_max=3900 y_min=200 y_max=3900

Now I get a new error from dmesg:

> [  284.996888] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[  284.998890] flexfb: module is from the staging directory, the quality is unknown, you have been warned.
[  285.009228] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[  285.010312] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
**[  285.010353] fbtft_device: spi_busnum_to_master(0) returned NULL
[  285.010361] fbtft_device: failed to register SPI device**

Any idea ?

notro, 2016-05-31:
> fbtft_device: spi_busnum_to_master(0) returned NULL

This means that the spi controller device/driver is not loaded.

From https://github.com/notro/fbtft/wiki#install:
To use fbtft_device instead of a DT overlay, add to /boot/config.txt:

dtparam=spi=on

obor00, 2016-05-31:
Thanks Notro  for your help. Now I have no more error in dmesg,, but still have nothing on TFT screen when doing a "FRAMEBUFFER=/dev/fb1 start".

The result of dmesg:

[  105.059735] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[  105.063704] flexfb: module is from the staging directory, the quality is unknown, you have been warned.
[  105.076436] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[  105.077145] spidev spi0.0: spidev spi0.0 500kHz 8 bits mode=0x00
[  105.077161] spidev spi0.1: spidev spi0.1 500kHz 8 bits mode=0x00
[  105.077184] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
[  105.077214] spidev spi0.0: Deleting spi0.0
[  105.078093] flexfb spi0.0: fbtft_request_gpios: 'reset' = GPIO13
[  105.078116] flexfb spi0.0: fbtft_request_gpios: 'dc' = GPIO15
[  105.078127] flexfb spi0.0: fbtft_request_gpios: 'cs' = GPIO24
[  105.078134] flexfb spi0.0: flexfb_verify_gpios_dc()
[  105.078142] flexfb spi0.0: fbtft_reset()
[  105.198192] flexfb spi0.0: init: write(0xB0) 0x00 
[  105.198273] flexfb spi0.0: init: write(0x11) 
[  105.198289] flexfb spi0.0: init: mdelay(250)
[  105.448311] flexfb spi0.0: init: write(0x3A) 0x55 
[  105.448384] flexfb spi0.0: init: write(0xC2) 0x44 
[  105.448407] flexfb spi0.0: init: write(0xC5) 0x00 0x00 0x00 0x00 
[  105.448437] flexfb spi0.0: init: write(0xE0) 0x0F 0x1F 0x1C 0x0C 0x0F 0x08 0x48 0x98 0x37 0x0A 0x13 0x04 0x11 0x0D 0x00 
[  105.448474] flexfb spi0.0: init: write(0xE1) 0x0F 0x32 0x2E 0x0B 0x0D 0x05 0x47 0x75 0x37 0x06 0x10 0x03 0x24 0x20 0x00 
[  105.448510] flexfb spi0.0: init: write(0xE2) 0x0F 0x32 0x2E 0x0B 0x0D 0x05 0x47 0x75 0x37 0x06 0x10 0x03 0x24 0x20 0x00 
[  105.448537] flexfb spi0.0: init: write(0x36) 0x28 
[  105.448556] flexfb spi0.0: init: write(0x11) 
[  105.448570] flexfb spi0.0: init: write(0x29) 
[  105.512404] flexfb spi0.0: Display update: 4701 kB/s, fps=0
[  105.512418] flexfb spi0.0: fbtft_register_backlight(): led pin not set, exiting.
[  105.512678] graphics fb1: flexfb frame buffer, 480x320, 300 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 48 MHz
[  105.512702] fbtft_device: GPIOS used by 'flexfb':
[  105.512708] fbtft_device: 'reset' = GPIO13
[  105.512712] fbtft_device: 'dc' = GPIO15
[  105.512716] fbtft_device: 'cs' = GPIO24
[  105.512726] spidev spi0.1: spidev spi0.1 500kHz 8 bits mode=0x00
[  105.512734] flexfb spi0.0: flexfb spi0.0 48000kHz 8 bits mode=0x00
root@raspberrypi:/home/pi# cat obo_lcd_driver 

Do you any idea what to do next ?

notro, 2016-06-03:
You can try and lower the speed. Apart from that the usual problems are wiring and a wrong init sequence

obor00, 2016-06-03:
ok, I finally succeed to configure that TFT touch screen on a Raspbian Jessie and Pi3. Here is what I did:
Start from a Raspbian jessie

```
Edit /boot/config.txt and add:

start_x=1
gpu_mem=128
core_freq=250
sdram_freq=400
over_voltage=0
dtparam=spi=on
dtoverlay=waveshare35a
dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=1,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,
dtparam=i2c_arm=on
dtoverlay=w1-gpio-pullup,gpiopin=4,extpullup=1

copy  waveshare35a.dtbo   to /boot/overlays/   

get the package xinput-calibrator:
wget http://www.raspberrypiwiki.com/download/images/xinput-calibrator.deb

sudo DISPLAY=:0.0 xinput_calibrator

copy the result conf lines to:  /usr/share/X11/xorg.conf.d/01-input.conf

add to  01-input.conf before end of section:
        Option "InvertX" "true" 
        Option "InvertY" "true" 

```

closed 2017-02-22 by notro