notro/fbtft #397: evtest success but console not un-blanking

diginfo, 2016-06-12:
Hi Notro;

I have a display that uses the xpt2046 which I understand is a nock-off of the ads-7846.

I'm using the rpi-display overlay without modification.

evtest can detect the touches and the x and y coord look correct, however the console does not un-blank when I touch the panel.

Any suggestions on what the reason for this may be ?

notro, 2016-06-12:
I assume X windows here. When I looked at backlight handling almost 2 years ago I had to make a script to even get it to turn off: https://github.com/notro/fbtft-spindle/wiki/Appendix#x-windows-backlight-script
rpi-display uses fb_ili9341 which doesn't support .fb_blank (fbtftops.blank) so I don't know how X windows turns off your backlight.

diginfo, 2016-06-12:
Thanks, but no this is the console, not x-windows.

Touching the screen fires the evtest events, but does not un-blank the display.

If I connect a keyboard and press a key, the display is un-blanked ?!

notro, 2016-06-13:
Oh, AFAIK only keypresses wakes the console.

diginfo, 2016-06-18:
Thanks, that would be logical since a text console does not support a mouse.

closed 2016-06-18 by diginfo